esp-rs/esp-idf-svc #3: ESP32-C3: Wifi access point remains in `is_transitional` state, causing an infinite loop

wassup-, 2021-08-01:
I'm trying to put my NodeMCU ESP32-C3M-Kit into Access Point mode.
This seems to work correctly, because I can see the AP in my list of networks, connect to the AP, and get assigned an IP address via DHCP.

However, it seems that the Wifi status remains in the `is_transitional` state, which causes an infinite loop in `EspWifi::wait_status_with_timeout`, and eventually a reboot of the device.

This is the code I'm using:
```rs
pub fn access_point() -> Result<Box<impl Wifi>> {
    let mut wifi = Box::new(EspWifi::new(
        Arc::new(EspNetif::new()?),
        Arc::new(EspSysLoop::new()?),
        Arc::new(EspDefaultNvs::new()?),
    )?);

    wifi.set_configuration(&Configuration::AccessPoint(AccessPointConfiguration {
        max_connections: 16,
        ..Default::default()
    }))?;

    // we never get here...
    debug!("Wifi access point started");
}
```

Relevant output:
```
I (586) esp_idf_svc::wifi: Driver initialized
I (586) esp_idf_svc::wifi: Event handlers registered
I (596) esp_idf_svc::wifi: Initialization complete
I (606) esp_idf_svc::wifi: Setting configuration: AccessPoint(AccessPointConfiguration { ssid: "iot-device", ssid_hidden: false, channel: 1, secondary_channel: None, protocols: EnumSet(P802D11B | P802D11BG | P802D11BGN), auth_method: None, password: "", max_connections: 16, ip_conf: Some(RouterConfiguration { subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dhcp_enabled: true, dns: Some(8.8.8.8), secondary_dns: Some(8.8.4.4) }) })
I (636) esp_idf_svc::wifi: Stopping
I (646) esp_idf_svc::wifi: Disconnect requested
I (646) esp_idf_svc::wifi: Stop requested
I (656) esp_idf_svc::wifi: About to wait for status
I (656) esp_idf_svc::wifi: Providing status: Status(Stopped, Stopped)
I (666) esp_idf_svc::wifi: Waiting for status done - success
I (676) esp_idf_svc::wifi: Stopped
I (676) esp_idf_svc::wifi: Wifi mode AP set
I (686) esp_idf_svc::wifi: Setting AP configuration: AccessPointConfiguration { ssid: "iot-device", ssid_hidden: false, channel: 1, secondary_channel: None, protocols: EnumSet(P802D11B | P802D11BG | P802D11BGN), auth_method: None, password: "", max_connections: 16, ip_conf: Some(RouterConfiguration { subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dhcp_enabled: true, dns: Some(8.8.8.8), secondary_dns: Some(8.8.4.4) }) }
I (726) esp_idf_svc::wifi: Setting AP IP configuration: RouterConfiguration { subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dhcp_enabled: true, dns: Some(8.8.8.8), secondary_dns: Some(8.8.4.4) }
I (736) esp_idf_svc::wifi: AP netif allocated: 0x3fca4e88
I (746) esp_idf_svc::wifi: AP IP configuration done
I (756) esp_idf_svc::wifi: AP configuration done
I (756) esp_idf_svc::wifi: Starting with status: Status(Stopped, Starting)
I (766) esp_idf_svc::wifi: Status is of operating type, starting
I (776) phy_init: phy_version 500,985899c,Apr 19 2021,16:05:08
I (886) wifi:set rx active PTI: 0, rx ack PTI: 0, and default PTI: 0
I (886) wifi:mode : softAP (7c:df:a1:a4:44:c1)
I (886) wifi:Total power save buffer number: 16
I (896) wifi:Init max length of beacon: 752/752
I (896) wifi:Init max length of beacon: 752/752
I (906) esp_idf_svc::wifi: Got wifi event: 12 
I (906) esp_idf_svc::wifi: Set status: Status(Stopped, Started(Waiting))
I (916) esp_idf_svc::wifi: Wifi event 12 handled
I (916) esp_idf_svc::wifi: Start requested
I (926) esp_idf_svc::wifi: About to wait for status with timeout 10s
I (926) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (5936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (10936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (15936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (20936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (25936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (30936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (35936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (40936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (45936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (50936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (55936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (60936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
I (65936) esp_idf_svc::wifi: Providing status: Status(Stopped, Started(Waiting))
[... 180 lines cut ...]
3fc986f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98710: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98730: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98750: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98770: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98790: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc987b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc987d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc987f0: 0x00000000 0x00000000 0x00000000 0x00000000 0xffffffff 0x3fca5f84 0x3fc960e0 0x3fc98c20
3fc98810: 0x420108a8 0x00000005 0x3fc8e050 0x42089de8 0x3fc967c0 0x00000000 0x00000000 0x00000000
3fc98830: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98850: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98870: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98890: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc988b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc988d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc988f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98910: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 0000000000000000

Rebooting...
```

Let me know if you need more information, or if I can help in any way.

ivmarkov, 2021-08-01:
Nope, the issue is as clear as it gets. :) Take a look [here](https://github.com/esp-rs/esp-idf-svc/blob/master/src/wifi.rs#L771).
Basically, once the AP is up, we are still "Waiting" for an ["IP" event to come up](https://github.com/esp-rs/esp-idf-svc/blob/master/src/wifi.rs#L785). However, the event that should come (ip_event_t_IP_EVENT_AP_STAIPASSIGNED) we are actually NOT processing, while we should. Let me see if I can fix it quickly.

ivmarkov, 2021-08-01:
OK, so I don't have time to test it ATM, as I'm trying to push the libstd upstreaming effort forward.
However, I've pushed a provisional fix in `esp-idf-svc`.

I've published to crates.io new versions of both `esp-idf-svc` and `embedded-svc`, so if you can test these, that would be great! Please do issue `cargo update` before trying to rebuild your binary crate.

ivmarkov, 2021-08-01:
Testing: nope, that's not it. IP_EVENT_AP_STAIPASSIGNED is not really what we need.
Will follow up.

ivmarkov, 2021-08-01:
Now should work. It seems we'll just not be notified with an IP event that the AP is IP-configured, as there is nothing to get notified about. The IP configuration is static...

closed 2021-08-01 by ivmarkov