esp-rs/esp-idf-svc #4: ESP32-C3-DevKitM-1: Boot loop when scanning for wifi APs

cdata, 2021-08-24:
Hello, I am following along with the instructions to run https://github.com/ivmarkov/rust-esp32-std-hello on my ESP32-C3-DevKitM-1.

I was able to build and flash the demo application to my board.

Unfortunately, it appears to boot loop when scanning for access points. Here is the full output from a single loop iteration:

```
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403842e6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x14
load:0x3fcd6114,len:0x17dc
load:0x403ce000,len:0x86c
load:0x403d0000,len:0x29b4
entry 0x403ce000
I (31) boot: ESP-IDF qa-test-v4.3-20201113-876-gd6a9 2nd stage bootloader
I (31) boot: compile time 19:39:33
I (32) boot: chip revision: 0
I (41) boot.esp32c3: SPI Speed      : 80MHz
I (41) boot.esp32c3: SPI Mode       : DIO
I (45) boot.esp32c3: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
W (55) bootloader_random: RNG for ESP32-C3 not currently supported
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 sec_cert         unknown          3f 00 0000d000 00003000
I (80) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (88) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (95) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (103) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (110) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (118) boot:  6 fctry            WiFi data        01 02 00340000 00006000
I (125) boot:  7 coredump         Unknown data     01 03 00350000 00010000
I (133) boot: End of partition table
I (137) esp_image: segment 0: paddr=0x00020020 vaddr=0x3c0c0020 size=0x296b8 (169656) map
I (174) esp_image: segment 1: paddr=0x000496e0 vaddr=0x3fc8e200 size=0x03500 ( 13568) load
I (177) esp_image: segment 2: paddr=0x0004cbe8 vaddr=0x40380000 size=0x03430 ( 13360) load
I (183) esp_image: segment 3: paddr=0x00050020 vaddr=0x42000020 size=0xb3a90 (735888) map
I (311) esp_image: segment 4: paddr=0x00103ab8 vaddr=0x40383430 size=0x0ad2c ( 44332) load
I (321) esp_image: segment 5: paddr=0x0010e7ec vaddr=0x50000000 size=0x00010 (    16) load
I (326) boot: Loaded app from partition at offset 0x20000
I (326) boot: Disabling RNG early entropy source...
W (330) bootloader_random: RNG for ESP32-C3 not currently supported
I (348) cpu_start: Pro cpu up.
I (360) cpu_start: Pro cpu start user code
I (360) cpu_start: cpu freq: 160000000
I (360) cpu_start: Application information:
I (363) cpu_start: Project name:     esp-idf
I (368) cpu_start: App version:      29b6481-dirty
I (374) cpu_start: Compile time:     Aug 24 2021 13:13:42
I (380) cpu_start: ELF file SHA256:  0000000000000000...
I (386) cpu_start: ESP-IDF:          4.3.0
I (390) heap_init: Initializing. RAM available for dynamic allocation:
I (398) heap_init: At 3FC96300 len 00029D00 (167 KiB): DRAM
I (404) heap_init: At 3FCC0000 len 0001F260 (124 KiB): STACK/DRAM
I (411) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (417) spi_flash: detected chip: generic
I (422) spi_flash: flash io: dio
I (426) sleep: Configure to isolate all GPIO pins in sleep state
I (432) sleep: Enable automatic switching of GPIO sleep configuration
I (440) cpu_start: Starting scheduler.
Hello, world from Rust!
More complex print ["foo", "bar"]
Result: 0, 1
Rust main thread: Thread { id: ThreadId(1), name: Some("main"), .. }
This is thread number 0, Thread { id: ThreadId(2), name: None, .. }
This is thread number 1, Thread { id: ThreadId(3), name: None, .. }
[... 73 lines cut ...]
3fc9fed0: 0x00000000 0x3fe00000 0x00000000 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28
3fc9fef0: 0x509f79fb 0x3fd34413 0x00000000 0x3ff00000 0x00000000 0x40240000 0x00000000 0x401c0000
3fc9ff10: 0x00000000 0x40140000 0x00000000 0x43500000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154
3fc9ff30: 0x3fc9fd60 0x3fc96314 0x3fc91f24 0x3fc91f24 0x3fc9ff30 0x3fc91f1c 0x00000002 0x3fc9de68
3fc9ff50: 0x3fc9de68 0x3fc9ff30 0x00000000 0x00000017 0x3fc9e52c 0x69666977 0x5f1a5600 0xc88e8184
3fc9ff70: 0x00212e8e 0x00000000 0x3fc9ff20 0x00000017 0x00000001 0x00000000 0x00000000 0x00000000
3fc9ff90: 0x3fc96b2c 0x3fc96b94 0x3fc96bfc 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
3fc9ffb0: 0x00000000 0x42099f0c 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9ffd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fff0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca0010: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca0030: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca0050: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca0070: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc90000 0x00000054 0x3fca0088 0x3fca0088
3fca0090: 0x3fca0088 0x3fca0088 0x00000000 0x3fca00a0 0xffffffff 0x3fca00a0 0x3fca00a0 0x00000000
3fca00b0: 0x3fca00b4 0xffffffff 0x3fca00b4 0x3fca00b4 0x00000000 0x00000001 0x00000000 0xf400ffff
3fca00d0: 0x00000000 0x6d5a935a 0x3fca0080 0x00000054 0x00000000 0x3fca00e0 0x00000000 0x00000000
3fca00f0: 0x00000000 0x3fca00f8 0xffffffff 0x3fca00f8 0x3fca00f8 0x00000000 0x3fca010c 0xffffffff

ELF file SHA256: 0000000000000000

Rebooting...
```

Any insight into what could be going on would be appreciated. I am able to provide any further details if it is helpful.

cdata, 2021-08-30:
Hello, sorry for the delay in replying. I just did what you suggested. After getting the scan example to build at the 4.3 release, I flashed it and it appears to work with this output:

```
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x181c
load:0x403ce000,len:0x8d4
load:0x403d0000,len:0x293c
entry 0x403ce000
I (32) boot: ESP-IDF v4.3-dirty 2nd stage bootloader
I (32) boot: compile time 22:25:45
I (32) boot: chip revision: 2
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (44) boot.esp32c3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=16d10h ( 93456) map
I (114) esp_image: segment 1: paddr=00026d38 vaddr=3fc8dc00 size=03428h ( 13352) load
I (117) esp_image: segment 2: paddr=0002a168 vaddr=40380000 size=05eb0h ( 24240) load
I (124) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=82914h (534804) map
I (212) esp_image: segment 4: paddr=000b293c vaddr=40385eb0 size=07c58h ( 31832) load
I (218) esp_image: segment 5: paddr=000ba59c vaddr=50000000 size=00010h (    16) load
I (223) boot: Loaded app from partition at offset 0x10000
I (224) boot: Disabling RNG early entropy source...
I (241) cpu_start: Pro cpu up.
I (253) cpu_start: Pro cpu start user code
I (253) cpu_start: cpu freq: 160000000
I (253) cpu_start: Application information:
I (256) cpu_start: Project name:     scan
I (261) cpu_start: App version:      v4.3-dirty
I (266) cpu_start: Compile time:     Aug 29 2021 22:25:55
I (272) cpu_start: ELF file SHA256:  f8b948e4a87efb16...
I (278) cpu_start: ESP-IDF:          v4.3-dirty
I (283) heap_init: Initializing. RAM available for dynamic allocation:
I (290) heap_init: At 3FC95BA0 len 0002A460 (169 KiB): DRAM
I (296) heap_init: At 3FCC0000 len 0001F260 (124 KiB): STACK/DRAM
I (303) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (310) spi_flash: detected chip: generic
I (314) spi_flash: flash io: dio
W (318) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using t
he size in the binary image header.
I (332) sleep: Configure to isolate all GPIO pins in sleep state
I (338) sleep: Enable automatic switching of GPIO sleep configuration
I (345) cpu_start: Starting scheduler.
I (355) pp: pp rom version: 8459080
I (355) net80211: net80211 rom version: 8459080
I (365) wifi:wifi driver task: 3fc9ede4, prio:23, stack:6656, core=0
I (365) system_api: Base MAC address is not set
I (375) system_api: read default base MAC address from EFUSE
I (375) wifi:wifi firmware version: c7d0450
I (385) wifi:wifi certification version: v7.0
I (385) wifi:config NVS flash: enabled
I (385) wifi:config nano formating: disabled
I (395) wifi:Init data frame dynamic rx buffer num: 32
I (395) wifi:Init management frame dynamic rx buffer num: 32
I (405) wifi:Init management short buffer num: 32
I (405) wifi:Init dynamic tx buffer num: 32
I (415) wifi:Init static tx FG buffer num: 2
I (415) wifi:Init static rx buffer size: 1600
I (415) wifi:Init static rx buffer num: 10
I (425) wifi:Init dynamic rx buffer num: 32
I (425) wifi_init: rx ba win: 6
I (435) wifi_init: tcpip mbox: 32
I (435) wifi_init: udp mbox: 6
I (435) wifi_init: tcp mbox: 6
I (445) wifi_init: tcp tx win: 5744
I (445) wifi_init: tcp rx win: 5744
I (455) wifi_init: tcp mss: 1440
I (455) wifi_init: WiFi IRAM OP enabled
I (455) wifi_init: WiFi RX IRAM OP enabled
I (465) phy_init: phy_version 500,985899c,Apr 19 2021,16:05:08
I (585) wifi:set rx active PTI: 0, rx ack PTI: 0, and default PTI: 0
I (585) wifi:mode : sta (7c:df:a1:43:40:14)
I (585) wifi:enable tsf
I (2685) scan: Total APs scanned = 6
I (2685) scan: SSID             Macaque Moving Crew
I (2685) scan: RSSI             -40
I (2685) scan: Authmode         WIFI_AUTH_WPA2_PSK
I (2685) scan: Pairwise Cipher  WIFI_CIPHER_TYPE_CCMP
I (2695) scan: Group Cipher     WIFI_CIPHER_TYPE_CCMP
I (2695) scan: Channel          11

I (2705) scan: SSID             Macaque Guest
I (2705) scan: RSSI             -40
I (2705) scan: Authmode         WIFI_AUTH_WPA2_PSK
I (2715) scan: Pairwise Cipher  WIFI_CIPHER_TYPE_CCMP
I (2715) scan: Group Cipher     WIFI_CIPHER_TYPE_CCMP
I (2725) scan: Channel          11

I (2725) scan: SSID             Macaque Moving Crew
I (2735) scan: RSSI             -62
I (2735) scan: Authmode         WIFI_AUTH_WPA2_PSK
I (2745) scan: Pairwise Cipher  WIFI_CIPHER_TYPE_CCMP
I (2745) scan: Group Cipher     WIFI_CIPHER_TYPE_CCMP
I (2755) scan: Channel          11

I (2755) scan: SSID             Macaque Guest
I (2765) scan: RSSI             -63
I (2765) scan: Authmode         WIFI_AUTH_WPA2_PSK
I (2765) scan: Pairwise Cipher  WIFI_CIPHER_TYPE_CCMP
I (2775) scan: Group Cipher     WIFI_CIPHER_TYPE_CCMP
I (2775) scan: Channel          11

I (2785) scan: SSID             Lighthouse Primus
I (2785) scan: RSSI             -89
I (2795) scan: Authmode         WIFI_AUTH_WPA2_PSK
I (2795) scan: Pairwise Cipher  WIFI_CIPHER_TYPE_TKIP_CCMP
I (2805) scan: Group Cipher     WIFI_CIPHER_TYPE_TKIP
I (2805) scan: Channel          6

I (2815) scan: SSID             ORBI96-Guest
I (2815) scan: RSSI             -91
I (2815) scan: Authmode         WIFI_AUTH_WPA2_PSK
I (2825) scan: Pairwise Cipher  WIFI_CIPHER_TYPE_CCMP
I (2825) scan: Group Cipher     WIFI_CIPHER_TYPE_CCMP
I (2835) scan: Channel          8
```

rthomas, 2022-07-09:
Did you have any further info on this @cdata? I'm hitting the same issue on my `C3` devkit.

Vollbrecht, 2023-03-29:
@ivmarkov this might not be a problem anymore after commit 8aac6d0dd41953f16c58c85f14ebb7720c719309 ish or what do you think?

JuliDi, 2023-04-27:
The issue seems to persist with ESP IDF v5.0 (setup via espup and esp-idf-template).
I have fixed the toolchain to `channel = "nightly-2023-03-09"`
With IDF v4.4.1 everything worked fine.
<details><summary>Details</summary>
<pre>
<code>
I (5444) wifi:wifi firmware version: 57982fe
I (5444) wifi:wifi certification version: v7.0
I (5454) wifi:config NVS flash: enabled
I (5454) wifi:config nano formating: disabled
I (5454) wifi:Init data frame dynamic rx buffer num: 32
I (5464) wifi:Init management frame dynamic rx buffer num: 32
I (5464) wifi:Init management short buffer num: 32
I (5474) wifi:Init dynamic tx buffer num: 32
I (5474) wifi:Init static tx FG buffer num: 2
I (5484) wifi:Init static rx buffer size: 1600
I (5484) wifi:Init static rx buffer num: 10
I (5484) wifi:Init dynamic rx buffer num: 32
I (5494) wifi_init: rx ba win: 6
I (5494) wifi_init: tcpip mbox: 32
I (5504) wifi_init: udp mbox: 6
I (5504) wifi_init: tcp mbox: 6
I (5504) wifi_init: tcp tx win: 5744
I (5514) wifi_init: tcp rx win: 5744
I (5514) wifi_init: tcp mss: 1440
I (5524) wifi_init: WiFi IRAM OP enabled
I (5524) wifi_init: WiFi RX IRAM OP enabled
I (5524) esp_idf_svc::wifi: Driver initialized
I (5534) esp_idf_svc::wifi: Stop requested
I (5534) esp_idf_svc::wifi: Stopping
I (5544) mvg_monitor::wifi: Searching for Wifi network <NAME>
I (5554) esp_idf_svc::wifi: About to scan for access points
I (5554) esp_idf_svc::wifi: Stop requested
I (5564) esp_idf_svc::wifi: Stopping
I (5564) esp_idf_svc::wifi: Disconnect requested
I (5574) esp_idf_svc::wifi: Stop requested
I (5574) esp_idf_svc::wifi: Stopping
I (5584) wifi:Deinit lldesc rx mblock:10
I (5594) esp_idf_svc::wifi: Driver deinitialized
I (5594) esp_idf_svc::wifi: Dropped
I (5594) esp_idf_svc::nvs: Dropped
I (5594) esp_idf_svc::eventloop: Dropped
</code>
</pre>
</p>
</details>

Vollbrecht, 2023-04-27:
> The issue seems to persist with ESP IDF v5.0 (setup via espup and esp-idf-template). I have fixed the toolchain to `channel = "nightly-2023-03-09"` With IDF v4.4.1 everything worked fine.
> Details

your output doesn't imply you run the std-demo. could you clarify what you are actually doing? i don't see any loop here in your code just dropping out of context. you are scanning for any wifi AP's?

JuliDi, 2023-04-27:
No, there is no loop. Just scanning for an AP and suddenly it crashes.
Iirc I pretty much copy&pasted the code from the example, though.

```rust
pub fn wifi(
    ssid: &str,
    psk: &str,
    modem: impl peripheral::Peripheral<P = esp_idf_hal::modem::Modem> + 'static,
    sysloop: EspSystemEventLoop,
) -> anyhow::Result<Box<EspWifi<'static>>> {
    
    if ssid.is_empty() {
        anyhow::bail!("missing WiFi name")
    }
    if psk.is_empty() {
        info!("Wifi password is empty");
    }

    let mut wifi = Box::new(EspWifi::new(
        modem,
        sysloop.clone(),
        Some(EspDefaultNvsPartition::take()?),
    )?);

    info!("Searching for Wifi network {}", ssid);

    let ap_infos = wifi.scan()?;

    let ours = ap_infos.into_iter().find(|a| a.ssid == ssid);

    let channel = if let Some(ours) = ours {
        info!(
            "Found configured access point {} on channel {}",
            ssid, ours.channel
        );
        Some(ours.channel)
    } else {
        info!(
            "Configured access point {} not found during scanning, will go with unknown channel",
            ssid
        );
        None
    };

    info!("setting Wifi configuration");
    wifi.set_configuration(&wifi::Configuration::Client(ClientConfiguration {
        ssid: ssid.into(),
        password: psk.into(),
        channel,
        ..Default::default()
    }))?;

    wifi.start()?;
    info!("Starting wifi...");

    if !WifiWait::new(&sysloop)?
        .wait_with_timeout(Duration::from_secs(20), || wifi.is_started().unwrap())
    {
        bail!("Wifi did not start");
    }

    info!("Connecting wifi...");
    wifi.connect()?;

    while !EspNetifWait::new::<EspNetif>(wifi.sta_netif(), &sysloop)?.wait_with_timeout(
        Duration::from_secs(7),
        || {
            wifi.is_connected().unwrap()
                && wifi.sta_netif().get_ip_info().unwrap().ip != Ipv4Addr::new(0, 0, 0, 0)
        },
    ) {
        wifi.connect()?;
    }

    let ip_info = wifi.sta_netif().get_ip_info()?;

    info!("Wifi DHCP info: {:?}", ip_info);

    Ok(wifi)
}
```

Vollbrecht, 2023-04-27:
@JuliDi your code is not working because you get an Error: ESP_ERR_WIFI_NOT_STARTED . and as the name suggest -> you should use wifi.start() before the first wifi.scan() than it is working, at least it does for me tested on my esp32c3 hardware. So it has nothing todo with the initial issue, can you confirm that?

Vollbrecht, 2023-04-27:
keep in mind: if you are using wokwi to simulate this, this also don't work, but works on real hardware!

JuliDi, 2023-04-28:
This kinda fixed it! Thank you :)

Stil, it is somewhat interesting. When I am on v4.4.1 and make the change as you suggested, it fails (though it works for 5.0)! See here:
```
I (5282) esp_idf_svc::wifi: Driver initialized
I (5282) esp_idf_svc::wifi: Stop requested
I (5282) esp_idf_svc::wifi: Stopping
I (5282) esp_idf_svc::wifi: Start requested
I (5392) esp_idf_svc::wifi: Starting
I (5392) mvg_monitor::wifi: Starting wifi...
I (5392) esp_idf_svc::wifi: About to wait for duration 20s
I (5392) esp_idf_svc::wifi: Waiting done - success
I (5402) mvg_monitor::wifi: Searching for Wifi network <NAME>
I (5402) esp_idf_svc::wifi: About to scan for access points
I (5412) esp_idf_svc::wifi: Disconnect requested
I (5422) esp_idf_svc::wifi: Disconnecting
I (5422) esp_idf_svc::wifi: Stop requested
I (5432) esp_idf_svc::wifi: Stopping
I (7532) esp_idf_svc::wifi: Found 4 access points
I (7532) esp_idf_svc::wifi: About to get info for found access points
```

closed 2023-09-21 by ivmarkov