esp-rs/esp-idf-svc #6: Documentation fails to build

lily-mara, 2021-09-24:
The docs do not build on docs.rs

[build log](https://docs.rs/crate/esp-idf-svc/0.20.1/builds/437848)

```
# rustc version
rustc 1.57.0-nightly (5ecc8ad84 2021-09-19)
# docs.rs version
docsrs 0.6.0 (432e30a 2021-09-12)

# build log
[INFO] running `Command { std: "docker" "create" "-v" "/home/cratesfyi/workspace/builds/esp-idf-svc-0.20.1/target:/opt/rustwide/target:rw,Z" "-v" "/home/cratesfyi/workspace/builds/esp-idf-svc-0.20.1/source:/opt/rustwide/workdir:ro,Z" "-v" "/home/cratesfyi/workspace/cargo-home:/opt/rustwide/cargo-home:ro,Z" "-v" "/home/cratesfyi/workspace/rustup-home:/opt/rustwide/rustup-home:ro,Z" "-e" "SOURCE_DIR=/opt/rustwide/workdir" "-e" "CARGO_TARGET_DIR=/opt/rustwide/target" "-e" "DOCS_RS=1" "-e" "CARGO_HOME=/opt/rustwide/cargo-home" "-e" "RUSTUP_HOME=/opt/rustwide/rustup-home" "-w" "/opt/rustwide/workdir" "-m" "3221225472" "--cpus" "2" "--user" "1001:1001" "--network" "none" "ghcr.io/rust-lang/crates-build-env/linux@sha256:a89201d6b5b9fd45c15d5952ece0a0129e7e80cb26ec023fd59cf09bc26f1604" "/opt/rustwide/cargo-home/bin/cargo" "+nightly" "rustdoc" "--lib" "-Zrustdoc-map" "-Zunstable-options" "--config=doc.extern-map.registries.crates-io=\"https://docs.rs/{pkg_name}/{version}/x86_64-unknown-linux-gnu\"" "-j2" "--" "-Z" "unstable-options" "--emit=invocation-specific" "--resource-suffix" "-20210919-1.57.0-nightly-5ecc8ad84" "--static-root-path" "/" "--cap-lints" "warn" "--disable-per-crate-search", kill_on_drop: false }`
[INFO] [stdout] f9c12e04431be89cc02118eae1b9c9eee59977444a9b61c055ef566694544387
[INFO] [stderr] WARNING: Your kernel does not support swap limit capabilities or the cgroup is not mounted. Memory limited without swap.
[INFO] running `Command { std: "docker" "start" "-a" "f9c12e04431be89cc02118eae1b9c9eee59977444a9b61c055ef566694544387", kill_on_drop: false }`
[INFO] [stderr]    Compiling esp-idf-sys v0.20.0
[INFO] [stderr] error: failed to run custom build command for `esp-idf-sys v0.20.0`
[INFO] [stderr] 
[INFO] [stderr] Caused by:
[INFO] [stderr]   process didn't exit successfully: `/opt/rustwide/target/debug/build/esp-idf-sys-a2bf8941e6b245cc/build-script-build` (exit status: 1)
[INFO] [stderr]   --- stdout
[INFO] [stderr]   Installer version: 1.0.2
[INFO] [stderr]   Platform: Linux-5.4.0-1037-aws-x86_64-with-glibc2.29
[INFO] [stderr]   Python version: 3.8.10 (default, Jun  2 2021, 10:49:15) 
[INFO] [stderr]   [GCC 9.4.0]
[INFO] [stderr]   Python path: /usr/bin/python3
[INFO] [stderr]   Creating a virtual environment at /.platformio/penv
[INFO] [stderr]   /usr/bin/python3
[INFO] [stderr] 
[INFO] [stderr]   --- stderr
[INFO] [stderr]   Error: Compatible PlatformIO Core not found.
[INFO] [stderr]   Reason: PlatformIO executable not found in `/.platformio/penv/bin`
[INFO] [stderr]   Traceback (most recent call last):
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connection.py", line 169, in _new_conn
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/util/connection.py", line 73, in create_connection
[INFO] [stderr]     File "/usr/lib/python3.8/socket.py", line 918, in getaddrinfo
[INFO] [stderr]       for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
[INFO] [stderr]   socket.gaierror: [Errno -3] Temporary failure in name resolution
[INFO] [stderr] 
[INFO] [stderr]   During handling of the above exception, another exception occurred:
[INFO] [stderr] 
[INFO] [stderr]   Traceback (most recent call last):
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connectionpool.py", line 699, in urlopen
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connectionpool.py", line 382, in _make_request
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connectionpool.py", line 1010, in _validate_conn
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connection.py", line 353, in connect
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connection.py", line 181, in _new_conn
[INFO] [stderr]   urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7f96e5f5b340>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution
[INFO] [stderr] 
[INFO] [stderr]   During handling of the above exception, another exception occurred:
[INFO] [stderr] 
[INFO] [stderr]   Traceback (most recent call last):
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/requests/adapters.py", line 439, in send
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connectionpool.py", line 755, in urlopen
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/util/retry.py", line 574, in increment
[INFO] [stderr]   urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='api.registry.platformio.org', port=443): Max retries exceeded with url: /v3/packages/platformio/tool/python-portable (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f96e5f5b340>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
[INFO] [stderr] 
[INFO] [stderr]   During handling of the above exception, another exception occurred:
[INFO] [stderr] 
[INFO] [stderr]   Traceback (most recent call last):
[INFO] [stderr]     File "/tmp/.tmpJjdIul", line 68, in <module>
[INFO] [stderr]       main()
[INFO] [stderr]     File "/tmp/.tmpJjdIul", line 60, in main
[INFO] [stderr]       bootstrap()
[INFO] [stderr]     File "/tmp/.tmpJjdIul", line 46, in bootstrap
[INFO] [stderr]       pioinstaller.__main__.main()
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/pioinstaller/__main__.py", line 131, in main
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/click/core.py", line 829, in __call__
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/click/core.py", line 782, in main
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/click/core.py", line 1236, in invoke
[... 11 lines cut ...]
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/requests/sessions.py", line 542, in request
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/requests/sessions.py", line 655, in send
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/requests/adapters.py", line 516, in send
[INFO] [stderr]   requests.exceptions.ConnectionError: HTTPSConnectionPool(host='api.registry.platformio.org', port=443): Max retries exceeded with url: /v3/packages/platformio/tool/python-portable (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f96e5f5b340>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
[INFO] [stderr]   Error: Compatible PlatformIO Core not found.
[INFO] [stderr]   Reason: PlatformIO executable not found in `/.platformio/penv/bin`
[INFO] [stderr]   Error: EOF while parsing a value at line 1 column 0
[INFO] running `Command { std: "docker" "inspect" "f9c12e04431be89cc02118eae1b9c9eee59977444a9b61c055ef566694544387", kill_on_drop: false }`
[INFO] running `Command { std: "docker" "rm" "-f" "f9c12e04431be89cc02118eae1b9c9eee59977444a9b61c055ef566694544387", kill_on_drop: false }`
[INFO] [stdout] f9c12e04431be89cc02118eae1b9c9eee59977444a9b61c055ef566694544387
```

It seems that the docs.rs build environment does not have a networking stack available, so the platformio installation script fails. Is there any way to provide fallback information in the build.rs script so that the docs can be built without networking? I'm also not able to build the docs locally:

```
error: failed to run custom build command for `esp-idf-sys v0.20.1 (/Users/lily/projects/github.com/esp-rs/esp-idf-sys)`

Caused by:
  process didn't exit successfully: `/Users/lily/projects/github.com/esp-rs/esp-idf-sys/target/debug/build/esp-idf-sys-84056bef3f12ccb1/build-script-build` (exit status: 1)
  --- stdout
  Found compatible PlatformIO Core 5.2.0 -> /Users/lily/.platformio/penv/bin/platformio

  --- stderr
  Error: Cannot derive default PIO platform, MCU and frameworks for target 'aarch64-apple-darwin'
```

lily-mara, 2021-09-24:
Oh I opened this on the wrong project, I'm very sorry.

closed 2021-09-24 by lily-mara

ivmarkov, 2021-09-24:
I don't know which is the correct project for this issue. Might well be this very project. :)

The problem is not the Internet. It is here:
` Error: Cannot derive default PIO platform, MCU and frameworks for target 'aarch64-apple-darwin'`

The problem is that the `esp-idf-sys` crate - and by extension - any crate that depends on that one (like `esp-idf-svc`) **cannot** be built for standard Rust targets like `aarch64-apple-darwin`. 
Only for ESP targets, like the `*-espidf` ones.

Now, I've never even looked into what it would take to build the docs for `esp-idf-sys` and friends. Btw, are there any docs in these crates really? I don't think so. In any case, this needs some attention, so if you want, you can actually reopen this issue.

ivmarkov, 2021-09-24:
Hmm, you are right that it is actually two things: a) Internet b) the need for a custom target during the build

lily-mara, 2021-09-24:
Since the issue is in `esp-idf-sys`, lets move the discussion to https://github.com/esp-rs/esp-idf-sys/issues/15