esp-rs/esp-idf-svc #7: Allow usage in a CMake-based build

maelp, 2021-09-29:
Hi (cross-linking from https://github.com/espressif/rust-esp32-example/issues/34)

thanks so much for your work on embedded rust, and in particular on ESP chips :)

We're a company using ESP on our hardware, and we'd love to move our codebase to Rust. Because that move will be incremental, we'd like to compile our C/C++ codebase and switch some modules to Rust incrementally. For this, it seems more natural to use a "CMake-based" build that would compile our Rust modules as a library, and link our code to it. And of course, we'd love to use `esp-idf-svc` :)

For now, we saw https://github.com/ivmarkov/rust-esp32-std-hello which uses `esp-idf-svc`, but uses a "cargo-first" build which seems less natural for us.

Would it be possible to allow building `esp-idf-svc` using a CMake build so that it can be included in a CMake project like for instance https://github.com/espressif/rust-esp32-example ?

ivmarkov, 2021-10-01:
@igrr, @georgik - to make the CMake-first build compatible with `esp-idf-sys` (and this with `esp-idf-svc` and friends), we really only need to define three additional environment variables and pass them down to Cargo [here](https://github.com/espressif/rust-esp32-example/blob/main/components/rustlib/CMakeLists.txt#L37). The names of the env variables are completely arbitrary:
* `CARGO_CMAKE_BUILD_INCLUDES` - this should be a delimited list of all include directories of all ESP-IDF components. If that makes it easier, having additional - user-components - on that list will not be a problem, I think. Btw, I tried with `get_target_property(CARGO_CMAKE_BUILD_INCLUDES ${COMPONENT_TARGET} COMPONENT_INCLUDES)`
* `CARGO_CMAKE_BUILD_SDKCONFIG` - this should be the full path to the `sdkconfig` file used during the build. A less ergonomic alternative would be for this environment variable to directly contain all `CONFIG_*` options - somehow delimited - from the `sdkconfig`
* `CARGO_CMAKE_BUILD_COMPILER` - the full path to the compiler (as in GCC) used during the build. We need this so as to derive a proper sysroot for the `Bindgen` utility.

Any help (in terms of `CMake` syntax) as to from where to get the content of these three variables would be greatly appreciated!

ivmarkov, 2021-10-05:
@georgik Can you help with ^^^? I have the (minimal) changes to the `esp-idf-sys` crate ready, but unless we figure out the above problem, there is no way to test these...

maelp, 2021-10-05:
That would be really nice if we could do it, we'd love to be able to start hacking on Rust with `esp-idf-svc` :)

ivmarkov, 2021-10-06:
* Necessary changes (6 lines) to a CMake project embedding rust lib [here](https://github.com/espressif/rust-esp32-example/pull/38).
* Patch to `esp-idf-sys` to support CMake-first projects [here](https://github.com/esp-rs/esp-idf-sys/pull/17).