esp-rs/esp-idf-svc #22: Map ESP-IDF's mbedTLS to rust-native-tls

ivmarkov, 2021-11-08:
... or alternatively, map ESP-IDF's partial OpenSSL implementation to `rust-openssl`.

Both solutions would require a lot of work and will likely result in patches against one of these upstream crates.

igrr, 2021-11-08:
Just a note, esp-idf will drop support for OpenSSL wrapper in 5.0 release. If it is necessary for rust, there is an option to extract the wrapper into a separate component, and maintain it as part for esp-rs organization.

MabezDev, 2021-11-22:
I think our best bet is to add support for mbedtls to `rust-native-tls`. There is [rust-mbedtls](https://github.com/fortanix/rust-mbedtls), however we probably need to PR some changes so that it it's possible to link against the mbedtls component in esp-idf instead of [building from source](https://github.com/fortanix/rust-mbedtls/blob/cef4e6995ab8dd31560136076e6320920c669955/mbedtls-sys/build/build.rs#L121).

I'll look into building rust-mbedtls against `esp-idf`, and I'll also open an issue upstream in `rust-native-tls` to make sure they are  open to the implementation.

MabezDev, 2022-11-04:
See https://github.com/sfackler/rust-native-tls/issues/211#issuecomment-1301049087 for an update

bsodmike, 2022-11-04:
Thanks @MabezDev

juliankrieger, 2024-06-27:
Is this issue still open? Brian Smith seems to have added support for little endian architectures in Ring a couple of weeks ago. See more here: https://www.reddit.com/r/rust/comments/1d7zzb8/which_tls_crate_do_you_recommend_to_use_for_an/

It seems like this commit (https://github.com/briansmith/ring/commit/ed5b2a86cccfe30e670b33536fc7e494fb5e9e19) is not yet included in a release and I've not yet tried it, but it seems like `ring` should then work on esp32 systems?

danielmeza, 2025-07-25:
@juliankrieger have you confirm the last comment?