esp-rs/esp-idf-svc #34: Any plans to support esp8266?

ganeshrvel, 2021-12-13:
I would love to see the support for esp8266 getting added as well. Are there any plans for it?

MabezDev, 2021-12-13:
Due to the fact that the esp8266 is not currently supported by esp-idf, this is unlikely. There is a bare metal hal: https://github.com/esp-rs/esp8266-hal available, in which most peripherals are covered but unfortunately WiFi/BT does not yet work.

closed 2021-12-13 by MabezDev

ganeshrvel, 2021-12-13:
I have been using the 8266 HAL. But I'm looking for wifi connectivity support which is absent in it. 

Is there a plan to support WiFi in 8266 HAL?

MabezDev, 2021-12-13:
I'm not aware of anyone actively working on it, but it should be possible to use the standalone binary blobs from https://github.com/espressif/ESP8266_NONOS_SDK/tree/master/lib to interact with the WiFi/BT hardware.