esp-rs/esp-idf-svc #35: ESP32-C3 stack overflow for wifi in a thread

BrendanBall, 2022-01-02:
Hi

I'm getting a stackoverflow when trying to connect to wifi in a thread on the C3. 
See https://github.com/BrendanBall/light-switch/blob/wifi_in_thread/src/main.rs . It is mostly copied from the demo project.

monitor output:
```
➜  light-switch git:(master) ✗ cargo espmonitor --chip esp32c3 /dev/ttyUSB0
ESPMonitor 0.5.2

Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

Opening /dev/ttyUSB0 with speed 115200
WARNING: Flash image /home/brendan/dev/projects/light-switch/target/riscv32imc-unknown-none-elf/debug/light-switch does not exist (you may need to build it)
Resetting device... done
2) spi_flash: �ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x403ce000,len:0x928
load:0x403d0000,len:0x2ce0
entry 0x403ce000
I (30) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (30) boot: compile time 12:10:40
I (30) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32c3: SPI Speed      : 80MHz
I (45) boot.esp32c3: SPI Mode       : DIO
I (50) boot.esp32c3: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table
I (98) boot_comm: chip revision: 3, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=1e4d0h (124112) map
I (132) esp_image: segment 1: paddr=0002e4f8 vaddr=3fc8e800 size=01b20h (  6944) load
I (133) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=968d8h (616664) map
I (230) esp_image: segment 3: paddr=000c6900 vaddr=3fc90320 size=01830h (  6192) load
I (231) esp_image: segment 4: paddr=000c8138 vaddr=40380000 size=0e604h ( 58884) load
I (247) esp_image: segment 5: paddr=000d6744 vaddr=50000000 size=00010h (    16) load
I (251) boot: Loaded app from partition at offset 0x10000
I (251) boot: Disabling RNG early entropy source...
I (267) cpu_start: Pro cpu up.
I (280) cpu_start: Pro cpu start user code
I (280) cpu_start: cpu freq: 160000000
I (280) cpu_start: Application information:
I (282) cpu_start: Project name:     esp-idf
I (287) cpu_start: App version:      e596c0d-dirty
I (293) cpu_start: Compile time:     Dec 27 2021 16:32:26
I (299) cpu_start: ELF file SHA256:  0000000000000000...
I (305) cpu_start: ESP-IDF:          4.3.1
I (310) heap_init: Initializing. RAM available for dynamic allocation:
I (317) heap_init: At 3FC95C10 len 0002A3F0 (168 KiB): DRAM
I (323) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (330) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (337) spi_flash: detected chip: generic
I (341) spi_flash: flash io: dio
I (345) sleep: Configure to isolate all GPIO pins in sleep state
I (352) sleep: Enable automatic switching of GPIO sleep configuration
I (359) cpu_start: Starting scheduler.
I (364) pp: pp rom version: 9387209
I (364) net80211: net80211 rom version: 9387209
***ERROR*** A stack overflow in task pthread has been detected.
Core  0 register dump:
MEPC    : 0x403858a2  RA      : 0x40386018  SP      : 0x3fc92100  GP      : 0x3fc8f000
TP      : 0x3fc80314  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x00000001
S0/FP   : 0x00000003  S1      : 0x00000001  A0      : 0x3fc92118  A1      : 0x3c0aab28
A2      : 0x00000003  A3      : 0x3fc92145  A4      : 0x00000001  A5      : 0x3fc96000
A6      : 0x40140000  A7      : 0x00000000  S2      : 0x00001881  S3      : 0x00000001
S4      : 0x00000000  S5      : 0x3c0aa314  S6      : 0x00000000  S7      : 0x00000017
S8      : 0x3fcdf928  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000001  T6      : 0x3ff00000
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000
Stack memory:
3fc92100: 0x00000000 0x00000000 0x00000000 0x40389936 0x00000000 0x00000000 0x452a2a2a 0x524f5252
3fc92120: 0x202a2a2a 0x74732041 0x206b6361 0x7265766f 0x776f6c66 0x206e6920 0x6b736174 0x68747020
3fc92140: 0x64616572 0x73616820 0x65656220 0x6564206e 0x74636574 0x002e6465 0x00000000 0x00000000
3fc92160: 0x00000000 0x3c0aaaec 0x3fc9e734 0x3c0aab14 0x00000000 0x80000003 0x00000000 0x4038803a
3fc92180: 0x00000000 0x80000003 0x3fca107c 0x4038911e 0x403801ba 0x00000000 0x00000000 0x4038019e
3fc921a0: 0x00000001 0x3fc921a8 0xffffffff 0x3fc9a314 0x3fc9a314 0x00000001 0x3fc998b8 0xffffffff
3fc921c0: 0x3fc998b8 0x3fc998b8 0x00000000 0x3fc921d0 0xffffffff 0x3fc921d0 0x3fc921d0 0x00000000
3fc921e0: 0x3fc921e4 0xffffffff 0x3fc921e4 0x3fc921e4 0x00000000 0x3fc921f8 0xffffffff 0x3fc921f8
3fc92200: 0x3fc921f8 0x00000001 0x3fc9e704 0xffffffff 0x3fc9e704 0x3fc9e704 0x00000000 0x3fc92220
3fc92220: 0xffffffff 0x3fc92220 0x3fc92220 0x00000000 0x3fc92234 0xffffffff 0x3fc92234 0x3fc92234
3fc92240: 0x00000000 0x3fc92248 0xffffffff 0x3fc92248 0x3fc92248 0x00000000 0x3fc9225c 0xffffffff
3fc92260: 0x3fc9225c 0x3fc9225c 0x00000000 0x3fc92270 0xffffffff 0x3fc92270 0x3fc92270 0x00000000
3fc92280: 0x3fc92284 0xffffffff 0x3fc92284 0x3fc92284 0x00000000 0x3fc92298 0xffffffff 0x3fc92298
3fc922a0: 0x3fc92298 0x00000000 0x3fc922ac 0xffffffff 0x3fc922ac 0x3fc922ac 0x00000000 0x3fc922c0
3fc922c0: 0xffffffff 0x3fc922c0 0x3fc922c0 0x00000000 0x3fc922d4 0xffffffff 0x3fc922d4 0x3fc922d4
3fc922e0: 0x00000000 0x3fc922e8 0xffffffff 0x3fc922e8 0x3fc922e8 0x00000000 0x3fc922fc 0xffffffff
3fc92300: 0x3fc922fc 0x3fc922fc 0x00000000 0x3fc92310 0xffffffff 0x3fc92310 0x3fc92310 0x00000000
3fc92320: 0x3fc92324 0xffffffff 0x3fc92324 0x3fc92324 0x00000000 0x3fc92338 0xffffffff 0x3fc92338
3fc92340: 0x3fc92338 0x00000000 0x3fc9234c 0xffffffff 0x3fc9234c 0x3fc9234c 0x00000000 0x3fc92360
3fc92360: 0xffffffff 0x3fc92360 0x3fc92360 0x00000001 0x3fc92374 0xffffffff 0x3fca1080 0x3fca1080
3fc92380: 0x00000000 0x3fc92388 0xffffffff 0x3fc92388 0x3fc92388 0x00000001 0x3fc9239c 0xffffffff
3fc923a0: 0x3fc9c158 0x3fc9c158 0x00000000 0x3fc923b0 0xffffffff 0x3fc923b0 0x3fc923b0 0x00000000
3fc923c0: 0x3fc923c4 0xffffffff 0x3fc923c4 0x3fc923c4 0x00000003 0x3fc923d8 0xffffffff 0x3fc979c4
3fc923e0: 0x3fc9d1a8 0x00000000 0x3fc923ec 0xffffffff 0x3fc923ec 0x3fc923ec 0x3c0b2610 0x00000005
3fc92400: 0x3c0a9eb4 0x0000000d 0x3c0a9aac 0x00000015 0x3c0b0f38 0x0000001d 0x3c0b47c0 0x00000025
3fc92420: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc92440: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc92460: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc92480: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc924a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc924c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc924e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
ELF file SHA256: 0000000000000000
```

ivmarkov, 2022-01-02:
Oh - I thought this is an invalid instruction or suchlike. Stack overflows are not real errors, but a normal situation, as the esp32 has a very tiny stack size.

To fix it, just create your thread with 7000 or 10000 bytes of stack using the Rust `thread::Builder` API. There is an example of that in the demo.

BTW, there is also a way to set larger default stack size for the main task, as well as for all explicitly created threads from rust using `sdkco fig.defaults`.

ivmarkov, 2022-01-02:
And by the way you should put a `thread::sleep` **inside** the code of your explicitly created thread, or else the thread will exit once the connection is established, thus dropping the wifi instance which will result in wifi shutdown. Which kind of makes the whole wifi initialization in a separate thread unnecessary. That is, unless you move back the initialized wifi I stance back to the main thread once it's initialization is complete.

closed 2022-01-02 by ivmarkov

ivmarkov, 2022-01-02:
And you are actually not "connecting" the wifi at all of course.

BrendanBall, 2022-01-02:
Ye I thought after I changed the version of idf that it started to produce invalid instruction, but then I couldn't reproduce that again so I must've forgotten to reflash it or something. 

Thanks for the advice.