esp-rs/esp-idf-svc #39: HTTP headers wrongly treated as case-sensitive by client

AlyoshaVasilieva, 2022-01-19:
EspHttpResponse defines headers as `BTreeMap<String, String>`: https://github.com/esp-rs/esp-idf-svc/blob/a39c1199b1468d6b581d457cdc752bdf5a38ed1d/src/http/client.rs#L314-L317

If a server sends `Content-Type: something`, then `resp.content_type()` returns `None` because [embedded_svc asks for `content-type`](https://github.com/esp-rs/embedded-svc/blob/47c21851fb47141629057ceb2f834c4139eb666e/src/http.rs#L59-L61) and `EspHttpResponse` does a case-sensitive search: https://github.com/esp-rs/esp-idf-svc/blob/a39c1199b1468d6b581d457cdc752bdf5a38ed1d/src/http/client.rs#L337-L339

I *think* the fix is replacing the `String` key with something like [`Uncased`](https://docs.rs/uncased/0.9.6/uncased/struct.Uncased.html)

ivmarkov, 2022-01-20:
`Uncased` seems to cover our use case perfectly. Are you up for a PR?

AlyoshaVasilieva, 2022-01-20:
I'll try to work on it within the next day or two

sehraf, 2022-01-20:
What is the advantage of `uncased` over something simple like `to_lowercase()`?

AlyoshaVasilieva, 2022-01-20:
Uncased doesn't require any additional allocations or modification of strings

PR #40 (2022-01-21): HTTP header keys are case-insensitive
Use the `uncased` crate when HTTP client is enabled to ensure this.

Fixes #39 in my brief test of my project, tested on an ESP32-C3.
closed 2022-01-21 by ivmarkov