esp-rs/esp-idf-svc #42: mqtt client panics

sh0rez, 2022-01-22:
Trying to get the experimental MQTT client to work, but having no luck.

The following code leads to the panic below, however no actually useful error message describing the panic is printed. Decoding the backtrace leads into the mqtt client internals.

```rust
fn main() -> Result<()> {
    // Temporary. Will disappear once ESP-IDF 4.4 is released, but for now it is necessary to call this function once,
    // or else some patches to the runtime implemented by esp-idf-sys might not link properly.
    esp_idf_sys::link_patches();

    // setup logging
    esp_idf_svc::log::EspLogger::initialize_default();

    let netif_stack = Arc::new(EspNetifStack::new()?);
    let sys_loop_stack = Arc::new(EspSysLoopStack::new()?);
    let default_nvs = Arc::new(EspDefaultNvs::new()?);
    let wifi = wifi(netif_stack, sys_loop_stack, default_nvs);

    let mqtt_cfg = MqttConfig {
        client_id: Some("foobar"),

        ..Default::default()
    };

    EspMqttClient::new("mqtt://broker:1883", &mqtt_cfg);
    return Ok(());
}
```

Running leads to the following panic:

```
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr 
Core  1 register dump:
PC      : 0x401760c3  PS      : 0x00060436  A0      : 0x800dbf91  A1      : 0x3ffcc590  
A2      : 0x3f40325c  A3      : 0x0000002b  A4      : 0x3ffcc5b0  A5      : 0x3f40324c  
A6      : 0x3f40332c  A7      : 0x00060b23  A8      : 0x8008c7c8  A9      : 0x3ffcc570  
A10     : 0x00000003  A11     : 0x00060423  A12     : 0x00060420  A13     : 0x3ffc8014  
A14     : 0x3ffc9afc  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Backtrace:0x401760c0:0x3ffcc590 0x400dbf8e:0x3ffcc5b0 0x400dc028:0x3ffcc5e0 0x400d5fe0:0x3ffcc610 0x400dbfa2:0x3ffcc630 0x4017a87a:0x3ffcc650 0x4017a2e9:0x3ffcc680 0x400ef733:0x3ffcc6c0 0x400ef751:0x3ffcc6e0 0x400f021d:0x3ffcc700 0x4008c539:0x3ffcc730
ELF file SHA256: 0000000000000000
Rebooting...
```

Decoding using `addr2line` gives:

```
~/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/panicking.rs:87
~/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/result.rs:1298
~/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.36.3/src/mqtt/client.rs:371
~/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.36.3/src/mqtt/client.rs:101
~/.rustup/toolchains/esp/lib/rustlib/src/rust/library/alloc/src/boxed.rs:1705
~/work/diox-rs/.embuild/platformio/packages/framework-espidf/components/esp_event/esp_event.c:145
~/work/diox-rs/.embuild/platformio/packages/framework-espidf/components/esp_event/esp_event.c:582 (discriminator 3)
~/work/diox-rs/.embuild/platformio/packages/framework-espidf/components/mqtt/esp-mqtt/mqtt_client.c:926
~/work/diox-rs/.embuild/platformio/packages/framework-espidf/components/mqtt/esp-mqtt/mqtt_client.c:913
~/work/diox-rs/.embuild/platformio/packages/framework-espidf/components/mqtt/esp-mqtt/mqtt_client.c:1371
~/work/diox-rs/.embuild/platformio/packages/framework-espidf/components/freertos/port/xtensa/port.c:168
```

jan-br, 2022-01-22:
Same here, cant get it to work either.
An example would be really nice.

ivmarkov, 2022-01-30:
Working on this. Probably will have something later today.

ivmarkov, 2022-01-30:
Should work now. Demo snippet: https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/src/main.rs#L710

closed 2022-01-30 by ivmarkov