esp-rs/esp-idf-svc #43: EspTimer panics as soon as it elapses (tries to call the callback)

martinserts, 2022-01-23:
I would like to use a countdown timer. `Callback` in current implementation is mandatory, although my use case does not need it. I need it to implement timeout of a blocking operation (`nb::Error::WouldBlock` and `timer.is_scheduled()`).

In any case, I cannot get it to work - the timer panics as soon as it times out and tries to call the callback.

```
Started.                                                                                                                                                                [59/1275]
Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.                                                                                          
Core  0 register dump:                                                                                                                                                           
MEPC    : 0x4200037a  RA      : 0x420061f8  SP      : 0x3fc8fbc0  GP      : 0x3fc8ba00           
TP      : 0x3fc861d0  T0      : 0x00000000  T1      : 0x00000000  T2      : 0x00000000           
S0/FP   : 0x3fc91a34  S1      : 0x4200036c  A0      : 0x00000000  A1      : 0x00000000           
A2      : 0x0023d166  A3      : 0x60023044  A4      : 0x3fc8e000  A5      : 0x600c2000           
A6      : 0x00000000  A7      : 0x00000000  S2      : 0x00000000  S3      : 0x00000000           
S4      : 0x0023d16f  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000           
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000           
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000           
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000           
MHARTID : 0x00000000   
Stack memory:                                                                                                                                                                    
3fc8fbc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x420062b8
3fc8fbe0: 0x00000000 0x00000000 0x00000000 0x40386a20 0x00000000 0x00000000 0x00000000 0x00000000
3fc8fc00: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x27146414 0x01010107 0x00000001 0x00000000
3fc8fc20: 0xffffffff 0x7fefffff 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000 0x3fe00000
3fc8fc40: 0x00000000 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413
3fc8fc60: 0x00000000 0x3ff00000 0x00000000 0x40240000 0x00000000 0x401c0000 0x00000000 0x40140000
3fc8fc80: 0x00000000 0x43500000 0x3fc8b200 0x00001c74 0xa5a5a5a5 0x00000154 0x3fc8fb40 0x3fc8dea4
3fc8fca0: 0x3fc8d670 0x3fc8d670 0x3fc8fc98 0x3fc8d668 0x00000003 0x130f9a0c 0x43369cfe 0x3fc8fc98
3fc8fcc0: 0x00000000 0x00000016 0x3fc8ec94 0x5f707365 0x656d6974 0xd4210072 0x00f4ebf0 0x00000000
3fc8fce0: 0x3fc8fc90 0x00000016 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc8e6bc 0x3fc8e724
3fc8fd00: 0x3fc8e78c 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x42009f5c
3fc8fd20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc8fd40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc8fd60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc8fd80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc8fda0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc8fdc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc8fde0: 0x00000000 0x00000000 0x3fc80000 0x0000000c 0x3fc8fe00 0x00000000 0x3fc8fde8 0x0000000c
3fc8fe00: 0x09c4000a 0x00000000 0x3fc8fe20 0x0000000c 0x3fc930c0 0x00000000 0x3fc8fe08 0x0000000c
3fc8fe20: 0x0c86000a 0x00000000 0x3fc9309c 0x00001d58 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fe40: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fe60: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fe80: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fea0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fec0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fee0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8ff00: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8ff20: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8ff40: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8ff60: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8ff80: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8ffa0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
ELF file SHA256: 0000000000000000
Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382250
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x403ce000,len:0x928
load:0x403d0000,len:0x2ce0
entry 0x403ce000
```

The code:
```
use embedded_hal_0_2::prelude::*;
use embedded_svc::timer::{Timer, TimerConfiguration, TimerService};
use esp_idf_hal::delay;
use esp_idf_svc::timer::EspTimerService;
use esp_idf_sys::EspError;
use std::time::Duration;

fn timer_elapsed() -> Result<(), EspError> {
    Ok(())
}

fn run_timer() -> Result<(), EspError> {
    let timer_service = EspTimerService::new()?;
    let timer_config = TimerConfiguration {
        name: None,
        skip_unhandled_events: true,
    };
    let mut timer = timer_service.timer(&timer_config, &timer_elapsed)?;
    timer.once(Duration::from_millis(2000))?;

    let mut delay = delay::FreeRtos;
    delay.delay_ms(3000_u32);

    Ok(())
}

fn main() {
    esp_idf_sys::link_patches();
    println!("Started.");
    println!("FINISHED: {:?}", run_timer());
}
```

What am I doing wrong?

The library calls `esp_timer_create` with locally (on stack) allocated reference `arg: &unsafe_callback as *const _ as *mut _,`, which gets garbage collected.
https://github.com/esp-rs/esp-idf-svc/blob/master/src/timer.rs#L104-L111

ivmarkov, 2022-01-23:
This is very much work in progress, and I probably should not have exposed it even under the "experimental" flag. I'll let you know once it is read and tested (and close the bug therefore).

ivmarkov, 2022-01-29:
Should work now, see the updated [demo](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/src/main.rs#L438).

closed 2022-01-29 by ivmarkov