esp-rs/esp-idf-svc #47: Yielded async tasks never restarted

anichno, 2022-02-05:
When trying to use smol for async tasks, tasks which try to yield time with `smol::Timer` never get reactivated.

Repo with a minimal example: https://github.com/anichno/esp32-smol-issue

Expected Behavior: 
```
main task started
main task print
main task print
main task print
main task print
...
```

Actual Behavior:
```
main task started
main task print
```

ivmarkov, 2022-02-07:
Indeed. There seems to be a bug in the `notify`  logic of the poller. Working on it...

ivmarkov, 2022-02-07:
OK, so the polling code seems to be correct!
The culprit turned out to be [something completely different](https://github.com/esp-rs/rust/issues/106).

In short:
* Until the above issue is fixed, this code will NOT work on xtensa multicore - (as in the stock `esp32` chip)
* Your example does work on single-core Riscv (`esp32c3`) - I tried that, and probably also on single-core Xtensa (`esp32s3`). If you have one of these, just rebuild for them and flash

I'm closing this not because the problem is solved, but because the problem is elsewhere.
Hopefully, we'll be able to fix the other (atomics) problem quickly. Personally, I also need working `smol` for the multicore Xtensa esp32 chip. And... atomics not working is very severe anyway.

Small detail: FYI - I'll update the patched `smol` dependencies maintained in the `esp-rs-compat` org to their latest versions in the coming hours.

closed 2022-02-07 by ivmarkov