esp-rs/esp-idf-svc #50: Can't import chrono

Codeleiche, 2022-02-16:
Hello there, when I try to add chrono to my project I get the following error message:
`
Compiling time v0.1.43
error[E0609]: no field `tm_gmtoff` on type `tm`
   --> /home/some_pc/.cargo/registry/src/[github.com](http://github.com/)-1ecc6299db9ec823/time-0.1.43/src/[sys.rs:314](http://sys.rs:314/)
:30
    |
314 |             let gmtoff = [out.tm](http://out.tm/)_gmtoff;
    |                              ^^^^^^^^^ unknown field
    |
    = note: available fields are: `tm_sec`, `tm_min`, `tm_hour`, `tm_mday`, `tm_mon` ... and 4 others

For more information about this error, try `rustc --explain E0609`.
error: could not compile `time` due to previous error
warning: build failed, waiting for other jobs to finish...
error: build failed
`
any idea on how to solve that? 
Thanks in advance,
David

MabezDev, 2022-02-16:
That is quite an old version of `time`, the latest is now `0.3.7` which compiles fine on espidf platform.

Seems there are some maintenance issues with chrono, see: https://github.com/meilisearch/milli/pull/450 for a start. Maybe you can do the same as meilisearch and just use the time crate directly?

Codeleiche, 2022-02-17:
I'll check that, It's an dependency of another crate that ue NaiveDateTime, maybe I can convert it

Codeleiche, 2022-02-17:
ok, I tried to convert the given NaiveDateTime to a time::PrimitiveDateTime with time v0.3.7 but can't import tie macros via 
```
use time::{PrimitiveDateTime, Date, Time, macros::{date, time}};
```
am I missing something? I get
```
error[E0432]: unresolved import `time::macros`
  --> src/main.rs:47:43
   |
47 | use time::{PrimitiveDateTime, Date, Time, macros::{date, time}};// macros::{date, datetime, time}};
   |                                           ^^^^^^ could not find `macros` in `time`
```

closed 2022-02-18 by Codeleiche