esp-rs/esp-idf-svc #56: esp_netif_obj isn't thread safe

ganeshrvel, 2022-03-05:
I was trying to access the `wifi` object within a thread but it is throwing this error:
```
202 |     thread::Builder::new().stack_size(32768).spawn(move || {
    |                                              ^^^^^ `*mut esp_netif_obj` cannot be sent between threads safely
    |
    = help: within `EspWifi`, the trait `Send` is not implemented for `*mut esp_netif_obj`
    = note: required because it appears within the type `EspNetif`
    = note: required because it appears within the type `std::option::Option<EspNetif>`
    = note: required because it appears within the type `EspWifi`
    = note: required because of the requirements on the impl of `Sync` for `std::sync::Mutex<EspWifi>`
    = note: required because of the requirements on the impl of `Send` for `Arc<std::sync::Mutex<EspWifi>>`
    = note: required because it appears within the type `[closure@src/main.rs:202:52: 204:6]`
note: required by a bound in `Builder::spawn`
   --> /Users/username/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/thread/mod.rs:379:12
    |
379 |         F: Send + 'static,
    |            ^^^^ required by this bound in `Builder::spawn`

```

This happens after I wrapped wifi object inside a mutex and arc.
```
    let wifi = EspWifi::new(netif_stack, sys_loop_stack, default_nvs).expect("Need wifi");
    let wifi_arc = Arc::new(Mutex::new(wifi));

    thread::Builder::new().stack_size(32768).spawn(move || {
        let status = wifi_arc.clone().lock().unwrap().get_status();
    });

```

ganeshrvel, 2022-03-05:
This issue was fixed in `esp-idf-svc: 0.37.2`. Closing this ticket.

closed 2022-03-05 by ganeshrvel

thomas725, 2023-02-12:
I'm using esp-idf-svc version = "0.45.0" - and get the same error, has it been reintroduced or did I do something wrong?
```
   Compiling playground-2023-01-15 v0.1.0 (/home/thomas/Development/esp-rs/playground-2023-01-15_esp-idf_v4.4)
error[E0277]: `*mut esp_netif_obj` cannot be shared between threads safely
   --> src/main.rs:174:37
    |
174 |       let timer = timer_service.timer(move || {
    |  _______________________________-----_^
    | |                               |
    | |                               required by a bound introduced by this call
175 | |         
176 | |         let do_reconnect = if let Ok(wifi_ro) = wifi.read() {
177 | |             if !wifi_ro.is_connected().unwrap_or(false) {
...   |
193 | |         }
194 | |     })?;
    | |_____^ `*mut esp_netif_obj` cannot be shared between threads safely
    |
    = help: within `EspWifi<'static>`, the trait `Sync` is not implemented for `*mut esp_netif_obj`
    = note: required because it appears within the type `EspNetif`
    = note: required because it appears within the type `EspWifi<'static>`
    = note: required for `RwLock<EspWifi<'static>>` to implement `Sync`
    = note: required for `std::sync::Arc<RwLock<EspWifi<'static>>>` to implement `Send`
note: required because it's used within this closure
   --> src/main.rs:174:37
    |
174 |     let timer = timer_service.timer(move || {
    |                                     ^^^^^^^
note: required by a bound in `EspTimerService::<T>::timer`
   --> /home/thomas/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.45.0/src/timer.rs:173:50
    |
173 |     pub fn timer(&self, callback: impl FnMut() + Send + 'static) -> Result<EspTimer, EspError> {
    |                                                  ^^^^ required by this bound in `EspTimerService::<T>::timer`

For more information about this error, try `rustc --explain E0277`.
error: could not compile `playground-2023-01-15` due to previous error
```

cs-clarence, 2023-12-17:
getting this error as well

```
    --> src/lib.rs:176:6
     |
176  |     .layer(layers);
     |      ^^^^^ `*mut esp_idf_sys::esp_netif_obj` cannot be shared between threads safely

```

ivmarkov, 2023-12-17:
> getting this error as well for version `0.47.3`:
> 
> ```
>     --> src/lib.rs:176:6
>      |
> 176  |     .layer(layers);
>      |      ^^^^^ `*mut esp_idf_sys::esp_netif_obj` cannot be shared between threads safely
> ```

Well... the the [ESP IDF Netif API documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html) claims that the Netif APIs are thread-safe, but if you look at the actual implementation, I think it is anything but thread-safe.

@igrr Is this indeed a bug in the ESP IDF documentation? While Netif state-changing functions [are delegated to the LwIP thread](https://github.com/espressif/esp-idf/blob/master/components/esp_netif/lwip/esp_netif_lwip.c#L1622), getters [do read directly without LwIP thread delegation or any locking](https://github.com/espressif/esp-idf/blob/master/components/esp_netif/lwip/esp_netif_lwip.c#L1624). So I think the Netif code has data races, if you try to use it from multiple threads, i.e. it is not `Sync` in Rust speak, only `Send` (as in you can call it from any thread, but not concurrently from multiple threads).

ivmarkov, 2023-12-17:
@cs-clarence if you really need a `Sync` behavior, put the `EspNetif` instance inside a`std::sync::Mutex` for now.

cs-clarence, 2023-12-18:
@ivmarkov I already tried wrapping it with an `Arc<Mutex>` but it still has errors

```
`*mut esp_idf_svc::sys::esp_netif_obj` cannot be shared between threads safely
within `esp_idf_svc::wifi::AsyncWifi<esp_idf_svc::wifi::EspWifi<'_>>`, the trait `std::marker::Sync` is not implemented for `*mut esp_idf_svc::sys::esp_netif_obj`
required for `&esp_idf_svc::wifi::AsyncWifi<esp_idf_svc::wifi::EspWifi<'_>>` to implement `std::marker::Send`
```

ivmarkov, 2023-12-18:
Just use an async mutex to wrap all of `AsyncWifi`.

cs-clarence, 2023-12-29:
It is already async (`tokio::sync::Mutex`). Anyway, I circumvented around the issue by using `tokio::task::spawn_local` which will accept a `Future` that is not `Send`. I just have to spawn my top level future within a `LocalSet`. Note that this will ensure that all futures will only be executed on the thread where `LocalSet` is created, which is probably fine.

ivmarkov, 2023-12-29:
> It is already async (`tokio::sync::Mutex`).

If it **IS** wrapped in `tokio::sync::Mutex` then [it IS `Sync` already](https://docs.rs/tokio/latest/tokio/sync/struct.Mutex.html#impl-Sync-for-Mutex%3CT%3E), so I'm not really sure what your problem is, from then on. I think you might be wrapping it in a wrong way and not even realizing it. The error mesage says:
```
[&esp_idf_svc::wifi::AsyncWifi<esp_idf_svc::wifi::EspWifi<'_>](note: required for `&esp_idf_svc::wifi::AsyncWifi<esp_idf_svc::wifi::EspWifi<'_>>` to implement `std::marker::Send`)
```

What are you wrapping in `tokio::sync::Mutex`? `AsyncWifi<EspWifi<>>` or `&AsyncWifi<EspWifi<>>`? If the latter, that cannot work, as `&AsyncWifi<EspWifi<>>` is NOT `Send`, and therefore, `tokio::sync::Mutex<&AsyncWifi<EspWifi<>>>` is NOT `Sync`.

> Anyway, I circumvented around the issue by using `tokio::task::spawn_local` which will accept a `Future` that is not `Send`. I just have to spawn my top level future within a `LocalSet`. Note that [...]

Yeah. This is like saying that you actually don't need `Sync` as you'll only do a single thread. Completely fine, and BTW then there are alternatives to tokio. As in `async_executor::LocalExecutor` and `edge_executor::LocalExecutor`.

cs-clarence, 2023-12-29:
These are my typedefs:

```
pub type ArcAsyncMutex<T> = std::sync::Arc<tokio::sync::Mutex<T>>;
pub type Wifi = AsyncWifi<EspWifi<'static>>;
pub type ShareableWifi = ArcAsyncMutex<Wifi>;
```

The `ShareableWifi` is what I am using for the rest of my code when passing the the Wifi around it is sure to be `Send` + `Sync`. Passing them around is not really my problem, it is when it comes to using them.

For example, this code will not compile (note that wifi is `ShareableWifi`): 
```
if !wifi.lock().await.is_started()? { // this is fine because even though `is_started()` takes `&self` or `&AsyncWifi`, it isn't an async call so no issues with Send and Sync.
  wifi.lock().await.start().await?; // <-- the compiler points the problem here because of the async call
}
```
Note that `start` takes in `&mut self`,  `Self` is `AsyncWifi` in this case which is not `Sync` because `&mut AsyncWifi` is not `Send`. So even the the passed wifi is wrapped with `Arc<Mutex<>>`, the `lock()` call returns a `MutexGuard` which is `Send` in this case, but when you actually use it, the `MutexGuard` needs to be `deref coerced` into a `&mut AsyncWifi` which causes the error:

```
[&esp_idf_svc::wifi::AsyncWifi<esp_idf_svc::wifi::EspWifi<'_>](note: required for `&esp_idf_svc::wifi::AsyncWifi<esp_idf_svc::wifi::EspWifi<'_>>` to implement `std::marker::Send`)
```

ivmarkov, 2023-12-30:
TL;DR: 
* This has nothing to do with `esp_netif_obj_t` not being `Sync` and again: it should not and cannot be `Sync`.
* The crux was indeed using `&AsyncWifi<EspWifi<>>` instead of simply `AsyncWifi<EspWifi<>>` except that this was happening in the `esp-idf-svc` code, not in your code. :)
* I've pushed changes to [embedded-svc here](https://github.com/esp-rs/embedded-svc/commit/26731acf99430dd25914208aae81120c01dd5500) and to [esp-idf-svc here](https://github.com/esp-rs/esp-idf-svc/commit/3ebaf6337dc00d8a4fa9a4f02640508c22a0faac). If you use these from GIT, your problem should be solved (and  `esp_netif_obj_t` is still not `Sync`!)

Longer story:

> These are my typedefs:
> 
> ```rust
> pub type ArcAsyncMutex<T> = std::sync::Arc<tokio::sync::Mutex<T>>;
> pub type Wifi = AsyncWifi<EspWifi<'static>>;
> pub type ShareableWifi = ArcAsyncMutex<Wifi>;
> ```
> 
> The `ShareableWifi` is what I am using for the rest of my code when passing the the Wifi around so it is sure to be `Send` + `Sync`. Passing them around is not really my problem, it is when it comes to using them.
> 
> For example, this code will not compile (note that wifi is `ShareableWifi`):
> 
> ```rust
> if !wifi.lock().await.is_started()? { // this is fine because 
>                                                           // even though `is_started()` 
>                                                           // takes `&self` or `&AsyncWifi`, 
>                                                           // it isn't an async call 
>                                                           // so no issues with Send and Sync.
> 
>   wifi.lock().await.start().await?; // <-- the compiler points the problem here 
>                                                        // because of the async call
> }
> ```

Yes, I agree with all of the above. That was the issue for me too, and yeah, it should work even if `esp_netif_obj_t` is **not** `Sync`.

> Note that `start` takes in `&mut self`, `Self` is `AsyncWifi` in this case which is not `Sync` because `&mut AsyncWifi` is not `Send`. 
> So even though the passed wifi is wrapped with `Arc<Mutex<>>`, the `lock()` call returns a `MutexGuard` which is `Send` in this case, but when you actually use it, the `MutexGuard` needs to be `deref coerced` into a `&mut AsyncWifi` in the `start()` call which causes the error:

I don't think this is correct. I think you might be having a too literal image of how futures are desugared...

> Anyway, it all comes back to `*mut esp_idf_sys::esp_netif_obj` not being thread safe which also affects its dependents.

Fortunately not completely. Or else how would [this code](https://github.com/ivmarkov/issue56) compile now?

cs-clarence, 2023-12-30:
Tried the new commits and can confirm that it works. Thanks!