esp-rs/esp-idf-svc #61: Wifi Issues (ESP32)

Traijan1, 2022-04-03:
Hello, 
I'm hoping I'm in the right place to ask this.

I started with Rust on the ESP32 today and wanted to do some basic Web Server and Wifi for testing purpose.
The issue is, that my ESP32 disconnects from Wifi after it started. I get an IP Address but after that it completly shutdowns from Wifi.

My code is: 
```rust
fn wifi_init() -> String {
    let mut wifi = EspWifi::new(
        Arc::new(EspNetifStack::new().unwrap()),
        Arc::new(EspSysLoopStack::new().unwrap()),
        Arc::new(EspDefaultNvs::new().unwrap())
    )
    .expect("Failed to initialize WiFi");

    wifi.set_configuration(&Configuration::Client(ClientConfiguration {
        ssid: env!("WIFI_SSID").into(),
        password: env!("WIFI_PASSWORD").into(),
        ..Default::default()
    })).unwrap();

    wifi.wait_status_with_timeout(Duration::from_secs(10), |s| {
        !s.is_transitional()
    }).expect("Could not connect to WiFi");

    let status = wifi.get_status();

    let ip = match status.0 {
        embedded_svc::wifi::ClientStatus::Started(ClientConnectionStatus::Connected(ClientIpStatus::Done(settings))) => format!("{}", settings.ip),
        _ => return Err(anyhow!("Not in Client Mode")).unwrap()
    };

    ip
}
```

and my main:

```rust
fn main() {
    esp_idf_sys::link_patches();
    let ip = wifi_init();
    println!("IP is {}", ip);
}
```

The output is following:

```
Opening COM3 with speed 115200
Resetting device... done
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
load:0x4007c428,len:4840
entry 0x4007c6a0
I (418) cpu_start: Pro cpu up.
I (418) cpu_start: Starting app cpu, entry point is 0x40081254
I (0) cpu_start: App cpu up.
I (432) cpu_start: Pro cpu start user code
I (432) cpu_start: cpu freq: 160000000
I (432) cpu_start: Application information:
I (437) cpu_start: Project name:     libespidf
I (442) cpu_start: App version:      1
I (446) cpu_start: Compile time:     Apr  3 2022 14:47:57
I (452) cpu_start: ELF file SHA256:  0000000000000000...
I (458) cpu_start: ESP-IDF:          c29343e
I (463) heap_init: Initializing. RAM available for dynamic allocation:
I (470) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (476) heap_init: At 3FFB7408 len 00028BF8 (162 KiB): DRAM
I (483) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (489) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (495) heap_init: At 400933B0 len 0000CC50 (51 KiB): IRAM
I (503) spi_flash: detected chip: generic
I (506) spi_flash: flash io: dio
I (511) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (602) wifi:wifi driver task: 3ffc0b88, prio:23, stack:6656, core=0
I (602) system_api: Base MAC address is not set
I (602) system_api: read default base MAC address from EFUSE
I (602) wifi:wifi firmware version: 63017e0
I (612) wifi:wifi certification version: v7.0
I (612) wifi:config NVS flash: disabled
I (612) wifi:config nano formating: disabled
I (622) wifi:Init data frame dynamic rx buffer num: 32
I (622) wifi:Init management frame dynamic rx buffer num: 32
I (632) wifi:Init management short buffer num: 32
I (632) wifi:Init dynamic tx buffer num: 32
I (642) wifi:Init static rx buffer size: 1600
I (642) wifi:Init static rx buffer num: 10
I (642) wifi:Init dynamic rx buffer num: 32
I (652) wifi_init: rx ba win: 6
I (652) wifi_init: tcpip mbox: 32
I (662) wifi_init: udp mbox: 6
I (662) wifi_init: tcp mbox: 6
I (4022) esp_netif_handlers: sta ip: 192.168.178.32, mask: 255.255.255.0, gw: 192.168.178.1
I (4022) wifi:state: run -> init (0)
I (4022) wifi:pm stop, total sleep time: 452671 us / 761315 us
W (4022) wifi:<ba-del>idx
I (4032) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (4042) wifi:flush txq
I (4042) wifi:stop sw txq
I (4042) wifi:lmac stop hw txq
W (4042) wifi:hmac tx: ifx0 stop, discard
I (4052) wifi:Deinit lldesc rx mblock:10
IP is 192.168.178.32
```

After research it sounds like it is a memory issue? Could this be? Because if yes, how could I make it better to work. I would really like to use this for some projects.

I appreciate any help!

ivmarkov, 2022-04-03:
What do you expect? Right after connecting, your main function ends and thus the wifi variable is dropped, so wifi disconnects and overall the execution of your app ends?

closed 2022-04-03 by ivmarkov

Traijan1, 2022-04-03:
Sorry I didn't realized that my wifi object is dying after the function

ivmarkov, 2022-04-03:
Np :) You have to keep the wifi object around and then also do something useful in your main() right after wifi is initialized.