esp-rs/esp-idf-svc #62: posting to eventloop from interrupt broken

brianmay, 2022-04-10:
Somehow this change:

https://github.com/esp-rs/esp-idf-svc/commit/b5d8295fe02e8fd98755ad40bdea0fae4d957eb0

Broke posting from an interrupt. The post request returns `EspError(258)`.

This does not appear to be related to the new panic call. Rather it is because we call `isr_post_raw` instead of `post_raw`.

The following change fixes it:

```diff
diff --git a/src/eventloop.rs b/src/eventloop.rs
index 2be74f80..b10ce0f4 100644
--- a/src/eventloop.rs
+++ b/src/eventloop.rs
@@ -603,7 +603,7 @@ where
     fn post(&mut self, payload: &P, wait: Option<Duration>) -> Result<bool, Self::Error> {
         if interrupt::active() {
             #[cfg(esp_idf_esp_event_post_from_isr)]
-            let result = P::serialize(payload, |raw_event| self.isr_post_raw(raw_event));
+            let result = P::serialize(payload, |raw_event| self.post_raw(raw_event, wait));
 
             #[cfg(not(esp_idf_esp_event_post_from_isr))]
             let result = {
```

ivmarkov, 2022-04-10:
Hm. I just quickly re-tested it in https://github.com/ivmarkov/ruwm/commit/e4c7be202a04ac247cec427091bb1c9325a7fd0f
"Works on my machine", so to say.

Are you doing `cargo clean; cargo build` after setting `CONFIG_ESP_EVENT_POST_FROM_ISR=y` in `sdkconfig.defaults`? You should.

Background:
As per the ESP-IDF documentation, calling `post_raw` (which calls ESP-IDF's `esp_event_post_to`) from an ISR handler should *not* be done. Instead, we should be calling `isr_post_raw` (which calls ESP-IDF's `esp_event_isr_post_to`).

`esp_event_post_to` might misbehave when called from an ISR. 
* It calls a bunch of APIs which I'm not sure are safe to be called from an ISR (`calloc`, `xSemaphoreTakeRecursive`, `xTaskGetCurrentTaskHandle` and `xQueueSendToBack`), with `calloc` and `xQueueSendToBack` (with ticks count) being particularly troublesome, as they would try to use mutexes and wait in an ISR context, which is not good at all
* When called from an ISR, `esp_event_post_to` will also **not** yield / re-prioritize the awoken even loop task if it has higher priority than the task that was running before the ISR was called

So we have to do whatever possible to make the ISR-dedicated APIs work.

With that said, I'm suspecting you and @pyaillet might be (ab)using the event loop API for things where there are better (lighter-weight and more correct) alternatives.

Are you using an event loop to post that a GPIO "button" was pressed? If so, then you should really make sure that you are using a dedicated event loop per button (which is wasteful! that's one thread per button), or else your logic is flawed. You see, event loop is a broadcast queue, but it also has a limited size. If you are multiplexing "button pressed" events from multiple buttons on it - and if your queue size is small - and since you cannot wait when posting from an ISR - you might miss a button press as the queue is full when you are posting it. Extremely small chance but logically possible.

Where I'm getting at, is that [this thing](https://github.com/esp-rs/esp-idf-svc/blob/master/src/notify.rs) might be more appropriate in these scenarios. It is a bit lighter weight compared to FreeRTOS events' queue, and also it does not exhibit the "queue is full" behavior, because it supports "bit per button" (up to 32 bits): if your button was pressed before and the thread has not processed that yet, you'll set a bit which was already 1 to 1, thus "coalescing" multiple events into one, but that's not a big deal anyway, as long as the coalesce logic is doing it "per button", and since each button has a dedicated bit, that works. Unlike the event queue, which might be full because of the events of *other* buttons.

brianmay, 2022-04-10:
If it wasn't picking up on the `CONFIG_ESP_EVENT_POST_FROM_ISR`, then I think the above change I did that fixed it wouldn't have any affect.

I have no idea why the `isr_post_raw` is not working. There is no overflow of the queue going on, this happens for the first event.

Your point on maybe the EventLoop is the wrong tool for the job might be valid. It was what I was advised to use on the chat forum a while back. I will have a look at the `EspNotify` approach when I have time.

brianmay, 2022-04-10:
Would really help heaps if you tagged releases in git...

If this `EspNotify` a new thing? I can't find it in `esp-idf-sys` 0.31.0 which I believe is the latest release.

brianmay, 2022-04-10:
Is there an example anywhere for `EspNotify`? Am doing what I think should be correct, but the subscribe isn't getting called. So wonder if I am doing something wrong.

pyaillet, 2022-04-10:
I finally took the time to refactor my project to use EspBackgroundNotify (sorry for the long delay, I was working on something else).

I have an example working [here](https://github.com/pyaillet/twatch-idf-rs/pull/8).

It might be a bit overly complex right now, as I was aiming at making it work.

Thanks for your remarks @ivmarkov and for the report @brianmay.

brianmay, 2022-04-10:
Silly me, was letting the subscription expire. Seems to be OK now. At least for early preliminary tests.

pyaillet, 2022-04-10:
For the record I faced the same problem with the error `EspError(258)` when posting to the `EspBackgroundEventLoop` from the `ISR` despite the option `CONFIG_ESP_EVENT_POST_FROM_ISR` being set.
Could this be a problem with the EventType Serialization ?

Anyway, using the `EspBackgroundEventLoop` I faced a strange `stackoverflow` which I don't seem to get anymore.

brianmay, 2022-04-10:
Yes, `EspNotify` seems to be working fine for me.

Next step try to get the new interrupt subscription stuff working. My first attempt had difficulties, but not related to this issue, so won't discuss here.

ivmarkov, 2022-04-10:
> For the record I faced the same problem with the error `EspError(258)` when posting to the `EspBackgroundEventLoop` from the `ISR` despite the option `CONFIG_ESP_EVENT_POST_FROM_ISR` being set. [...]

`EspError(258)` is `ESP_ERR_INVALID_ARG`

You are likely hitting this line of code: https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_event/esp_event.c#L917

When posting from an ISR, the data you are posting should be 4 bytes big max, as the ISR poster cannot call `calloc`. How big is the data you are posting, i.e. `mem::size_of::<WhateverRustTypeYouSerialize>()`?

pyaillet, 2022-04-10:
Ok, on my side, this was clearly the problem as I was not aware of this limitation and tried to post something larger.

Now, using `EspBackgroundNotify` as you suggested is working fine and seems the best way to go, so thanks for your advice :)

ivmarkov, 2022-04-10:
If @brianmay confirms that he is also trying to post a payload which when serialized (or its `size_of` if you use `Copy` "serialization") is > 4 bytes, I'll close that.

brianmay, 2022-04-10:
In my case:

```
#[derive(Copy, Clone, Debug)]
struct EventLoopMessage(i32, Value);
```

Yes, looks like this is more then 4 bytes,..,  Ok, closing.

closed 2022-04-10 by brianmay