esp-rs/esp-idf-svc #63: EspWifi::wait_status_with_timeout() seems not resetting after deep sleep.

mgaggero, 2022-04-10:
Hi, I'm working with a code for ESP32-C3 (RISC-V core) that uses a wifi client connection to send some data to a HTTP server and then goes to deep sleep for 10 seconds. I'm using _wait_status_with_timeout()_ to wait for the connection to be operating or timeout after 40 seconds, and _drop()_ to stop the wifi and shut down the connection gracefully before the call to _esp_deep_sleep()_.

Powering or resetting the device, it actually works for 3 deep sleep cycles after which _wait_status_with_timeout()_ immediately returns claiming a timeout that is not occurred (few ms are elapsed after the call to the function instead of 40 seconds).

It seems to me that _wait_status_with_timeout()_ timer is initialized at the first call after the device reset but that is not stopped during the sleep and  cleared after the wake up, still keeping to run (the timeout occurs exactly after more than 40 second after the device powering or reset).
There's some errors in my code or it is a bug?

This is an excerpt of the code that runs on **ESP32-C3**:

```
//#[allow(unused_imports)]
use std::time::Duration;

use embedded_svc::http::{client::*, Status};
use embedded_svc::wifi;
use embedded_svc::wifi::{Wifi, ApStatus, ClientStatus, ClientConnectionStatus, ClientIpStatus};
use embedded_svc::wifi::{ClientConfiguration, Configuration};

use esp_idf_svc::wifi::EspWifi;
use esp_idf_svc::netif::*;
use esp_idf_svc::sysloop::*;
use esp_idf_svc::nvs::*;
use esp_idf_svc::http::client::*;

fn main() -> Result<(), anyhow::Error> {
    esp_idf_svc::log::EspLogger::initialize_default();

    let netif_stack = Arc::new(EspNetifStack::new()?);
    let sys_loop_stack = Arc::new(EspSysLoopStack::new()?);
    let default_nvs = Arc::new(EspDefaultNvs::new()?);

    let mut wifi = EspWifi::new(
        netif_stack,
        sys_loop_stack,
        default_nvs).unwrap();

    wifi.set_configuration(&Configuration::Client(
        ClientConfiguration {
            ssid: "wifinet".into(),
            password: "wifipass".to_string(),
            ..Default::default()
        }
    ))?;    
            
    wifi.wait_status_with_timeout(Duration::from_secs(40), |status| !status.is_transitional())
        .map_err(|e| error!("Unexpected Wifi status: {:?}", e));
    
    let status = wifi.get_status();
    
    if let wifi::Status(
        ClientStatus::Started(ClientConnectionStatus::Connected(ClientIpStatus::Done(_ip_settings))),
        ApStatus::Stopped
    ) = wifi.get_status() {
        println!("Wifi is connected.");
    } else {
        error!("Unexpected Wifi status: {:?}", status);
    } 
        
    let http_url = format!("http://192.168.1.100:8080/");
    let http_line = format!("Hello, World!");
    println!("sending data {} to {}", http_line, http_url);

    let mut client = EspHttpClient::new_default()?;

    let request = client.post(&http_url)?;

    match request.send_str(http_line) {
        Ok(response) =>
            match response.status() {
                100..200 => error!("Unexpected HTTP response code: {}", response.status()),
                200..400 => debug!("Message sent: {}", response.status()),
                400..600 => error!("HTTP error code {}", response.status()),
                _ => println!("HTTP Response: {}", response.status()),
            }
        Err(_e) => { error!("HTTP Error"); }
    }

    println!("Turning off Wifi.");
    drop(wifi);
    println!("Going to sleep.");
    unsafe {
        esp_idf_sys::esp_deep_sleep(1000_0000);
    }

    #[allow(unreachable_code)]
    Ok(())
}
```

Massimo.

P.S.: in the code above, I found that I must set the deep sleep for 1000_000 usec in order to get a 10 seconds sleep time. Why?

TrueBrain, 2022-07-16:
Ran into this exact same issue. Turns out, 2 days ago someone actually solved the problem: https://github.com/esp-rs/esp-idf-hal/pull/100.

Add in `Cargo.toml`:
```
[patch.crates-io]
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal", branch = "master" }
```

Fixes the problem now. And I guess otherwise you have to wait for a release of `esp-idf-hal` :)

MabezDev, 2022-07-22:
This should now be fixed in relese v0.42.1 :)

closed 2022-07-22 by MabezDev

arrisde, 2022-07-24:
> This should now be fixed in release v0.42.1 :)

Had the same issue on ESP32 and got it resolved with @TrueBrain's suggestion, while esp-idf-svc v0.42.1 did not fix it for me...

MabezDev, 2022-07-24:
> Had the same issue on ESP32 and got it resolved with @TrueBrain's suggestion, while esp-idf-svc v0.42.1 did not fix it for me...

Oops! Sorry got mixed up on which crate I submitted the fix :sweat_smile:.