esp-rs/esp-idf-svc #67: undefined reference to `_binary_x509_crt_bundle_start'

svenstaro, 2022-04-24:
I'm trying to make use of the bundled certificates as per #37. This works fine as far as I can tell but the problem is that we don't have the [magic `asm()` from esp-idf](https://github.com/espressif/esp-idf/blob/45c1d1cba212b2012d53a55fcf9d338a959c2ece/components/mbedtls/esp_crt_bundle/esp_crt_bundle.c#L21) exposed anywhere in esp-idf-svc and so linking to these inline asm binary positions fails. I don't currently see a good way to fix this.

svenstaro, 2022-04-24:
So this fails because they are using [target_add_binary_data](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#embedding-binary-data) which embeds the data [like so](https://github.com/espressif/esp-idf/blob/master/tools/cmake/scripts/data_file_embed_asm.cmake). I'm not sure there's a good way to imitate this.

ivmarkov, 2022-04-24:
How about defining a bunch of `extern "C"` vars in Rust code? That have the same names as `${varname}*`? Would that not work?

svenstaro, 2022-04-24:
I did this:
```rust
extern "C" {
    static _binary_x509_crt_bundle_start: *const u8;
    static _binary_x509_crt_bundle_end: *const u8;
}

fn main() -> Result<()> {
}
```
and no dice:
```
  = note: Running ldproxy
          Error: Linker .embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
         .embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/mbedtls/libmbedtls.a(esp_crt_bundle.c.obj):(.literal.esp_crt_bundle_attach+0x0): undefined reference to `_binary_x509_crt_bundle_start'
          collect2: error: ld returned 1 exit status
```

svenstaro, 2022-04-24:
Hey @sergiogasquez sorry for the random ping but I couldn't help but notice that you did [essentially the same thing](https://github.com/SergioGasquez/ferrous-training/blob/e354d660c1b7b54163118cb42724e4ad3db2ba0d/intro/http-client/solution/src/main.rs#L40) just a few days ago. Would you be able to weigh in how you solved this particular issue when calling
```rust
    let mut client = EspHttpClient::new(&EspHttpClientConfiguration {
        use_global_ca_store: true,
        crt_bundle_attach: Some(esp_crt_bundle_attach),
        ..Default::default()
    })?;
```
?
Any hints would be super greatly appreciated!

svenstaro, 2022-04-25:
Whoops that's weird! GitHub search only showed yours and not the original one.

closed 2022-04-25 by ivmarkov