esp-rs/esp-idf-svc #68: set_hostname isnt working.

ganeshrvel, 2022-04-27:
`set_hostname` doesn't seem to be working. The connected **ESP32**'s hostname is still coming as `espressif`.

code snippet:
```
        let netif_stack = Arc::new(EspNetifStack::new()?);

        let netif_stack_clone = Arc::<esp_idf_svc::netif::EspNetifStack>::clone(&netif_stack);
        let interface_stack_config = InterfaceStack::Sta.get_default_configuration();
        let esp_netif = EspNetif::new(netif_stack_clone, &interface_stack_config)?; 
        esp_netif.set_hostname("my-device-host-name")?;

        let sys_loop_stack = Arc::new(EspSysLoopStack::new()?);
        let default_nvs = Arc::new(EspDefaultNvs::new()?);

        let esp_wifi = EspWifi::new(netif_stack, sys_loop_stack, default_nvs)?;
```

The output of `esp_netif.set_hostname("my-device-host-name")` is coming as `OK`

ivmarkov, 2022-04-27:
It doesn't work because `EspWifi` has its own internal `EspNetif` instance (actually, two, if AP is also enabled), and is not related at all to the `EspNetif` instance you are explicitly creating in the code snippet.

To set the host name (this is only possible for STA, and when DHCP is enabled btw), just use [this](https://github.com/esp-rs/embedded-svc/blob/master/src/ipv4.rs#L145).

closed 2022-04-27 by ivmarkov

ganeshrvel, 2022-04-27:
Taking cues from the above comment I got it working:

```
    let dhcp_conf = DHCPClientSettings {
        hostname: Some("my-custom-hostname"),
    };

    let ip_conf = ipv4::ClientConfiguration::DHCP(dhcp_conf);
    let cl = ClientConfiguration {
        ssid: "<ssid>",
        password: "<password>",
        channel: "<channel>",
        bssid: None,
        auth_method: AuthMethod::default(),
        ip_conf: Some(ip_conf),
    };

    let netif_stack = Arc::new(EspNetifStack::new()?);
    let sys_loop_stack = Arc::new(EspSysLoopStack::new()?);
    let default_nvs = Arc::new(EspDefaultNvs::new()?);

    let esp_wifi = EspWifi::new(netif_stack, sys_loop_stack, default_nvs)?;
    esp_wifi.set_configuration(&Configuration::Client(cl)).unwrap();
```