esp-rs/esp-idf-svc #73: Feature request - Set client wifi config without restarting

usbalbin, 2022-05-05:
Hi!

First of, Thank you all for the awesome set of esp related crates :)

---

I am trying to setup a device for the following scenario.

1. Device is not configured, start in `AP`-mode (likely `Mixed` to allow scanning)
2. The user connects to the AP using a phone or similar and in a web interface hosted by the device configure what Wifi the device should be connected to (need ability to scan)
3. The device tries to connect to the selected network (while maintaining connection to the phone through the AP)
4. If step 3 succeeded, turn of AP. Otherwise maintain connection with phone and allow for another try

First off, is this a scenario that should be supported by esp-idf-svc/embedded-svc?

Looking at the code I have stumbled upon some questions:
* `Wifi::scan()` restarts the wifi? Does it *have* to do that? or could something like `fn scan_with_optional_restart(restart: bool)` be exposed?
* `Wifi::set_configuration(wifi::Configuration::Mixed())` can not start without a client config (can not scan in Ap-mode)
  Calling EspWifi::set_configuration(wifi::Configuration::Mixed(Default::default(), ap_config)) fails with something about an invalid SSID

I tried these changes and they seem to make make it possible to use `EspWifi::set_configuration(wifi::Configuration::Mixed(Default::default(), ap_config))` to enter `Mixed` (and probably `Client`) without specifying client settings:

```diff
    fn set_client_conf(&mut self, conf: &ClientConfiguration) -> Result<(), EspError> {
         info!("Setting STA configuration: {:?}", conf);

-        let mut wifi_config = wifi_config_t {
-            sta: Newtype::<wifi_sta_config_t>::from(conf).0,
+        let mut wifi_config = if conf.ssid.is_empty() {
+            // Empty config
+            unsafe { core::mem::zeroed() }
+        } else {
+            wifi_config_t {
+                sta: Newtype::<wifi_sta_config_t>::from(conf).0,
+            }
         };

         esp!(unsafe { esp_wifi_set_config(wifi_interface_t_WIFI_IF_STA, &mut wifi_config) })?;

-        self.set_client_ip_conf(&conf.ip_conf)?;
+        if !conf.ssid.is_empty() {
+            self.set_client_ip_conf(&conf.ip_conf)?;
+        }

         info!("STA configuration done");
```

Also, would there be any way to expose some way to change client settings without affecting the Ap connection? Perhaps something like(pseudo code)

```rust
pub fn foo(&mut self, conf: &ClientConfiguration) -> Result<(), ErrorNotInMixedOrClientMode | EspError> {
    if not in Mixed or Client mode {
        return Err(ErrorNotInMixedOrClientMode);
    }
    wifi.set_client_conf(conf);
    if conf.ssid.is_empty() {
        status.client_status = Stopped;
    } else {
        status.client_status = Started;
    }
}
```

Please let me know if you would like me to split this into multiple issues and if you would be interested in a PR

ivmarkov, 2022-05-07:
I don't think step 3 is possible. Mixed mode (AP + STA) does require that both AP and STA are running on the same Wifi channel, because the esp32 has a single radio. However, when setting up the AP, you cannot know upfront what Wifi parameters the user would configure for the STA mode, so that you can configure the AP on the same channel.

usbalbin, 2022-05-09:
Oh, that does make sense. Thank you for the clarification :) . 

So I would have to do something more like(changed text in bold):
1. Device is not configured, start in AP-mode (likely Mixed to allow scanning)
2. The user connects to the AP using a phone or similar and in a web interface hosted by the device configure what Wifi the device should be connected to (need ability to scan)
3. The device tries to connect to the selected network **(by `wifi.set_configuration(Mixed(...))`, restarting the wifi closing the connection to the browser)**
4. **During this time the browser polls the esp for a status of how the connection went(and the phone will most likely reconnect to the ap after restart)**
5. If step 3 succeeded, turn off AP. Otherwise maintain connection with phone and allow for another try

However it would still be possible to perform a scan without restarting the Wifi right? It does at least *seem* to work in my tests with #76 combined with the above proposed `if conf.ssid.is_empty(){ empty config }`..

Again, thank you

ivmarkov, 2022-12-15:
Should now be fixed.

closed 2022-12-15 by ivmarkov