esp-rs/esp-idf-svc #81: Can't disable color logging

ZacJW, 2022-05-24:
I'm not sure if this should go here or in esp-idf-sys. I've been trying to disable the generation of terminal color control sequences when logging, and so in sdkconfig I have `# CONFIG_LOG_COLORS is not set` rather than `CONFIG_LOG_COLORS=y`. Doing this causes esp-idf-sys to generate without a `const CONFIG_LOG_COLORS` and I get a compiler error since esp-idf-svc expects it to be defined but equal to 0 in the case of disabling color logs.

```text
error[E0425]: cannot find value `CONFIG_LOG_COLORS` in this scope
   --> /home/zac/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.41.3/src/log.rs:107:12
    |
107 |         if CONFIG_LOG_COLORS == 0 {
    |            ^^^^^^^^^^^^^^^^^ not found in this scope

For more information about this error, try `rustc --explain E0425`.
error: could not compile `esp-idf-svc` due to previous error
```

I've tried this with both rust-esp32-std-demo and also a platformio based build and it's the same error in both. Has this branch ever executed? If so what am I doing wrong?

N3xed, 2022-05-25:
I'd guess you have to set it to `CONFIG_LOG_COLORS=n`.

ivmarkov, 2022-05-25:
> I'd guess you have to set it to `CONFIG_LOG_COLORS=n`.

I would keep it as a bug and will fix it shortly. There is a better way to express this by using
 `#[cfg(esp_idf_log_colors)]`.

ZacJW, 2022-05-25:
> I'd guess you have to set it to `CONFIG_LOG_COLORS=n`.

I did try that but it doesn't work. In a platformio based build it rewrites it back to `# CONFIG_LOG_COLORS is not set` and in a cargo based build it leaves sdkconfig as is, but still generates esp-idf-sys without the constant.

closed 2022-08-13 by ivmarkov