esp-rs/esp-idf-svc #83: Feature request: implement esp_wifi_set_country

leelists, 2022-05-28:
Hi!

First of, Thank you all for the awesome set of esp related crates :)

I'm porting a little c app from esp-idf to rust and search a simple way to call 

```
    const wifi_country_t cc = {
            .cc = "fr",
            .schan = 1,
            .nchan = 14,
            .max_tx_power = 255,
    };
    ESP_ERROR_CHECK(esp_wifi_set_country(&cc));
```

from rust, but i'm a bit stuck

Any clues are welcomed

MabezDev, 2022-05-30:
We don't have safe wrappers to do this yet, but in general you can always call esp-idf C functions with the esp_idf_sys crate.

E.g, a rough pseudo-ish code example:
```rust
use esp_idf_sys::*;

let cc = wifi_country_t {
  cc: "fr",
  schan: 1,
  nchan: 14,
  max_tx_power: 255,
  ..Default::default()
};

let res = esp!(unsafe {
  esp_wifi_set_country(&cc)
});
```

If this works for you, and you feel like contributing back, adding a safe wrapper around this to esp-idf-svc would be a helpful contribution :).

leelists, 2022-06-02:
Many thanks, I will see if i'am able to do that