esp-rs/esp-idf-svc #84: Crypto support

arlyon, 2022-06-01:
Hi!

I have been messing around w/ the esp and would like to add a hal impl for the hardware hmac signing on the esp32 c3 (and more). I assume that would happen in `esp-idf-svc`, but I'm opening this issue to make sure I'm targeting the right place.

As a side note, really enjoying working w/ this ecosystem! :tada:

ivmarkov, 2022-06-02:
I think you should rather do this in `esp-idf-hal`, as the hardware HMAC / DS is a "device" (and as a matter of fact, `EspWifi` and `EspEth*` should also be moved to `esp-idf-hal` as they also rely on hardware devices (the modem in the case of `EspWifi` and the SPI/EMAC controllers in the case of `EspEth`, but it is a gray area for those as they are mixing the ethernet layer with the IP layer in a single trait)).

Also, `esp-idf-hal`/`esp-idf-svc` is all about wrapping ESP-IDF drivers & APIs, so you probably need to wrap [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/hmac.html) and [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/ds.html) API in your impl. If you are looking to implement a bare-metal driver, it probably belongs to the `esp-hal` crate, not to `esp-idf-hal`.

What I would suggest is to first of all make sure that the above ESP-IDF APIs are exposed by `esp-idf-sys`. If they are not, you need to contribute a PR to `esp-idf-sys` which includes the relevant ESP-IDF headers [here](https://github.com/esp-rs/esp-idf-sys/blob/master/src/include/esp-idf/bindings.h).

Once you have the unsafe bindings to the HMAC/DS ESP-IDF APIs in `esp-idf-sys`, you can write your own code that calls these APIs to e.g. sign with the HMAC key. If this works, the next step indeed is to indeed write a typesafe "driver" in `esp-idf-hal` that actually wraps the unsafe ESP-IDF driver exposed in `esp-idf-sys` in the previous step.

PRs greatly appreciated!

arlyon, 2022-06-02:
I appreciate your direction. Will see if I can woth a solution together for simple upstream hmac, though I don't have an ETA until I explore further.

Cheers