esp-rs/esp-idf-svc #89: EspWifi - Panic on wifi drop

Nukeuler123, 2022-07-06:
Hello, i'm working with the Esp32_DevkitC_V4 board and iv been testing out allowing users to connect to the esp32 through a self created AP and having them interact with an HTTP server on the board to submit information about their home WiFi so the board can connect to their WiFi on reboot. Everything works well until the WiFi variable is dropped, where then the main thread panics.

My code

```
pub fn wifi_setup(nvs: Arc<EspDefaultNvs>) -> Result<(String, String), Box<dyn Error>> {
    let netif_stack: Arc<EspNetifStack> = Arc::new(EspNetifStack::new()?);
    let sys_loop_stack: Arc<EspSysLoopStack> = Arc::new(EspSysLoopStack::new()?);

    info!("Starting wifi");
    let mut wifi: Box<EspWifi> = Box::new(EspWifi::new(netif_stack, sys_loop_stack, nvs)?);

    wifi.set_configuration(&Configuration::AccessPoint(AccessPointConfiguration {
        ssid: "sensor_setup".into(),
        password: "Testing123".into(),
        max_connections: 3,
        auth_method: embedded_svc::wifi::AuthMethod::WPA2Personal,
        ip_conf: Some(RouterConfiguration {
            subnet: Subnet {
                gateway: Ipv4Addr::new(192, 168, 1, 1),
                mask: Mask(24),
            },
            dhcp_enabled: true,
            dns: Some(Ipv4Addr::new(192, 168, 1, 1)),
            secondary_dns: None,
        }),
        ..Default::default()
    }))?;

    info!("Starting setup server");
    let mut server = esp_idf_svc::http::server::EspHttpServer::new(&Default::default())?;
    let ssid: Arc<Mutex<String>> = Arc::new(Mutex::new(String::new()));
    let passwd: Arc<Mutex<String>> = Arc::new(Mutex::new(String::new()));

    let tmp_ssid = ssid.clone();
    let tmp_pass = passwd.clone();
    server
        .handle_get("/", |_req, resp| {
            resp.send_str(
                r#"
                <form method="post" action="setup">
                    <label for="ssid">WifiName:</label><br>
                    <input type="text" id="ssid" name="ssid"><br>
                    <label for="passwd">Password:</label><br>
                    <input type="password" id="passwd" name="passwd">
                    <input type="submit" value="Submit">
                </form> "#,
            )?;
            Ok(())
        })?
        .handle_post("/setup", move |mut req, resp| {
            //Process the form into the ssid and password
            let mut body = Vec::new();

            ToStd::new(req.reader()).read_to_end(&mut body)?;

            let ssid = url::form_urlencoded::parse(&body)
                .filter(|p| p.0 == "ssid")
                .map(|p| p.1.to_string())
                .next()
                .unwrap();
            let pass = url::form_urlencoded::parse(&body)
                .filter(|p| p.0 == "passwd")
                .map(|p| p.1.to_string())
                .next()
                .unwrap();

            *tmp_ssid.lock().unwrap() = ssid;
            *tmp_pass.lock().unwrap() = pass;

            resp.send_str(
                "Wifi setup completed, please reconnect to your home wifi to continue installation",
            )?;

            Ok(())
        })?;

    //Wait for the user to enter in their wifi's SSID and Pass, very janky but it gets the job done. 
    info!("waiting for configured AP info");
    while ssid.lock().unwrap().len() == 0 || passwd.lock().unwrap().len() == 0 {
        std::thread::sleep(Duration::from_millis(250));
    }

    let true_ssid = ssid.lock().unwrap().to_string();
    let true_pass = passwd.lock().unwrap().to_string();

    info!("Got AP info, killing wifi and server");
    drop(server);
    std::thread::sleep(Duration::from_millis(250));
    drop(wifi); //<Crash here
    Ok((true_ssid, true_pass))
}

```

The error (and things happening before that)
```
I (61590) iot_temp_firmware::wifi_setup: Got AP info, killing wifi and server
I (61590) esp_idf_svc::http::server: Unregistered Httpd server handler 3 for URI "/setup"
I (61590) esp_idf_svc::http::server: Unregistered Httpd server handler 1 for URI "/"
I (61700) esp_idf_svc::http::server: Httpd server stopped
I (61950) esp_idf_svc::wifi: Stopping
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: EspError(-1)', /home/esp/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.42.0/src/wifi.rs:792:30
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
abort() was called at PC 0x400e1102 on core 0
Backtrace:0x40081caa:0x3ffba8500x40088a3d:0x3ffba870 0x4008edaa:0x3ffba890 0x400e1102:0x3ffba900 0x400e10f6:0x3ffba920 0x400fa7f6:0x3ffba940 0x400fb320:0x3ffba960 0x400faf51:0x3ffbaa00 0x400faf00:0x3ffbaa30 0x400fb679:0x3ffbaa50 0x400d6708:0x3ffbaa80 0x400d8717:0x3ffbaac0 0x400d4475:0x3ffbab20 0x400d5cb2:0x3ffbab70 0x4016337b:0x3ffbae80 0x400d5e17:0x3ffbaea0 0x400e04b9:0x3ffbaf30 0x4016706b:0x3ffbaf50 
ELF file SHA256: 0000000000000000
Rebooting...
```

It should be noted that before wifi was enabled the object `esp_idf_svc::nvs_storage::EspNvsStorage` was created so the flash memory could be checked to see if the wifi credentials were already stored on the system. After the check was done the object was dropped.

Thank you for your time.

TrueBrain, 2022-08-02:
Ran into the same issue, and made a somewhat smaller example and did some triage:

```rust
fn test() {
    let netif_stack = Arc::new(EspNetifStack::new()?);
    let sys_loop_stack = Arc::new(EspSysLoopStack::new()?);
    let default_nvs = Arc::new(EspDefaultNvs::new()?);

    let wifi = EspWifi::new(netif_stack, sys_loop_stack, default_nvs)?;

    wifi.set_configuration(&Configuration::AccessPoint(
        AccessPointConfiguration {
            ssid: "test".into(),
            ..Default::default()
        },
    ))?;
}
```

After calling this function, `wifi` drops (and calls `drop()`, which calls `clear_all()`, which calls `stop()`, which returns an error on `esp_wifi_disconnect()`. In `drop()`, the `clear_all()` is unwrapped, causing a crash.

Documentation suggests `esp_wifi_disconnect()` should return `ESP_ERR_WIFI_NOT_STARTED`, and that is handled here:
https://github.com/esp-rs/esp-idf-svc/blob/cc6588d59ae67d634b3babbf09ca1b7a92209cba/src/wifi.rs#L574

But instead, the error-code is `-1`.

I made https://github.com/esp-rs/esp-idf-svc/compare/master...TrueBrain:esp-idf-svc:wifi-disconnect-without-connect?expand=1 to bypass the problem, but honestly, I think that is just a hack, and not a proper solution. Hopefully others have some clues in why `esp_wifi_disconnect()` is not returning `ESP_ERR_WIFI_NOT_STARTED`.

MabezDev, 2022-08-05:
Thanks for investigating the issue!

I think the plan would be to change the `Drop` impl to not panic if it fails to disconnect, a "best effort" approach to shutting everything down. This probably means duplicating most of `clear_all` and ignoring the errors (maybe print a warning?).

@TrueBrain Would you mind PR'ing something like that?

TrueBrain, 2022-08-05:
I totally agree that having `drop` doing some best effort would be better, but it actually doesn't fully solve this issue. The problem also happens when switching from AP to STA.

<details>
  <summary>Take the following example</summary>

```rust
use log::error;
use std::sync::Arc;

use esp_idf_sys::EspError;
use esp_idf_svc::netif::EspNetifStack;
use esp_idf_svc::nvs::EspDefaultNvs;
use esp_idf_svc::sysloop::EspSysLoopStack;
use esp_idf_svc::wifi::EspWifi;

use embedded_svc::wifi::{AccessPointConfiguration, ClientConfiguration, Configuration, Wifi};

fn main() -> Result<(), EspError> {
    esp_idf_svc::log::EspLogger::initialize_default();

    let netif_stack = Arc::new(EspNetifStack::new()?);
    let sys_loop_stack = Arc::new(EspSysLoopStack::new()?);
    let default_nvs = Arc::new(EspDefaultNvs::new()?);

    let mut wifi = EspWifi::new(netif_stack, sys_loop_stack, default_nvs)?;

    wifi.set_configuration(&Configuration::AccessPoint(
        AccessPointConfiguration {
            ssid: "test".into(),
            ..Default::default()
        },
    ))?;

    match wifi.set_configuration(&Configuration::Client(
        ClientConfiguration {
            ssid: "myap".into(),
            password: "password".into(),
            ..Default::default()
        },
    )) {
        Ok(_) => {}
        Err(e) => {
            error!("Couldn't change to STA: {:?}", e);
            return Err(e);
        }
    }

    Ok(())
}
```
</details>

This too triggers the `Couldn't change to STA` because of the same issue. So sadly enough, your suggestion would side-step the problem for now.

My earlier question, why `ESP_ERR_WIFI_NOT_STARTED` wasn't working, is answered by this comment in the IDF codebase:
https://github.com/espressif/esp-idf/blob/4c3322ed6003b87422bd79fe9e67f1cb347c6e2f/components/esp_wifi/include/esp_wifi.h#L64-L65

Basically, it means something different then I expected. It doesn't mean there was no `esp_wifi_connect`, but that there wasn't a `esp_wifi_start`. And in this case the latter is done, so it makes sense `ESP_ERR_WIFI_NOT_STARTED` isn't returned. So it seems IDF has no explicit error-state when `esp_wifi_disconnect` was called without a `esp_wifi_connect`. Which brings me back to my earlier solution to this problem:
https://github.com/esp-rs/esp-idf-svc/compare/master...TrueBrain:esp-idf-svc:wifi-disconnect-without-connect?expand=1

An alternative solution would be to just always ignore any error `esp_wifi_disconnect` gives, no matter if it is called from `drop`, or from `do_scan`, or from `set_configuration`.

@MabezDev : what do you think is the best solution here?

MabezDev, 2022-08-05:
Hmm. In the case of `ESP_FAIL: other WiFi internal errors` we really need to be able to know what that error is. Perhaps the error is emitted as a WiFi event in the event loop? If we can capture that event after disconnect (or switching to STA) we may be able to handle it more appropriately than just ignoring it.

TrueBrain, 2022-08-05:
The other solution that comes to mind is to track if `esp_wifi_connect` was called. If not, skip the `esp_wifi_disconnect`. My above branch does something similar, but depends on the status, instead of "is connect called".

TrueBrain, 2022-08-05:
> Hmm. In the case of `ESP_FAIL: other WiFi internal errors` we really need to be able to know what that error is. Perhaps the error is emitted as a WiFi event in the event loop? If we can capture [...]

The only event I see from my example application, is `ApStarted`. No other events are emitted.

Honestly, from an IDF point-of-view, I guess this makes sense: why call disconnect when you didn't call connect? It is only the abstraction that this library gives that causes this issue, I guess. Don't get me wrong, I love this library, as it saves me writing a lot of boilerplate .. but I guess this is just a result of that.

Now I understand a bit more what is actually going on, I think my original approach is the easiest one: just check the STA state to know if `esp_wifi_disconnect` is needed.

Nukeuler123, 2022-08-06:
Thanks for looking into this yall, had a suspicion that it had to do with the unwrap in the drop call

PR #113 (2022-08-08): Fix: only call esp_wifi_disconnect() when esp_wifi_connect() was called
ESP-IDF seems to assume that you only call esp_wifi_disconnect
when you called esp_wifi_connect first. Otherwise it will
simply return -1, without explaining further why.

The ESP_ERR_WIFI_NOT_STARTED is only returned when esp_wifi_start()
wasn't called. But ClientStatus cannot be Connected if it hasn't.
As such, that check is no longer required.

Fixes #89, and see there for more discussion about this topic.
closed 2022-08-08 by MabezDev