esp-rs/esp-idf-svc #97: Corrupted HTTP Request due to usage of httpd_resp_set_status

taunusflieger, 2022-07-16:
The implementation of <code>EspHttpResponseHeaders::send</code> is corrupting the HTTP response. The implementation uses <code>httpd_resp_set_status()</code> which requires that the provided raw pointer is valid until the response is send. This is with the current implementation not the case. The lifetime of the dereferenced raw c++ pointer is to short (only valid within the <code>EspHttpResponseHeaders::send()</code> function, but no longer when the response gets transmitted.

I'm not aware that Rust allows lifetimes for raw pointer, so I assume the best solution would be to refactor the <code>EspHttpResponseHeaders::send()</code> function and move the code into the <code>EspHttpResponseWrite::write</code> function to ensure that the lifetime of the raw pointer can be managed as required by the ESP_IDF api.

https://github.com/esp-rs/esp-idf-svc/blob/1cad9a87d439d715681c323ddb8972a0bc70b6c5/src/http/server.rs#L601

ivmarkov, 2022-07-28:
I know we have discussed it and there is a large refactoring of the code coming... (where I'm spending most of my free time) yet I finally looked at the current code and the fix is one-liner, so I'll push it until the refactored code lands. 

Can you test if it works for you after that?

ivmarkov, 2022-07-28:
Fixed with this: https://github.com/esp-rs/esp-idf-svc/commit/cc6588d59ae67d634b3babbf09ca1b7a92209cba

closed 2022-08-06 by ivmarkov