esp-rs/esp-idf-svc #105: Question: How to obtain Mac Address of Wifi Adapter?

Snapstromegon, 2022-07-21:
This is probably an easy one:

I'm trying to write a custom WifiMgr which handles wifi reconnects and creating an Wifi AP if no WiFI is available.
To avoid SSID duplicates and keep them identifyable, I want to use the last 3 bytes of the mac as hex in the SSID.

**What I want to do:**

Get the default Mac Address of the Wifi Interface before starting a connection/AP.

**What I've tried:**

I've tried some different variations of this already:
```rust
let netif = EspNetif::new(
    Arc::new(EspNetifStack::new()?),
    &InterfaceConfiguration {
        interface_stack: InterfaceStack::Sta,
        ..Default::default()
    },
)?;
println!("Netif: {:?}", netif.get_mac());
```

**What I get:**

But this always prints the mac as only zeros.

**I also tried this:**

```rust
let mut wifi = EspWifi::new(
  Arc::new(EspNetifStack::new()?),
  Arc::new(EspSysLoopStack::new()?),
  Arc::new(EspDefaultNvs::new()?)
)?;

let status = wifi.get_status();
let wifi_config = wifi.get_configuration();
println!("Status: {status:?}, wifi config: {wifi_config:?}");
```

**Used Hardware:**

[M5StampC3](https://docs.m5stack.com/en/core/stamp_c3) - ESP32-C3 based board

I'm probably close, but the docs don't really help me out and I feel a little lost.

MabezDev, 2022-07-21:
I don't _think_ the netif mac is set by default? I think the mac comes from `esp_read_mac` (which can be set in many ways) and you'll need to call `netif_mac_set` to set it on the netif. More details here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#mac-address. I don't know if that's the intended behaviour though, it might be a bug.

Snapstromegon, 2022-07-21:
Thanks a lot @MabezDev! Your answer gave me the right hint.

To everyone just looking for the answer as of now:

```rust
use esp_idf_sys::{esp_mac_type_t_ESP_MAC_WIFI_STA, esp_read_mac};
let mut mac = [0 as u8; 6];
unsafe { esp_read_mac(mac.as_mut_ptr(), esp_mac_type_t_ESP_MAC_WIFI_STA); }
info!("Mac: {mac:02X?}");
```

This will give you the full base mac address from the ESP.

While the current behavior of the crate is not completely wrong, (at least according to my experience) it's unexpected. Especicially since the docs state:
> It is also possible to call the function [esp_netif_set_mac()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html#_CPPv417esp_netif_set_macP11esp_netif_tA_7uint8_t) to set the specific MAC used by a network interface after network initialization. But it is recommended to use the base MAC approach documented here to avoid the possibility of the original MAC address briefly appearing on the network before being changed.

As a developer I would've expected that `esp_netif_set_mac()` should only be called when I want to set a custom MAC and not when I want the default one.

Even more so in the abstracting svc crate I'd expect the behavior that way, since there is [EspNetif::get_mac](https://esp-rs.github.io/esp-idf-svc/esp_idf_svc/netif/struct.EspNetif.html#method.get_mac), but not `set_mac`.

After reading into this, I would expect the [esp_idf_svc::netif::InterfaceConfiguration](https://esp-rs.github.io/esp-idf-svc/esp_idf_svc/netif/struct.InterfaceConfiguration.html#) struct to contain an `mac: Option<[u8; 6]>` which allows to set a mac manually and alternatively sets the default mac based on the InterfaceStack.

This would mean a breaking change, but since I expect most people use `InterfaceConfiguration::default()` in some shape anyways, I don't expect such a change to impact many people anyways.

PR #107 (2022-07-23): #105 Add option to set a mac for an interface and use the default if none is set
This PR sets the default mac for any initialized mac (if it can be done reasonably) and allows to set a custom max during creation (or at a later point in time).

That way the `get_mac` method doesn't require an unsafe call to [esp_netif_set_mac](https://esp-rs.github.io/esp-idf-svc/esp_idf_sys/fn.esp_netif_set_mac.html) to initially set the mac address.

This is a breaking change, as it modifies the members of the InterfaceConfiguration struct.

closes #105
closed 2022-07-23 by ivmarkov