esp-rs/esp-idf-svc #106: Feature Request / RFC: Allow subscribing to wifi and ip event seperately

Snapstromegon, 2022-07-22:
At the moment the callback for [EspWifi::subscribe](https://esp-rs.github.io/esp-idf-svc/esp_idf_svc/wifi/struct.EspWifi.html#method.subscribe) does not pass the event into the callback, because the Subscription is to a combination of WifiEvent and IpEvent.

For some applications it would be useful to get the event instead of just an info that something happened.

To achieve this, I see two options, which are possible to implement together and **I'd be willing to create a PR for this**.

**Option 1: Two `subscribe_*` methods**

This options contains adding two new methods to the [esp_idf_svc::wifi::EspWifi](https://esp-rs.github.io/esp-idf-svc/esp_idf_svc/wifi/struct.EspWifi.html#) struct: 

1. `subscribe_wifi_events(&mut self, callback: impl for<'a> FnMut(&'a WifiEvent) + Send + 'static) -> Result<Self::Subscription, Self::Error>)`
2. `subscribe_ip_events(&mut self, callback: impl for<'a> FnMut(&'a IpEvent) + Send + 'static) -> Result<Self::Subscription, Self::Error>)`

This would allow users to clearly define which events they want to listen on and also provide a way to actually receive the triggering events.

This approach can be implemented without making a breaking change.

**Option 2: Changing the `subscribe` signature**

This option would change the current method signature in a way, that allows returning the events. This could be achieved like this:

1. Change the callback param to `(Option<&WifiEvent>, Option<&IpEvent>)` and fill the corresponding
2. Add a new type `enum EspWifiEvent { Wifi(&WifiEvent), Ip(&IpEvent) }` (simplified) and using that as the callback argument

This would be a breaking change to the API.

**Currently possible options:**

ATM you could of course subscribe to the corresponding event bus yourself and reimplement the logic inside of esp-idf-svc, but I think this is neither ergonomic, nor a good pattern.

ivmarkov, 2022-07-23:
> Currently possible options:

> ATM you could of course subscribe to the corresponding event bus yourself and reimplement the logic inside of esp-idf-svc, but I think this is neither ergonomic, nor a good pattern.

Have you looked [here](https://github.com/esp-rs/esp-idf-svc/blob/master/src/eventloop.rs#L34), and [here](https://github.com/esp-rs/esp-idf-svc/blob/master/src/eventloop.rs#L623)? This exists since *before* `EspWifi` implemented (a restricted functionality of) the EventBus trait as well, and is a public-facing API.

Just subscribe to the system event bus (note that `EspWifi` anyway takes an `Arc` ref to it) and listen to Wifi, or IP or Eth events or anything else you could deserialize with a custom `EspTypedEventDeserializer` impl of yours at will.

The above should be just as ergonomic as subscribing to the `EspWifi` instance itself. As for not being a good pattern... if you can elaborate?

Snapstromegon, 2022-07-23:
To be honest, at the time of writing I did not yet understand the code in the EspWifi subscribe method and thought more of it would be required boilerplate.

I also thought that the provided subscribe method was also there, so you could avoid touching the EventBus. I expected it to signal "something happened related to Wifi" instead of "something changed wifi", which is just a misunderstanding. This is also the main reason for #108 to exist.

Nonetheless I think passing through the event to the subscriber in some form would be helpful so the subscriber can further filter and know what actually happened and benefit from existing filtering of the EspWifi module like the prefiltered IpEvents.

When the user subscribes to the EventBus directly, they need to reimplement e.g. the `is_our_*_ip_event()` filters.

So to put it in a nutshell:

- The comment regarding pattern was mainly because of a misunderstanding of what the subscribe method should do and because a user has to reimplement filters already present in EspWifi
- I still think it would be helpful to pass the events to the subscriber in some form
- I personally think the filtering to events that only change the current state is unexpected and should at least be documented at some point (but I also know that documentation is still ongoing).

ivmarkov, 2022-07-23:
> Nonetheless I think passing through the event to the subscriber in some form would be helpful so the subscriber can further filter and know what actually happened and benefit from existing filtering [...]

Putting documentation aside (where you do have a point), how about doing the minimal set of changes:
- We can make the `is_*` stuff public
- We need an "async" WiFi scan, which I think is anyway what is driving your request. In fact **this** is the trickiest part and I strongly suggest we concentrate our thinking here instead
- The semantics of the `EspWifi`-specific event bus would remain to signify "changes visible in the status" as opposed to "any event" which the system event bus gives you anyway
- The `Status` struct needs to be extended with a `bool` or an `enum` which reports the scan status

Subscription (for now) would still be done on the system event loop for "any event" and subscribing to `EspWifi` would give you status changes only (but the status level of detail will grow over time).

ivmarkov, 2022-07-23:
Again, I would really like to stress that doing an async scan is the thing which requires serious thinking. An ergonomic async scan might lead to a "real" async `Wifi` trait, which then might mean a complete rethinking whether we need `EspWifi` to even implement the `EventBus` interface. None of this is applicable to the event loops of ESP-IDF, where `EventBus` serves great (IMO).

Snapstromegon, 2022-07-23:
I'm not trying to use scan in an async fashion. _ScanDone_ was just one of the EspWifi events that get filtered out and was the one that I found it on when doing my testing.

I'm writing kind of a wifi manager around EspWifi and for the structure of my code it would be helpful to get the wifi events somewhere other than in the code that calls `scan()` (even if it's just because I share the EspWifi instance). That would e.g. allow me to implement a simple metric counter like "was connected x% of the time", "scanned Y times for wifi networks", "z clients connected to AP" and all of this just with a handle to EspWifi.

I know that this can be achieved in other ways, but I just thought it'd be easier this way.

ivmarkov, 2022-07-23:
OK in that case if you only contribute public API (perhaps - on the `EspWifi`, not on the `Wifi` trait itself) that allows "recognizing" some events as ours that would be a good enough first step IMO.

I'm unwilling to do deeper redesigns of `Wifi` ATM, until after our async story is complete and until after `Wifi` works 100% OK for bare metal too. In other words, I want any potential changes to `Wifi` to also be informed by bare-metal and async, or else we might end up redesigning this stuff twice.

Snapstromegon, 2022-07-23:
If I see it correctly, `EspWifi` only allows one instance to exist. Therefore all `WifiEvent`s are relevent to that single `EspWifi` instance.

I will create a PR that adds one public fn to `EspWifi` to determine wether or not an `IpEvent` belongs to the `EspWifi` instance. This should then solve all my current problems and I agree that bare-metal and async are higher prio.

ivmarkov, 2022-07-23:
> If I see it correctly, `EspWifi` only allows one instance to exist. Therefore all `WifiEvent`s are relevent to that single `EspWifi` instance.
>
> I will create a PR that adds one public fn to [...]

Sorry for asking but can you do the same for `EspEth`? This is not a single instance though.

Snapstromegon, 2022-07-23:
Yes, I can try to do that too.

PR #109 (2022-08-02): Publish method for filtering ip_events from outside
Fixes #106 and allows third parties to test wether or not an IpEvent is related to an EspWifi instance.
This allows using the eventloop to get all wifi events and to filter for relevant events.
commit 9201f1c7 (2022-08-02): Publish method for filtering ip_events from outside

Fixes #106 and allows third parties to test wether or not an IpEvent is related to an EspWifi instance.
This allows using the eventloop to get all wifi events and to filter for relevant events.
closed 2022-08-02 by ivmarkov