esp-rs/esp-idf-svc #108: Bug: ScanDone Wifi events don't reach subscribers

Snapstromegon, 2022-07-22:
When subscribing to wifi events on EspWifi, some events that don't change the state of the connection are not passed to subscribers.

The example for me is the ScanDone event. This happens because of this check: https://github.com/esp-rs/esp-idf-svc/blob/master/src/wifi.rs#L1065

What is the reason why this check exists (and similar on the IpEvent subscription)?
I think all events should reach the subscribers who then can filter them themselves (the IpEvent is okay to filter on the station).

ivmarkov, 2022-07-23:
You have filed two issues (#108) and (#107) on the same topic. I'll close #108 - this one - and will continue commenting in #107.

To address your specific question though:
> When subscribing to wifi events on EspWifi, some events that don't change the state of the connection are not passed to subscribers.

> The example for me is the ScanDone event. This happens because of this check: https://github.com/esp-rs/esp-idf-svc/blob/master/src/wifi.rs#L1065

> What is the reason why this check exists (and similar on the IpEvent subscription)?
I think all events should reach the subscribers who then can filter them themselves (the IpEvent is okay to filter on the station).

Have you looked at what *event type* the `EventBus` adaptation of `EspWifi` exposes? It is `()`. In other words, *by design* you don't have any detail on the actual event which happened. This means, direct subscription to `EspWifi` (as opposed to subscribing to the system event bus - see my comment in #107) is only usable for indication if the *status* of the `EspWifi` has changed. Hence why this check you are referring to.

Regarding "ScanDone" - I'm not really sure how you plan to use it even if it was an option, given that `EspWifi` does not even expose currently a non-blocking `scan` method. If/when it does, the status field will probably get a new field - `scanning` or such like and the current event notification approach should continue to work as designed.

closed 2022-07-23 by ivmarkov

Snapstromegon, 2022-07-23:
Ahh, I see that the idea behind EspWifi's subscribe method is different from what I expected (I thought it was to simplify getting WifiEvents, so you don't have to touch the eventloop, but it's more to only get when the EspWifi actually has some internal change).