esp-rs/esp-idf-svc #110: Panic while scanning WiFi APs

jlocash, 2022-07-27:
Hi all, I'm fairly new to Rust on ESP, so apologies if I'm doing something wrong or this has been covered already. I'm starting a small project and I'm trying to connect to my WiFi network, however during the scan a panic occurs:
```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ()', /Users/josh/.cargo/registry/src/github.com-1ecc6299db9ec823/heapless-0.7.15/src/string.rs:295:25
```

My Machine: MacOS 12.5 (M1 MBP)
Toolchain version: esp-1.62.1.0
Target: xtensa-esp32-espidf (ESP32-CAM board)

My project was generated using the cargo template and selected idf version 4.4. Here's my main.rs
```rust
use anyhow::bail;
use embedded_svc::ipv4;
use std::sync::Arc;
use std::thread;
use std::time::Duration;

use embedded_svc::ping::Ping;
use embedded_svc::wifi::{
    ApStatus, ClientConfiguration, ClientConnectionStatus, ClientIpStatus, ClientStatus,
    Configuration, Status, Wifi,
};
use esp_idf_sys as _; // If using the `binstart` feature of `esp-idf-sys`, always keep this module imported

use esp_idf_svc::netif::EspNetifStack;
use esp_idf_svc::nvs::EspDefaultNvs;
use esp_idf_svc::sysloop::EspSysLoopStack;
use esp_idf_svc::wifi::EspWifi;

use esp_idf_svc::ping;

use log::*;

#[allow(dead_code)]
#[cfg(not(feature = "qemu"))]
const SSID: &str = env!("RUST_ESP32_WIFI_SSID");
#[allow(dead_code)]
#[cfg(not(feature = "qemu"))]
const PASS: &str = env!("RUST_ESP32_WIFI_PASS");

fn main() -> anyhow::Result<()> {
    // Temporary. Will disappear once ESP-IDF 4.4 is released, but for now it is necessary to call this function once,
    // or else some patches to the runtime implemented by esp-idf-sys might not link properly.
    esp_idf_sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    // enable anyhow stack traces
    std::env::set_var("RUST_BACKTRACE", "1");

    info!("Creating netif stack");
    let netif_stack = Arc::new(EspNetifStack::new()?);
    info!("Creating sys loop stack");
    let sys_loop_stack = Arc::new(EspSysLoopStack::new()?);
    info!("Creating nvs");
    let default_nvs = Arc::new(EspDefaultNvs::new()?);
    info!("Initializing WiFi");
    let _wifi = wifi(netif_stack, sys_loop_stack, default_nvs)?;
    loop {
        info!("useless loop!");
        thread::sleep(Duration::from_secs(10));
    }
}

fn ping(ip_settings: &ipv4::ClientSettings) -> anyhow::Result<()> {
    info!("About to do some pings for {:?}", ip_settings);

    let ping_summary =
        ping::EspPing::default().ping(ip_settings.subnet.gateway, &Default::default())?;
[... 63 lines cut ...]
        bail!("Unexpected Wifi status: {:?}", status);
    }

    Ok(wifi)
}
```
And log output:

```
...

I (3351) esp_idf_svc::wifi: Found 18 access points
I (3351) esp_idf_svc::wifi: About to get info for found access points
I (3351) esp_idf_svc::wifi: Got info for 18 access points
I (3351) esp_idf_svc::wifi: Found access point AccessPointInfo { ssid: "joshnet", bssid: [176, 106, 65, 202, 45, 158], channel: 1, secondary_channel: None, signal_strength: 202, protocols: EnumSet(), auth_method: WPA2Personal }
I (3371) esp_idf_svc::wifi: Found access point AccessPointInfo { ssid: "SpectrumSetup-5B", bssid: [136, 222, 124, 105, 26, 89], channel: 6, secondary_channel: None, signal_strength: 202, protocols: EnumSet(), auth_method: WPA2Personal }
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ()', /Users/josh/.cargo/registry/src/github.com-1ecc6299db9ec823/heapless-0.7.15/src/string.rs:295:25
stack backtrace:
abort() was called at PC 0x4012b9b2 [abort_return<int>:/Users/josh/Git/esp32-mqtt-node/.embuild/espressif/esp-idf/release-v4.4/components/cxx/cxx_exception_stubs.cpp:33] on core 0
Backtrace:0x40081b9e [panic_abort:/Users/josh/Git/esp32-mqtt-node/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/panic.c:402]:0x3ffbd7600x40087e59 [esp_system_abort:/Users/josh/Git/esp32-mqtt-node/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/esp_system.c:128]:0x3ffbd780 0x4008e3fe [abort:/Users/josh/Git/esp32-mqtt-node/.embuild/espressif/esp-idf/release-v4.4/components/newlib/abort.c:46]:0x3ffbd7a0 0x4012b9b2 [abort_return<int>:/Users/josh/Git/esp32-mqtt-node/.embuild/espressif/esp-idf/release-v4.4/components/cxx/cxx_exception_stubs.cpp:33]:0x3ffbd810 0x400fbae7 [std::backtrace_rs::backtrace::libunwind::trace:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93]:0x3ffbd830 0x400f8004 [std::sys_common::backtrace::_print_fmt:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/sys_common/backtrace.rs:66]:0x3ffbd860 0x401097ce [core::fmt::write:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/fmt/mod.rs:1196]:0x3ffbd8d0 0x400e6fa8 [std::io::Write::write_fmt:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/io/mod.rs:1654]:0x3ffbd920 0x400f7f2b [std::sys_common::backtrace::_print:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/sys_common/backtrace.rs:48]:0x3ffbd970 0x400efadb [std::panicking::default_hook::{{closure}}:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panicking.rs:??]:0x3ffbd9c0 0x400ef978 [std::panicking::default_hook:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panicking.rs:314]:0x3ffbda20 0x400efc83 [std::panicking::rust_panic_with_hook:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panicking.rs:698]:0x3ffbda90 0x400f81cf [std::panicking::begin_panic_handler::{{closure}}:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panicking.rs:??]:0x3ffbdb00 0x400f8095 [std::sys_common::backtrace::__rust_end_short_backtrace:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/sys_common/backtrace.rs:138]:0x3ffbdb30 0x400efb68 [rust_begin_unwind:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panicking.rs:584]:0x3ffbdb60 0x4010a4a8 [core::panicking::panic_fmt:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/panicking.rs:142]:0x3ffbdb90 0x4010c083 [core::result::unwrap_failed:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/result.rs:1785]:0x3ffbdbd0 0x400dc6fd [core::result::Result<T,E>::unwrap:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/result.rs:1078]:0x3ffbdc30 0x400dc7be [<heapless::string::String<_> as core::convert::From<&str>>::from:/Users/josh/.cargo/registry/src/github.com-1ecc6299db9ec823/heapless-0.7.15/src/string.rs:295]:0x3ffbdc60 0x400dc198 [<T as core::convert::Into<U>>::into:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/convert/mod.rs:550]:0x3ffbdc80 0x400d8489 [esp_idf_svc::wifi::<impl core::convert::From<esp_idf_svc::private::common::Newtype<&esp_idf_sys::bindings::wifi_ap_record_t>> for embedded_svc::wifi::AccessPointInfo>::from:/Users/josh/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.42.1/src/wifi.rs:170]:0x3ffbdca0 0x400d79ed [<T as core::convert::Into<U>>::into:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/convert/mod.rs:550]:0x3ffbdcf0 0x400d5c58 [esp32_mqtt_node::wifi:/Users/josh/Git/esp32-mqtt-node/src/main.rs:83]:0x3ffbddc0 0x4017220b [core::ops::function::FnOnce::call_once:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/ops/function.rs:248]:0x3ffbe100 0x400d56b4 [std::rt::lang_start::{{closure}}:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/rt.rs:145]:0x3ffbe120 0x400e6c65 [core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/ops/function.rs:280]:0x3ffbe140 0x400d56a2 [std::rt::lang_start:/Users/josh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/rt.rs:144]:0x3ffbe160 0x400d62b5 [main:??:??]:0x3ffbe190 0x400db877 [app_main:/Users/josh/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.6/src/start.rs:46]:0x3ffbe1b0 0x40176887 [main_task:/Users/josh/Git/esp32-mqtt-node/.embuild/espressif/esp-idf/release-v4.4/components/freertos/port/port_common.c:141]:0x3ffbe1d0
ELF file SHA256: 0000000000000000
Rebooting...

...
```

ivmarkov, 2022-07-28:
Yep, sorry about that. It seems you have an SSID around you which is longer than 30 characters. I was probably sleeping when checking wikipedia. SSIDs can be up to 32 characters, but I have encoded 30 chars here: https://github.com/esp-rs/embedded-svc/blob/master/src/wifi.rs#L142

I have now fixed it in `embedded-svc` master. Can you patch the dependency to `embedded-svc` in your binary crate to pull from GitHub master directly rather than from `crates.io`? You can check how this is done in the Cargo Book, where it talks about `[patch.crates-io]`.

Let me know if the patched version works, so that I can release a new `embedded-svc` increment and close this.

jlocash, 2022-07-28:
It looks like that did the trick! 🎉

decentclock, 2022-08-02:
Hello @ivmarkov 

Just reproduced this issue on both my personal project and on your `rust-esp32-demo` latest master.

On my personal project, adding the line `embedded-svc = { git = "https://github.com/esp-rs/embedded-svc.git" }` fixes the issue here.

However, when adding that same line to your `rust-esp32-demo`, I get the build error below:

```
   Compiling esp-idf-svc v0.42.1
error[E0437]: type `Read` is not a member of trait `Response`
   --> /Users/beraf/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.42.1/src/http/client.rs:342:5
    |
342 | /     type Read<'b>
343 | |     where
344 | |         'a: 'b,
345 | |     = &'b mut EspHttpResponse<'a>;
    | |__________________________________^ not a member of trait `Response`

error[E0407]: method `reader` is not a member of trait `Response`
   --> /Users/beraf/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.42.1/src/http/client.rs:347:5
    |
347 | /     fn reader(&mut self) -> Self::Read<'_> {
348 | |         self
349 | |     }
    | |_____^ not a member of trait `Response`

error[E0437]: type `Read` is not a member of trait `Request`
   --> /Users/beraf/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.42.1/src/http/server.rs:430:5
    |
430 | /     type Read<'b>
431 | |     where
432 | |         'a: 'b,
433 | |     = &'b mut EspHttpRequest<'a>;
    | |_________________________________^ not a member of trait `Request`

error[E0407]: method `get_request_id` is not a member of trait `Request`
   --> /Users/beraf/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.42.1/src/http/server.rs:435:5
    |
435 | /     fn get_request_id(&self) -> &'_ str {
436 | |         if let Some(id) = unsafe { self.id.get().as_ref().unwrap() }.as_ref() {
437 | |             id.as_ref()
438 | |         } else {
...   |
451 | |         }
452 | |     }
    | |_____^ not a member of trait `Request`

error[E0407]: method `query_string` is not a member of trait `Request`
   --> /Users/beraf/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.42.1/src/http/server.rs:454:5
    |
454 | /     fn query_string(&self) -> &str {
455 | |         if let Some(query_string) = unsafe { self.query_string.get().as_ref().unwrap() }.as_ref() {
456 | |             query_string.as_ref()
457 | |         } else {
...   |
491 | |         }
492 | |     }
    | |_____^ not a member of trait `Request`

error[E0407]: method `reader` is not a member of trait `Request`
   --> /Users/beraf/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.42.1/src/http/server.rs:494:5
    |
494 | /     fn reader(&mut self) -> Self::Read<'_> {
495 | |         self
496 | |     }
    | |_____^ not a member of trait `Request`

error[E0220]: associated type `Read` not found for `Self`
   --> /Users/beraf/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.42.1/src/http/client.rs:347:35
    |
347 |     fn reader(&mut self) -> Self::Read<'_> {
    |                                   ^^^^ associated type `Read` not found

error[E0220]: associated type `Read` not found for `Self`
   --> /Users/beraf/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.42.1/src/http/server.rs:494:35
    |
494 |     fn reader(&mut self) -> Self::Read<'_> {
    |                                   ^^^^ associated type `Read` not found

Some errors have detailed explanations: E0220, E0407, E0437.
For more information about an error, try `rustc --explain E0220`.
error: could not compile `esp-idf-svc` due to 8 previous errors
```

Let me know if you have any questions, and thanks for the fix!

MabezDev, 2022-08-05:
@decentclock the HTTP traits are going through some changes, hence the reason why they're behind the `experimental` feature :D 

Closing as the original issue is fixed :)

closed 2022-08-05 by MabezDev

ivmarkov, 2022-08-05:
@decentclock - there is a newly released `embedded-svc` increment, which should fix the issue in `rust-esp32-std-demo` if you do `cargo update`. The issue will be fixed with and without the "experimental" feature enabled (which is now again disabled by default for the demo, as the experimental stuff changes too often recently).

decentclock, 2022-08-05:
> @decentclock - there is a newly released `embedded-svc` increment, which should fix the issue in `rust-esp32-std-demo` if you do `cargo update`. The issue will be fixed with and without the [...]

@ivmarkov Can confirm the issue is fixed now in that repo as well :)