esp-rs/esp-idf-svc #111: Feature Request: Allow customization of httpd::Server configuration

Snapstromegon, 2022-07-28:
Right now you can only set the http and https port of the httpd server.

Internally this gets resolved to a bigger configuration with more interesting options like max_uri_handlers or in my case uri_match_fn.

Usecase:
I want to build a static fileserver using rust-embed and for that I need/want a wildcard registration (which require the uri_match_fn) so I can lookup the file in a folder and serve that then.

In addition to this the embedded_svc::httpd::Request struct would benefit from a "uri" property (or similar) to get which uri this request belongs to (I didn't see it on first sight). I wanted to check in here first if that's at all wanted by the project.

Workaround currently possible:
Register a handler on "/" and use a query parameter. This feels really dirty, as this is incompatible with many other things like providing a rest API.

ivmarkov, 2022-07-28:
> Right now you can only set the http and https port of the httpd server.
>
> Internally this gets resolved to a bigger configuration with more interesting options like max_uri_handlers or in my case [...]

I would suggest to instead look into the `embedded_svc::http::server` traits & configuration.
The `embedded_svc::httpd` one is scheduled for removal, as its API inherently requires allocation and heavy in-memory structures.

While the newer `embedded_svc::http::server` APIs are still marked as "experimental" and - in fact - will change significantly in the next few weeks, they are the future, as their traits are now proven not to require an allocating implementation and can be very efficiently implemented in pure Rust with no_std and no-alloc.

What's important for you is that I don't plan to change the `Configuration` struct of the ESP-IDF impl of the newer APIs, so you can look at it for reference and suggest changes / enhancements.

> In addition to this the embedded_svc::httpd::Request struct would benefit from a "uri" property (or similar) to get which uri this request belongs to (I didn't see it on first sight). I wanted to check in here first if that's at all wanted by the project.

It would be great to have it (for one, how do I even process HTTP server API-s where the parameters are encoded in the path rather than the query string itself?!) however there is no URI API in ESP-IDF itself.

We can introduce an optional trait - say - `Uri` in the newer APIs that will **not** be implemented by the ESP-IDF wrapper, but will be by its pure Rust equivalent.

> Workaround currently possible: Register a handler on "/" and use a query parameter. This feels really dirty, as this is incompatible with many other things like providing a rest API.

Indeed. Even though I don't think you necessarily need to register at "/". You can register at any path, but you **don't** have access to to that path, or its suffix.

Snapstromegon, 2022-07-28:
> I would suggest to instead look into the embedded_svc::http::server traits & configuration.

I took a look at the new http module and at first sight it looks great! Just the uri_match_fn is missing in the config and you could expose the URI from the EspHttpRequest.raw_req.

> It would be great to have it (for one, how do I even process HTTP server API-s where the parameters are encoded in the path rather than the query string itself?!) however there is no URI API in ESP-IDF itself.

This is a really common pattern in e.g. Rest APIs and there are many existing libraries for parsing URL strings with patterns.
The ESP-IDF already exposes the information which URL belongs to the request on the [http_req](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html#_CPPv49httpd_req) type which is currently used by the esp-idf-svc crate to create the Request implementation ([as seen here](https://esp-rs.github.io/esp-idf-svc/src/esp_idf_svc/httpd.rs.html#20-23) - same in the new http impl).
So IMO it should just be a matter of exposing that information. Handling that data could be left to the user.

> Indeed. Even though I don't think you necessarily need to register at "/". You can register at any path, but you don't have access to to that path, or its suffix during request processing.

Yes. you could register anywhere, "/" was just the first example and then you can handle requests like "/?file=index.html".

Snapstromegon, 2022-07-28:
One thing: It seems like the new implementation no longer supports middlewares.
This is a fairly useful feature for me, so I wanted to ask wether this is done intentional for some reason, or if it just happened to not beeing reimplemented?

ivmarkov, 2022-07-28:
1. Good catch for the Uri - I'll expose it!

2. The Middleware is still there in the new API. What is missing in the latest version is the notion of Registry, as it cannot be implemented in no-alloc way. By the way, the latest latest version is not in master but here: https://github.com/esp-rs/embedded-svc/tree/http-server-traits-rework

ivmarkov, 2023-01-21:
Most of this should be resolved with the new release of `esp-idf-svc` a couple of months ago. Feel free to open more specific bugs for anything you feel is not addressed yet.

closed 2023-01-21 by ivmarkov