esp-rs/esp-idf-svc #116: Commented Storage for EspNvsStorage implementation. No get/put functions on EspNvsStorage.

jBernavaPrah, 2022-08-06:
Hi, 
I'm new to esp-rs and I'm trying to play a little. 

I saw, thanks to github examples, that the version esp-idf-svc = "0.36.1", has convenient api for the storage. 

```rust
let nvs = Arc::new(EspDefaultNvs::new()?);
let mut storage = EspNvsStorage::new_default(nvs.clone(), "config", true)?;

if !storage.contains("configured")? {
    storage.put(WIFI_SSID_KEY, &env!("WIFI_SSID"))?;
    storage.put(WIFI_PASS_KEY, &env!("WIFI_PASS"))?;
    storage.put("configured", &"true")?;
    info!("stored creds");
}

let ssid: String = storage.get(WIFI_SSID_KEY)?.unwrap();
let password: String = storage.get(WIFI_PASS_KEY)?.unwrap();
``` 

But the latest version `esp-idf-svc = "0.42.1"`, the one that I'm using now, the implementation of the Storage for EspNvsStorage has been commented with a TODO.

```rust
// TODO
// impl Storage for EspNvsStorage {
//     fn get<'a, T>(&'a self, name: &str) -> Result<Option<T>, Self::Error>
//     where
//         T: serde::Deserialize<'a> {
//         todo!()
//     }

//     fn set<T>(&mut self, name: &str, value: &T) -> Result<bool, Self::Error>
//     where
//         T: serde::Serialize {
//         todo!()
//     }
// }
```

Is this because there is a different way to use those functions? In this case, an example will be super useful.

ivmarkov, 2022-08-06:
The `get`/`put` API will return (sort of) with the next major release of `embedded-svc`. You can preview it currently in branch `next`. It will however NOT be 100% backward compatible with what used to be there.

What used to be there was hard-wired to json serializarion which is (a) verbose (b) the way the API looked like it could not be implemented without allocatiins. Thw new embedded-svc definea the traits un a way where it is guaranteed thwy can be implemented inno_std and without allocations, as the current STD and allocating implementation based in esp-idf will hopefully not be the only o e in future.

You can keep this issue open until the `next` branxh is released.

jBernavaPrah, 2022-08-07:
Thanks for the response!

I’m not seeing the implementation of the Storage on the next branch, as you mentioned. :(

When do you think will be available the major update?

ivmarkov, 2022-08-07:
> Thanks for the response!
> 
> I’m not seeing the implementation of the Storage on the next branch, as you mentioned. :(

It is in the `next` branch of `embedded-svc`.
> When do you think will be available the major update?

In a few weeks.

DS3a, 2022-11-26:
Any updates regarding this? I too am new to `esp-rs` and find myself using this feature in a future project. I am open to contributing towards the `get`/`put` API, or something to enable access to the permanent flash memory on the esp.

ivmarkov, 2022-11-26:
This is released since a week. Check CHANGELOG.md in embedded-svc root and in esp-idf-svc root.

closed 2022-11-26 by ivmarkov