esp-rs/esp-idf-svc #122: The state of EspNvsStorage

tcbennun, 2022-08-15:
Could you please help me understand what the state of affairs is with the NVS storage part, @ivmarkov ? Looking to contribute.

1. Why does the `RawStorage` impl utilise both `u64` and `blob`? Is it just more space-efficient to store with `u64` if the blob is 8 bytes or less?
2. For the future `Storage` impl, are we to allow the user to choose a serialization format, or should we just impl with something efficient and embedded-friendly like [postcard](https://github.com/jamesmunns/postcard)?
3. Should the `Storage` trait have its own associated error type, separate from that of `StorageBase`, in order to encompass (de)serialization errors?

ivmarkov, 2022-08-15:
> Could you please help me understand what the state of affairs is with the NVS storage part, @ivmarkov ? Looking to contribute.
> 
> 1. Why does the `RawStorage` impl utilise both `u64` and `blob`? Is it just more space-efficient to store with `u64` if the blob is 8 bytes or less?

Yes. `RawStorage` impl for ESP-IDF is implemented by utilizing the [ESP-IDF NVS library](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html). Using BLOBs for everything is not very efficient.
 
> 2. For the future `Storage` impl, are we to allow the user to choose a serialization format, or should we just impl with something efficient and embedded-friendly like [postcard](https://github.com/jamesmunns/postcard)?

Users will be able to whichever serde they want [by implementing the `SerDe` trait](https://github.com/esp-rs/embedded-svc/blob/next/src/storage.rs#L112). Or just implement their own `Storage` impl, if they want to do something more exotic.

> 3. Should the `Storage` trait have its own associated error type, separate from that of `StorageBase`, in order to encompass (de)serialization errors?

The fact that `Storage` shares the error type with `StorageBase` does not mean that its impl (i.e. the `SerDe`-based impl that we'll provide out of the box) is obliged to share its error type with a potential `RawStorage` impl to which it might delegate. See [here](https://github.com/esp-rs/embedded-svc/blob/next/src/storage.rs#L161).

I might get rid of `EitherError` though in favour of a dedicated error type, but in any case the dedicated error type would still have an "either" semantics.

tcbennun, 2022-08-15:
Thanks. I didn't know about the next branch in embedded-svc, that makes things a lot clearer! When are you looking to merge that in?

ivmarkov, 2022-08-15:
In a week, max two. Need to remove/move elsewhere a bit more util stuff from embedded-svc and then it should be ready.

ivmarkov, 2023-01-21:
I think this is all addresses with the new release since a couple of months. There is a separate issue request for compatibility between the Rust storage impl and the C one.

closed 2023-01-21 by ivmarkov