esp-rs/esp-idf-svc #126: EspHttpClient doesn't support `keep-alive`

AnthonyGrondin, 2022-09-07:
## Description
The current implementation of EspHttpClient doesn't support the `keep-alive` functionnality of HTTP connections.

According to the [documentation](https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32/api-reference/protocols/esp_http_client.html#_CPPv424esp_http_client_config_t), we can specify the following attributes to `esp_http_client_config_t`:
 - [keep_alive_enable](https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32/api-reference/protocols/esp_http_client.html#_CPPv4N24esp_http_client_config_t17keep_alive_enableE)
 - [keep_alive_idle](https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32/api-reference/protocols/esp_http_client.html#_CPPv4N24esp_http_client_config_t15keep_alive_idleE)
 - [keep_alive_interval](https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32/api-reference/protocols/esp_http_client.html#_CPPv4N24esp_http_client_config_t19keep_alive_intervalE)
 - [keep_alive_count](https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32/api-reference/protocols/esp_http_client.html#_CPPv4N24esp_http_client_config_t16keep_alive_countE)

Also, the current implementation doesn't call [esp_http_client_close](https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32/api-reference/protocols/esp_http_client.html#_CPPv421esp_http_client_close24esp_http_client_handle_t) after the `keep-alive` timeout or max as been reached.

## Expected behavior

We should be able to pass the `keep-alive` attributes to `EspHttpClientConfiguration` either by specifiying each attributes, or by passing a struct, that contains the properties, whichever is the most idiomatic.

`esp_http_client_close` should also be called, after the connection has been idle for the number of seconds specified in timeout, or after reaching the maximum number of requests that can be sent on this connection.

## External documentation
https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Keep-Alive

ivmarkov, 2022-09-22:
@AnthonyGrondin Would you be willing to contribute a PR against `master` for this?

AnthonyGrondin, 2022-09-24:
I don't think I have the required knowledge to come up with an ideal implementation of it.

The main challenge here, is to work asynchronously, since the implementation has to work kind of like a watchdog. Upon a new request, a timer is started, with the value of the `keep-alive` header, and decrements each second. If another request is made during that period, before the timer reaches 0, it has to be reset from the value of the `keep-alive` header  in that request.

ivmarkov, 2022-10-01:
> I don't think I have the required knowledge to come up with an ideal implementation of it.
>
> The main challenge here, is to work asynchronously, since the implementation has to work kind of like a [...]

Are you sure that Espressif does not have an automatic keep-alive management?

closed 2026-02-05 by ivmarkov