esp-rs/esp-idf-svc #127: crash on changing `wifi.set_configuration`

chungwong, 2022-09-13:
Given the following snippet for changing WIfi config from `Mixed` to `ClientConfiguration`, it will crash the whole program
```rs
struct Wifi {
    esp_wifi: EspWifi,
}

impl Wifi {
  // Disable AP mode
  fn remove_ap(&mut self) -> anyhow::Result<()> {
   if let Ok(Mixed(client_configuration, _)) = self.esp_wifi.get_configuration() {
      info!("Change Wifi to ClientConfiguration");
      self.esp_wifi
      .set_configuration(&Client(client_configuration))?;
   }
   Ok(())
  }

  fn new() -> anyhow::Result<Self> {
    //...
    wifi.set_configuration(&Mixed(
      ClientConfiguration {
        ssid: ssid.into(),
        password: psk.into(),
        channel,
        auth_method,
        ..Default::default()
      },
      AccessPointConfiguration {
        ssid: "ESP32 C3".into(),
        password: "admin".into(),
        channel: channel.unwrap_or(1),
         ..Default::default()
      },
    ))?;

    OK(Self { esp_wifi: wifi });
   }
}

fn main() {
  let mut wifi = Wifi::new().unwrap();

  // both AP and STA mode are working
  

  // later on, BOOT button triggers this `remove_ap` and the whole program crashed
  wifi.remove_ap();
}
```

Logs
```bash
I (42864) esp_idf_svc::wifi: Getting configuration
I (42864) esp_idf_svc::wifi: Providing STA configuration: ClientConfiguration { ssid: "Chung", bssid: None, auth_method: WPA2Personal, password: "*******", channel: None, ip_conf: Some(DHCP(DHCPClientSettings { hostname: None })) }
I (42874) esp_idf_svc::wifi: Providing AP configuration: AccessPointConfiguration { ssid: "ESP32 C3", ssid_hidden: false, channel: 7, secondary_channel: None, protocols: EnumSet(), auth_method: None, password: "admin", max_connections: 10, ip_conf: Some(RouterConfiguration { subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dhcp_enabled: true, dns: Some(8.8.8.8), secondary_dns: Some(8.8.4.4) }) }
I (42914) esp_idf_svc::wifi: Configuration gotten: Mixed(ClientConfiguration { ssid: "Chung", bssid: None, auth_method: WPA2Personal, password: "*******", channel: None, ip_conf: Some(DHCP(DHCPClientSettings { hostname: None })) }, AccessPointConfiguration { ssid: "ESP32 C3", ssid_hidden: false, channel: 7, secondary_channel: None, protocols: EnumSet(), auth_method: None, password: "admin", max_connections: 10, ip_conf: Some(RouterConfiguration { subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dhcp_enabled: true, dns: Some(8.8.8.8), secondary_dns: Some(8.8.4.4) }) })
I (42964) hydro::wifi: Change Wifi to ClientConfiguration
I (42974) esp_idf_svc::wifi: Setting configuration: Client(ClientConfiguration { ssid: "Chung", bssid: None, auth_method: WPA2Personal, password: ""*******", channel: None, ip_conf: Some(DHCP(DHCPClientSettings { hostname: None })) })
I (42994) esp_idf_svc::wifi: Stopping
I (42994) wifi:state: run -> init (0)
I (43004) wifi:pm stop, total sleep time: 28474937 us / 32743363 us
W (43004) wifi:<ba-del>idx
W (43004) wifi:<ba-del>idx
I (43014) wifi:new:<7,0>, old:<7,1>, ap:<7,1>, sta:<7,0>, prof:7
I (43014) esp_idf_svc::wifi: Disconnect requested
I (43164) wifi:flush txq
I (43164) wifi:stop sw txq
I (43164) wifi:lmac stop hw txq
I (43164) esp_idf_svc::wifi: Stop requested
I (43164) esp_idf_svc::wifi: Stopped
[... 84 lines cut ...]
3fc9a2b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9a2d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9a2f0: 0x00000000 0x3f000000 0x00000054 0x3fc9a2fc 0x3fc9a2fc 0x3fc9a2fc 0x3fc9a2fc 0x00000000
3fc9a310: 0x3fc9a314 0xffffffff 0x3fc9a314 0x3fc9a314 0x00000001 0x3fc9a328 0xffffffff 0x3fc9b20c
3fc9a330: 0x3fc9b20c 0x00000000 0x00000001 0x00000000 0xaf00ffff 0x00000000 0xb33fffff 0x00000000
3fc9a350: 0x00000054 0x3fc9a354 0x3fc9a354 0x3fc9a354 0x3fc9a354 0x00000000 0x3fc9a36c 0xffffffff

ELF file SHA256: 0000000000000000
Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381836
0x40381836 - esp_restart_noos_dig
    at /home/chung/.espressif/esp-idf/v4.4.1/components/esp_system/esp_system.c:44
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x403ce000,len:0x930
0x403ce000 - _coredump_iram_start
    at ??:??
load:0x403d0000,len:0x2d28
0x403d0000 - _coredump_iram_start
    at ??:??
```

mohs8421, 2022-09-21:
I have the same issue, and I found a hint on how to solve it here: https://github.com/espressif/esp-idf/issues/4411
I assume this fix has to be incorporated into this library.

ivmarkov, 2022-10-09:
Should be fixed and released.

closed 2022-10-09 by ivmarkov