esp-rs/esp-idf-svc #128: Timer does not trigger

usbalbin, 2022-09-13:
The callback function does not seem to get called. There is no `Hi!` printed in the console.

```rust
fn main() {
    use esp_idf_sys as _;

    use std::time::Duration;
    use embedded_svc::timer::{TimerService, PeriodicTimer};
    use embedded_hal_0_2::prelude::_embedded_hal_blocking_delay_DelayMs;
    use esp_idf_hal::delay::FreeRtos;
    
    esp_idf_sys::link_patches();

    esp_idf_svc::timer::EspTimerService::new()
        .unwrap()
        .timer(|| println!("Hi!"))
        .unwrap()
        .every(Duration::from_secs(1))
        .unwrap();
    
    println!("Waiting for hi!");
    loop {
        FreeRtos.delay_ms(10000u16);
    }
}
```

same thing with `timer.after(...)`. `Waiting for hi!` is printed but nothing more, I have also tried toggling pins but nothing happens.

This is all with the untouched `sdkconfig.default` from the`cargo generate --git https://github.com/esp-rs/esp-idf-template cargo`

ivmarkov, 2022-09-13:
You are supposed to keep around the returned timer so that it does not get dropped immediately.

closed 2022-09-13 by ivmarkov

usbalbin, 2022-09-13:
Oh.. Thank you.