esp-rs/esp-idf-svc #135: Cannot understand EspHttpServer::handler signature

snaiperskaya96, 2022-09-25:
Hi, sorry to open a new issue for this. I was a bit reluctant but i cannot really figure it out by myself and the samples around the net all seem to use bits of code that are not there anymore (EspHttpServer::handle_get,get,at.../embedded_svc::http::server::registry,ResponseData etc) so i cannot even find a working example.

I'm a rust newbie, i'm having big troubles figuring out what the signature for the callback in  EspHttpServer::handler is. I just cannot decode what the requirements are.
If i use something like this 
```rust
    server.handler("/", embedded_svc::http::Method::Get, |_, _| Ok(()));
```
I get 
```rust

error[E0277]: the trait bound `for<'a, 'b> for<'r> fn(embedded_svc::http::server::Request<esp_idf_svc::http::server::EspHttpConnection<'r>>) -> Result<(), HandlerError> {test}: embedded_svc::http::server::Handler<&'a mut esp_idf_svc::http::server::EspHttpConnection<'b>>` is not satisfied
   --> src/main.rs:57:58
    |
57  |     server.handler("/", embedded_svc::http::Method::Get, test);
    |            -------                                       ^^^^ the trait `for<'a, 'b> embedded_svc::http::server::Handler<&'a mut esp_idf_svc::http::server::EspHttpConnection<'b>>` is not implemented for fn item `for<'r> fn(embedded_svc::http::server::Request<esp_idf_svc::http::server::EspHttpConnection<'r>>) -> Result<(), HandlerError> {test}`
    |            |
    |            required by a bound introduced by this call
    |
    = help: the following other types implement trait `embedded_svc::http::server::Handler<C>`:
              <&H as embedded_svc::http::server::Handler<C>>
              <CompositeHandler<M, H> as embedded_svc::http::server::Handler<C>>
              <FnConnectionHandler<F> as embedded_svc::http::server::Handler<&mut C>>
              <FnHandler<F> as embedded_svc::http::server::Handler<&mut C>>
note: required by a bound in `EspHttpServer::handler`
   --> /home/esp/.cargo/git/checkouts/esp-idf-svc-a28457b0e32c6283/e19819c/src/http/server.rs:264:12
    |
264 |         H: for<'a, 'b> Handler<&'a mut EspHttpConnection<'b>> + 'static,
    |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ required by this bound in `EspHttpServer::handler`
```

So could i get some pointers on what the library is expecting to see? 
Sorry again for opening an issue just for this, I just really want to use rust for my project rather than good ole' c++.

snaiperskaya96, 2022-09-25:
Hate to find the solution a few minutes after finally opening a ticket after hours of struggle but...
I was using the master branch of both embedded_svc and esp_idf_svc (don't remember how i ended up doing that) but sticking to the latest version with experimental = true turned it into a much easier to understand 
```rust
.handle(
            "/",
            embedded_svc::http::Method::Get,
            |req: EspHttpRequest, res: EspHttpResponse| {
                res.send_str("Hello");
                Ok(())
            },
        )
```
so im gonna close this for the time being. Sorry to have bothered you again.

closed 2022-09-25 by snaiperskaya96

Vollbrecht, 2022-09-25:
Everybody begins somewhere, no problem. The compiler try's to tell you that you don't fulfull the traitbounds that the handler Fn expects you from doing so. If you have a look at the handler Fn you see the following:
```
    fn set_handler<H>(
        &mut self,
        uri: &str,
        method: Method,
        handler: H,
    ) -> Result<&mut Self, Self::Error>
    where
        H: for<'a> Handler<Self::Request<'a>, Self::Response<'a>> + 'static,
    {
        let mut prefixed_uri = heapless::String::<N>::new();

        write!(&mut prefixed_uri, "{}{}", self.prefix, uri).unwrap();

        self.registry.set_handler(&prefixed_uri, method, handler)?;

        Ok(self)
    }
```
the important bits is handler is a Generic H that expects you to implement H with Request and Response. (look at the where line ) By given the compiler both | _, _ |  anonymous he cant now whether the first or the second anonymous should be a request or a response so he cant automatically do it for you, you need to be explicit. Trait bounds can look intimidating at start but you should try to get a hang of them. They are really powerfull

snaiperskaya96, 2022-09-25:
Thank you @Vollbrecht, i really appreciate you taking the time to explain this. I'll try to hit my head on trait bounds until I finally can properly understand them!  :)