esp-rs/esp-idf-svc #139: FirmwareInfo does not contain correct data

axos88, 2022-09-27:
I would expectthe cargo.toml package info (name, version) to be written to the app_info_t in the image header, but it's not. The build time is not updated either.

ivmarkov, 2022-09-27:
Would you contribute a PR?

axos88, 2022-09-27:
I did take a look, but I have no idea how the whole build process is put together. It seems that esp IDF itself would need to be modified as well... Not sure I can tackle this one. 

As far as I saw - at least for the build time - the issue is that the descriptor is put there by a variable in a special section of the resulting elf file. This is done during the build of esp-idf itself, and esp IDF uses the __date__ and __file__ macros supplied by the compiler, but those files are not going to be recompiled if the esp-idf-sys package is not rebuilt, so the subsequent builds of the rust stuff does not update the descriptor.

axos88, 2022-09-27:
I'm not sure if it's feasable to overwrite that variable value.

ivmarkov, 2022-09-28:
Can you point me to the build  code in esp-idf that does it?

axos88, 2022-09-28:
Yeah, I was on my phone yesterday already. Here's the link: https://github.com/espressif/esp-idf/blob/master/components/esp_app_format/esp_app_desc.c

ivmarkov, 2022-09-28:
I think the best approach is to first implement a PR for ESP-IDF where the whole declaration and subsequent initialization of the `esp_app_desc_t` structure is skipped in case there is a new configuration variable present (say, `CONFIG_APP_EXTERNAL_DESC`). In that case this code should instead just do `extern esp_app_desc_t` and expect some other code to provide this symbol.

Note also that we can initially implement the PR as a custom patch that triggers during the build of `esp-idf-sys`. I.e. see [here](https://github.com/esp-rs/esp-idf-sys/tree/master/patches) and [here](https://github.com/esp-rs/esp-idf-sys/blob/master/build/common.rs#L23).

Once we have the above, the `esp_app_desc_t` structure can actually be defined from *Rust* code, and there we have full control over it.

axos88, 2022-09-28:
Sounds like a good idea, I was thinking of something similar too. Did not think about the patches, but those would have to be generated dynamically (because of the build time), and somehow we would need to trigger the rebuild of the sys crate even if there were no changes (because of the generation of the build time again).

axos88, 2022-09-28:
Orrrr.... We could patch the esp_app_desc.c to remove the definition, and leave only an external declaration, and generate the symbol from rust code during the build rather than trying to inject it into C.

axos88, 2022-09-28:
Hah, yeah. I started tinkering with this one, but it won't work with an external definition, because - as far as I see, the process builds a libespidf.elf, and (i'm guessing) it would later link the rust code into that?...

Anyways, If I define a definition for esp_app_desc in the rust code, it complains that there are two definitions for the symbol. Perfectly normal.
But now if I remove the definition from esp_idf itself, during the link of libespidf.elf, it complains that the symbol is not defined.

Experimenting with __weak__ instead, and that seems to compile.

axos88, 2022-09-28:
Yep, that works, now we just need to figure out how to populate it with data.

ivmarkov, 2022-09-28:
* I agree `__weak__` is a MUCH better idea, and would most likely be upstreamed without any complaints as long as it works indeed!
* No, libespidf.elf is not linked and cannot be linked (it is an ELF file!) into Rust. This is just a side effect. We are instead linking all libXXX.a files that are created during the ESP-IDF compilation and we know about those by asking the ESP-IDF build process to output its linker args
* Populating with data is not a rocket science. Check [this](https://stackoverflow.com/a/48965642) for an inspiration.

axos88, 2022-09-28:
Yeah, please see my comments on the wip pr on the sys create. The build time is not as easy as pie unfortunately.

ivmarkov, 2022-10-03:
The PR is now merged and a new `esp-idf-sys` release is on its way.

@axos88 Can you link here the relevant ESP IDF issue for putting `__weak__` or create one?

axos88, 2022-10-03:
https://github.com/espressif/esp-idf/pull/9899

axos88, 2022-10-03:
It's already approved, but I'm not sure what that shasum means in the comment.

axos88, 2022-10-09:
https://github.com/espressif/esp-idf/pull/9899#event-7548426156 has been merged. I'll remove the master patch shortly

ivmarkov, 2022-10-09:
The `__weak__` fix is merged in ESP-IDF master. Not sure if they'll propagate to the release branches, but we have the `esp-idf-sys` patches for that anyway.

closed 2022-10-09 by ivmarkov