esp-rs/esp-idf-svc #159: can't get calls to EspWifi::new to compile

liebman, 2022-11-10:
I'm attempting to use Wifi from the current master. I've not been able to find a way to make the call to `EspWifi::new` work.  Almost every attempt I end up with `the trait DerefMut is not implemented for Modem`.  I'm using the current master from `esp-idf-hal` and `esp-idf-svc`.  I'm using an `esp32-s3`.

Here is my small test program:
```
use anyhow::Context;
use embedded_svc::wifi::ClientConfiguration;
use embedded_svc::wifi::{AuthMethod, Configuration};
use esp_idf_hal::{prelude::Peripherals};
use esp_idf_svc::nvs::EspDefaultNvsPartition;
use esp_idf_svc::{eventloop::EspSystemEventLoop, wifi::EspWifi};
use esp_idf_sys as _; // If using the `binstart` feature of `esp-idf-sys`, always keep this module imported

fn main() -> anyhow::Result<()> {
    // Temporary. Will disappear once ESP-IDF 4.4 is released, but for now it is necessary to call this function once,
    // or else some patches to the runtime implemented by esp-idf-sys might not link properly.
    esp_idf_sys::link_patches();
    println!("Running!");
    let ssid: heapless::String<32> = "something-not-secret".into();
    let psk: heapless::String<64> = "something-secret".into();

    let peripherals = Peripherals::take().context("failed to take Peripherals")?;
    let sysloop = EspSystemEventLoop::take()?;
    let nvs = EspDefaultNvsPartition::take()?;
    let _wifi = EspWifi::new(peripherals.modem, sysloop, Some(nvs))?;
    let conf = Configuration::Client(ClientConfiguration {
        ssid,
        auth_method: AuthMethod::WPA2Personal,
        password: psk,
        ..Default::default()
    });
    _wifi.set_configuration(&conf)?;
    _wifi.start();

    loop {}
}
```
and the results from the compiler:
```
➜ cargo build
   Compiling esp32-rust-wifi v0.1.0 (/Users/chris.l/Dropbox/git/esp32-rust-wifi)
error[E0277]: the trait bound `Modem: DerefMut` is not satisfied
   --> src/main.rs:20:30
    |
20  |     let _wifi = EspWifi::new(peripherals.modem, sysloop, Some(nvs))?;
    |                 ------------ ^^^^^^^^^^^^^^^^^ the trait `DerefMut` is not implemented for `Modem`
    |                 |
    |                 required by a bound introduced by this call
    |
    = help: the following other types implement trait `esp_idf_hal::peripheral::Peripheral`:
              esp_idf_hal::adc::ADC1
              esp_idf_hal::adc::ADC2
              esp_idf_hal::can::CAN
              esp_idf_hal::gpio::AnyIOPin
              esp_idf_hal::gpio::AnyInputPin
              esp_idf_hal::gpio::AnyOutputPin
              esp_idf_hal::gpio::chip::Gpio0
              esp_idf_hal::gpio::chip::Gpio1
            and 79 others
    = note: required for `Modem` to implement `esp_idf_hal::peripheral::Peripheral`
note: required by a bound in `EspWifi::<'d>::new`
   --> /Users/chris.l/Dropbox/git/esp-idf-svc/src/wifi.rs:853:21
    |
853 |         modem: impl Peripheral<P = M> + 'd,
    |                     ^^^^^^^^^^^^^^^^^ required by this bound in `EspWifi::<'d>::new`

For more information about this error, try `rustc --explain E0277`.
error: could not compile `esp32-rust-wifi` due to previous error
```

Any hints / pointers to make this work?

ivmarkov, 2022-11-10:
I think it would be much easier if you upload somewhere the code snippet from above as a minimal project, with its `Cargo.toml` and everything else, as there is nothing wrong with the above code in general.

I - for one - just recompiled RUWM for the `esp32s3` and I'm not seeing the error you are seeing. Moreover, there is no special branch for `esp32s3` which might be causing the error you are seeing.

So yeah, I believe you have somehow your `Cargo.toml` messed up. But to confirm this, I need a complete mini-project demonstrating the issue.

liebman, 2022-11-10:
Seems to have been an issue with the lockfile. Closing

closed 2022-11-10 by liebman

ekeranen, 2022-11-12:
I observed this issue as well. It went away after I upgraded esp-idf-hal:
```
esp-idf-hal = { version = "0.39", default-features = false, features = ["esp-idf-sys"], git = "https://github.com/esp-rs/esp-idf-hal" }
```

Previously, I was pinning a specific commit (0c3a17b4).