esp-rs/esp-idf-svc #166: changing wifi station hostname not working with new api

HannesGitH, 2022-11-19:
since `ip_conf` is no longer part of `ClientConfiguration` (as it was in #68) one would set the hostname when initializing the wifi like so (or at least that what i did):

```rust
let ipv4_client_cfg =
    embedded_svc::ipv4::ClientConfiguration::DHCP(embedded_svc::ipv4::DHCPClientSettings {
        hostname: Some(heapless::String::<30>::from("THIS SHOULD BE ITS HOSTNAME")),
        ..Default::default()
    });
let new_c = NetifConfiguration {
    ip_configuration: embedded_svc::ipv4::Configuration::Client(ipv4_client_cfg),
    ..NetifConfiguration::wifi_default_client()
};

let esp_wifi = EspWifi::wrap_all(
    WifiDriver::new(
        modem,
        EspSystemEventLoop::take().unwrap(),
        EspDefaultNvsPartition::take().ok(),
    )?,
    EspNetif::new_with_conf(&new_c)?,
    EspNetif::new(NetifStack::Ap)?,
)?;
```

wifi works (AP as well as STA), but my wifi router still shows him as the default "espressif" , not as what was provided in L3

`esp_wifi.sta_netif().get_hostname()` evals to the provided hostname though, but it seems like this doesn't effect the actual underlying config

teamplayer3, 2022-11-24:
When I tested it, it works fine.

HannesGitH, 2022-11-25:
i think it was caused by #168 as my setup was kinda bugged, i dont know why but as the STA mode now works fine, i think this is solved as well

closed 2022-11-25 by HannesGitH