esp-rs/esp-idf-svc #168: http server only reachable via ap not sta mode

HannesGitH, 2022-11-20:
in mixed (`Configuration::Mixed(c, ap)`) or STA mode the http routes wont get invoked when reaching via the external network.

in AP mode or in mixed mode via the internal/hosted network everything works fine..

using `esp_idf_svc::wifi::EspWifi` and `esp_idf_svc::http::server::EspHttpServer` , both with default config

Vollbrecht, 2022-11-20:
can you please specify what version of svc are you using. by routes not invoked you mean through the hostname? what are you trying to use ? hostname.local ? did you try it with the ip address directly ? what type of dns server are you using?

HannesGitH, 2022-11-20:
`esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc", version = "0.43.0", features = ["experimental"]}`

i tried via direct ip address which my router provided (192.168.178.27 in my case), but looking at `esp_wifi.sta_netif_mut().get_ip_info().unwrap().ip,` i get 0 (`0.0.0.0`), which is probably part of the culprit

Vollbrecht, 2022-11-20:
can you try waiting a bit. most of the time you dont directly get ip from your dhcp. please have an look at this example code https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/src/main.rs#L1265 how to wait on wifi init

HannesGitH, 2022-11-20:
<details>
 <summary>log (in mixed mode(ap first then adding sta))</summary>

```log
I (787) wifi:wifi driver task: 3ffc1f6c, prio:23, stack:6656, core=0
I (787) system_api: Base MAC address is not set
I (787) system_api: read default base MAC address from EFUSE
I (787) wifi:wifi firmware version: 2c582c7
I (797) wifi:wifi certification version: v7.0
I (797) wifi:config NVS flash: disabled
I (797) wifi:config nano formating: disabled
I (807) wifi:Init data frame dynamic rx buffer num: 32
I (807) wifi:Init management frame dynamic rx buffer num: 32
I (817) wifi:Init management short buffer num: 32
I (817) wifi:Init dynamic tx buffer num: 32
I (827) wifi:Init static rx buffer size: 1600
I (827) wifi:Init static rx buffer num: 10
I (827) wifi:Init dynamic rx buffer num: 32
I (837) wifi_init: rx ba win: 6
I (837) wifi_init: tcpip mbox: 32
I (847) wifi_init: udp mbox: 6
I (847) wifi_init: tcp mbox: 6
I (847) wifi_init: tcp tx win: 5744
I (857) wifi_init: tcp rx win: 5744
I (857) wifi_init: tcp mss: 1440
I (867) wifi_init: WiFi IRAM OP enabled
I (867) wifi_init: WiFi RX IRAM OP enabled
I (1617) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (1617) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (1777) wifi:mode : softAP (f0:08:d1:d2:f6:35)
I (1787) wifi:Total power save buffer number: 16
I (1787) wifi:Init max length of beacon: 752/752
I (1787) wifi:Init max length of beacon: 752/752
I (41867) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (41867) wifi:station: f4:d4:88:83:a5:09 join, AID=1, bgn, 20
W (41907) wifi:<ba-add>idx:2 (ifx:1, f4:d4:88:83:a5:09), tid:6, ssn:0, winSize:64
W (42107) wifi:<ba-add>idx:3 (ifx:1, f4:d4:88:83:a5:09), tid:0, ssn:2, winSize:64
I (42137) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.71.2
I (54867) wifi:station: f4:d4:88:83:a5:09 leave, AID = 1, bss_flags is 658531, bss:0x3ffcd0a8
I (54867) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
W (54877) wifi:<ba-del>idx
W (54877) wifi:<ba-del>idx
I (54927) wifi:flush txq
I (54927) wifi:stop sw txq
I (54927) wifi:lmac stop hw txq
I (54927) wifi:mode : sta (f0:08:d1:d2:f6:34)
I (54927) wifi:enable tsf
I (57427) wifi:flush txq
I (57427) wifi:stop sw txq
I (57427) wifi:lmac stop hw txq
I (57437) wifi:mode : sta (f0:08:d1:d2:f6:34) + softAP (f0:08:d1:d2:f6:35)
I (57447) wifi:enable tsf
I (57447) wifi:Total power save buffer number: 16
E (57457) wifi:sta is connecting, return error
I (57457) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (58197) wifi:state: init -> auth (b0)
I (58207) wifi:state: auth -> assoc (0)
I (58217) wifi:state: assoc -> run (10)
I (58227) wifi:connected with routerli, aid = 5, channel 1, BW20, bssid = 38:10:d5:ea:21:94
I (58227) wifi:security: WPA2-PSK, phy: bgn, rssi: -55
I (58227) wifi:pm start, type: 1
I (58287) wifi:AP's beacon interval = 102400 us, DTIM period = 1
connected to *ssid* as espressif (0.0.0.0) --------------------------- this is the aformentioned ip 
I (59067) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (59077) wifi:station: f4:d4:88:83:a5:09 join, AID=1, bgn, 20
W (59217) wifi:<ba-add>idx:2 (ifx:1, f4:d4:88:83:a5:09), tid:6, ssn:0, winSize:64
I (59257) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.71.2
W (59807) wifi:<ba-add>idx:3 (ifx:1, f4:d4:88:83:a5:09), tid:0, ssn:2, winSize:64
```
</details>

HannesGitH, 2022-11-20:
> can you try waiting a bit when init wifi. most of the time you dont directly get ip from your dhcp. please have an look at this example code https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/src/main.rs#L1265 how to wait on wifi init

Thanks!

sadly `wait_with_timeout` isnt available in 0.43.0 and the connection seems to be succesful (`esp_wifi.driver_mut().is_sta_connected()` is `true`, and my router displays an actual ip)

Vollbrecht, 2022-11-20:
> Thanks!
>
> sadly `wait_with_timeout` isnt available in 0.43.0 and the connection seems to be succesful (`esp_wifi.driver_mut().is_sta_connected()` is `true`, and my router displays an actual ip)

please look at the provided example. 0.43 include wait_with_timeout. but it is reworked and uses an new WifiWait struct now. its used in the example but otherwise look here : https://github.com/esp-rs/esp-idf-svc/blob/master/src/wifi.rs#L1196

Vollbrecht, 2022-11-20:
i didnt test the mixed mode myself but i am sure that at least pure station and pure ap mode works. please try to confirm first that pure sta mode works or not for you. after that we can look in mixed mode

HannesGitH, 2022-11-20:
Alright, thanks! 
I'll look into it this week

HannesGitH, 2022-11-21:
ok the example uses mixed mode as well, and modifiing my code to use the WifiWait and EspWifiWait still didnt do the trick, sadly (`Wifi did not connect or did not receive a DHCP lease`) 

but i'll run the given example as-is and see when it breaks

HannesGitH, 2022-11-21:
okay the example works fine (after incorperating https://github.com/ivmarkov/rust-esp32-std-demo/pull/117 and increasing flash size)

i'd close this for now as its probably my fault somewhere in reopen if it aint ;)

closed 2022-11-21 by HannesGitH