esp-rs/esp-idf-svc #182: Wifi init assert failed:/Segfault

reinismu, 2022-12-05:
Using latest commits from `esp-idf-svc` I can't get WIFI to work. It seems to scan correctly and do the right things to connect, but the it just crashes with
```
I (5583) esp_idf_svc::netif::status: Got IP event: DhcpIpAssigned(DhcpIpAssignment { netif_handle: 0x3fceb1bc, ip_settings: IpInfo { ip: 192.168.17.17, subnet: Subnet { gateway: 192.168.17.1, mask: Mask(24) }, dns: None, secondary_dns: None }, ip_changed: true })
assert failed: xQueueGenericSend queue.c:821 (!( ( pvItemToQueue == ((void *)0) ) && ( pxQueue->uxItemSize != ( UBaseType_t ) 0U ) ))
Backtrace:0x403759d2:0x3fce12f00x4037b6cd:0x3fce1310 0x40382005:0x3fce1330 0x4037c136:0x3fce1450 0x4037c315:0x3fce1490 0x42099991:0x3fce14b0 0x420999a4:0x3fce1500 
0x403759d2 - panic_abort
    at /home/reinis/projects/rfid/training/w5500-mdns/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/panic.c:402
0x4037b6cd - esp_system_abort
    at /home/reinis/projects/rfid/training/w5500-mdns/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/esp_system.c:128
0x40382005 - __assert_func
    at /home/reinis/projects/rfid/training/w5500-mdns/.embuild/espressif/esp-idf/release-v4.4/components/newlib/assert.c:85
0x4037c136 - xQueueGenericSend
    at /home/reinis/projects/rfid/training/w5500-mdns/.embuild/espressif/esp-idf/release-v4.4/components/freertos/queue.c:821
0x4037c315 - xQueueGiveMutexRecursive
    at /home/reinis/projects/rfid/training/w5500-mdns/.embuild/espressif/esp-idf/release-v4.4/components/freertos/queue.c:685
0x42099991 - esp_event_loop_run
    at /home/reinis/projects/rfid/training/w5500-mdns/.embuild/espressif/esp-idf/release-v4.4/components/esp_event/esp_event.c:628
0x420999a4 - esp_event_loop_run_task
    at /home/reinis/projects/rfid/training/w5500-mdns/.embuild/espressif/esp-idf/release-v4.4/components/esp_event/esp_event.c:115
```

Or in few cases with same code with Segfault...

Am I the only one?

I did the same thing half a year ago with `esp-idf-svc = { git = "https://github.com/otakuto/esp-idf-svc", rev = "b95d43c13fc9418b745a9521d4d9498140760eac"}` and it was working fine. 

Code I run
Cargo.toml
```toml
[profile.release]
opt-level = "s"

[profile.dev]
debug = true # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
pio = ["esp-idf-sys/pio"]

[dependencies]
esp-idf-hal = "0.39.3"
esp-idf-sys = { version = "0.31.11", features = ["binstart"] }
anyhow = "1"
embedded-hal = "1.0.0-alpha.9"
# esp-idf-svc = { version = "0.43.3" }
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc", rev = "790fee01f9b70743dcfa46724b7ed34943832c66"}

embedded-svc = "0.23.1"
log = "0.4.17"

[build-dependencies]
embuild = "0.30.4"
anyhow = "1"

```

```rust
use anyhow::{bail, Result};
use embedded_svc::wifi::{ClientConfiguration, Configuration, Wifi};
use esp_idf_svc::eventloop::EspSystemEventLoop;

use esp_idf_svc::netif::{EspNetif, EspNetifWait};
use esp_idf_svc::nvs::EspDefaultNvsPartition;
use esp_idf_svc::wifi::{EspWifi, WifiWait};
use esp_idf_sys as _; // If using the `binstart` feature of `esp-idf-sys`, always keep this module imported
use log::info;
use std::io::{Read, Write};
use std::sync::Mutex;

use std::sync::Arc;
use std::thread;
use std::time::Duration;

use esp_idf_hal::peripherals::Peripherals;

use embedded_hal::digital::OutputPin;

mod w5500;
mod wifi;

const SSID: &str = "test";
const PASS: &str = "test";

fn main() -> Result<()> {
    // Temporary. Will disappear once ESP-IDF 4.4 is released, but for now it is necessary to call this function once,
    // or else some patches to the runtime implemented by esp-idf-sys might not link properly.
    esp_idf_sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    let sysloop = EspSystemEventLoop::take()?;

    let nvs = EspDefaultNvsPartition::take()?;

    let peripherals = Peripherals::take().unwrap();

    let _wifi = {
        let modem = peripherals.modem;
        let sysloop = sysloop.clone();
        let nvs = nvs.clone();
        use std::net::Ipv4Addr;

        let mut wifi = Box::new(EspWifi::new(modem, sysloop.clone(), Some(nvs))?);

        info!("Wifi created, about to scan");

        let ap_infos = wifi.scan()?;

        let ours = ap_infos.into_iter().find(|a| a.ssid == SSID);

        let channel = if let Some(ours) = ours {
            info!(
                "Found configured access point {} on channel {}",
                SSID, ours.channel
            );
            Some(ours.channel)
        } else {
            info!(
                "Configured access point {} not found during scanning, will go with unknown channel",
                SSID
            );
            None
        };

        wifi.set_configuration(&Configuration::Client(ClientConfiguration {
            ssid: SSID.into(),
            password: PASS.into(),
            channel,
            ..Default::default()
        }))?;

        wifi.start()?;

        info!("Starting wifi...");

        if !WifiWait::new(&sysloop)?
            .wait_with_timeout(Duration::from_secs(20), || wifi.is_started().unwrap())
        {
            bail!("Wifi did not start");
        }

        info!("Connecting wifi...");

        wifi.connect()?;

        if !EspNetifWait::new::<EspNetif>(wifi.sta_netif(), &sysloop)?.wait_with_timeout(
            Duration::from_secs(20),
            || {
                wifi.is_connected().unwrap()
                    && wifi.sta_netif().get_ip_info().unwrap().ip != Ipv4Addr::new(0, 0, 0, 0)
            },
        ) {
            bail!("Wifi did not connect or did not receive a DHCP lease");
        }

        let ip_info = wifi.sta_netif().get_ip_info()?;

        info!("Wifi DHCP info: {:?}", ip_info);

        Some(wifi)
    };

    println!("WIFI Connected...");

    loop {
        // we are using thread::sleep here to make sure the watchdog isn't triggered
        thread::sleep(Duration::from_millis(10));
    }
}
```

ivmarkov, 2022-12-05:
Looks like a smashed stack to me (though it does not report as so). Can you try `CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=4096`?

reinismu, 2022-12-06:
> Looks like a smashed stack to me (though it does not report as so). Can you try `CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=4096`?

Thanks! That fixed it! Is there some doc where I could read up on these things? Feels like this won't be the last time i have to change something in `sdkconfig`

closed 2022-12-06 by reinismu