esp-rs/esp-idf-svc #183: HTTP Get request panics when LTO is enabled

JuliDi, 2022-12-06:
If I turn on Link Time Optimization in Cargo.toml, my http get request panics after calling submit.
From the logs it seems like it starts reading the headers and then crashes.

Funny enough, this happens only in debug builds. For release builds, LTO causes no issues 

Is this a known problem?

If you think this can be fixed on this crate's side, I could try to create an MWE of this.

ivmarkov, 2022-12-06:
LTO on Xtensa esp32* chips is causing known miscompilation problems. Would would be useful is to look at the `esp-rs` `rust` project for the LTO issue and add your code example there, so that folks are aware that there are other examples where we are hitting that.

I highly doubt otherwise that this has anytbhing to do with `esp-idf-svc`.

JuliDi, 2022-12-07:
Thanks for the input. It is really a weird issue.
I tried creating a small example but it does not seem to happen then. So I cannot reproduce it at the moment.
If this still occurs in the published version of the project, I will share a link to that in the other project and let them know.

closed 2022-12-07 by JuliDi