esp-rs/esp-idf-svc #188: WifiDriver stop behavior

teamplayer3, 2022-12-14:
I thought of the behavior of the `WifiDriver` to stop the wifi before configuring it. For the user, it's a little bit unintuitive that the wifi has to be restarted after configuring it (I think of a scenario where the driver gets reconfigured while running).

As in [esp-idf doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html?highlight=esp_wifi_set_mode) stated, the wifi can be configured no matter if wifi is stopped.
> Generally, the Wi-Fi driver should be configured before the Wi-Fi connection is set up. But this is NOT mandatory, which means that you can configure the Wi-Fi connection anytime, provided that the Wi-Fi driver is initialized successfully. However, if the configuration does not need to change after the Wi-Fi connection is set up, you should configure the Wi-Fi driver at this stage, because the configuration APIs (such as [esp_wifi_set_protocol()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv421esp_wifi_set_protocol16wifi_interface_t7uint8_t)) will cause the Wi-Fi to reconnect, which may not be desirable.

I would appreciate it if the API could be used as followed:
```rust
let driver = WifiDriver::new(...);
driver.configure(Ap...);
driver.start();

driver.configure(ApSta...);
driver.start_scan(...);
...
driver.get_scan_result();
driver.configure(Ap...);

driver.stop();
```

I haven't tested all possible scenarios, but the scenario described works. And as in esp-idf docs stated, it shouldn't be a problem.

taunusflieger, 2022-12-15:
I was also initially confused when I saw Wifi two times starting. Would be more intuitive to start it explicitly and not when the driver is created.

ivmarkov, 2022-12-15:
> I was also initially confused when I saw Wifi two times starting. Would be more intuitive to start it explicitly and not when the driver is created.

Wifi is not started when the driver is created.

closed 2022-12-15 by ivmarkov