esp-rs/esp-idf-svc #195: Correct way to wait for wifi to be connected

v-morlock, 2022-12-26:
Hi, 

in the online [training](https://github.com/ferrous-systems/espressif-trainings/blob/main/common/lib/esp32-c3-dkc02-bsc/src/wifi.rs) example for connecting a wifi network, they display a clear way to check whether the wifi is connected & has obtained an ip

```rs
    wifi.wait_status_with_timeout(Duration::from_secs(2100), |status| {
        !status.is_transitional()
    })
    .map_err(|err| anyhow::anyhow!("Unexpected Wifi status (Transitional state): {:?}", err))?;

    let status = wifi.get_status();

    if let wifi::Status(
        ClientStatus::Started(ClientConnectionStatus::Connected(ClientIpStatus::Done(
            _ip_settings,
        ))),
        _,
    ) = status
...
```

However, in newer versions of this library, neither `wifi.wait_status_with_timeout` nor `wifi.get_status` seem to be available. I've found the "WifiWait" struct, however, I'm not really sure how to use that to check the exact wifi status. I can also imagine some solution involving subscribing the sysloob, however, this seems kind of hacky to me.

Is there a recommended way to obtain the current wifi status / to wait for a certain status to occur?

Thanks!

ivmarkov, 2022-12-26:
The training is now out of date, as ~ a month ago backwards-incompatible versions of `esp-idf-hal` and `esp-idf-svc` were released.

Your best bet is to use the `WifiWait` and `EspNetifWait` structs, as demonstrated [here](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/src/main.rs#L1424) and [here](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/src/main.rs#L1434).

And yes, these structs are actually waiting for certain events on the ESP IDF system event loop to pop up, so rather than using those, you can also do something more sophisticated (and async) by subscribing and listening to events on the event loop yourself.

The above is not a hack BTW, but the recommended way to do this stuff.

closed 2022-12-26 by ivmarkov

v-morlock, 2022-12-29:
This makes total sense, thank you so much!