esp-rs/esp-idf-svc #199: Change wifi client hostname

noahpnio, 2022-12-26:
I cannot find a function to change the wifi hostname of my esp32 in client mode. How can I proceed?

taunusflieger, 2022-12-27:
I suggest to look into the ESP IDF documentation. It is important to understand that the Rust crates are sitting on top of IDF. Lots of concepts, but also settings are described there. 

To your question - have a look at https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32/api-reference/kconfig.html?highlight=config_lwip_local_hostname#config-lwip-local-hostname

ivmarkov, 2023-01-16:
There is no such thing as "hostname" in ESP IDF. What is there is a *suggestion* for a hostname that can be passed to your router *if* you configure your ESP with DHCP. Look at the `embedded-svc` `IpConfiguration` struct and specifically, the `hostname` optional parameter when DHCP is specified.

closed 2023-01-16 by ivmarkov