esp-rs/esp-idf-svc #206: Current nvs_get_blob() based NVS access can't read NVS partition created with IDF nvs_partition_gen.py

taunusflieger, 2023-01-03:
I use the IDF nvs_partition_gen.py tool to generate the nvs partition content using the following definition file:
```
key,type,encoding,value
certificates,namespace,,
ca_cert,file,string,../../certificates/aws-iot-rootCA.crt
cert,file,string,../../certificates/cert.pem
priv_key,file,string,../../certificates/privkey.pem
device_data,namespace,,
device_id,data,string,XXXXX-YYY-ZZZZZ
```
When checking access to the data using ```EspCustomNvs::contains()``` none of the keys are found. When replacing in ```EspNvs::len()``` ```nvs_get_blob()```  with ```nvs_get_str()``` the keys are found. While the IDF documentation states that both functions have a identical behavior except that they return different data types, there seems to be a difference.

I would propose to add additional functions which use ```nvs_get_str()``` / ```nvs_set_str()``` for handling access to string based key/value pairs. I'm open to create a PR to resolve the issue.

ivmarkov, 2023-01-07:
The current ESP Nvs Storage implementation uses a serde scheme which is slightly incompatible with the ESP IDF way of (de)serializing things. Basically, it encodes raw data always as a BLOB, if the data is longer than 7 bytes, and as u64 if the data is shorter or equal to 7 bytes. And then n top of this raw data persistence scheme you can layer your own Rust serde. If you want to interface with keys written by ESP IDF C code, or by the nvs partition Gen tool, you need an additional API - exposed directly on the `EspNvs` structure I guess, that can read/write by calling ESP IDF nvs_get_str, nvs_get_i8 and so on - separate getter/setter for each ESP NVS C type.

taunusflieger, 2023-01-07:
Thanks for the detailed explanation. Yes I concluded already that the serde serialization doesn't work together with the nvs partition gen tool. I've implemented nvs_get_str() / nvs_set_str() in directly in esp-idf-svc to enable a more ESP IDF C and toolchain compatible Nvs Storage access. My question would be, does it makes sense (I think it makes sense, otherwise we need to have a Rust based Nvs partition creation tool - maybe at some point ..) to add these two additional non-serde based functions into esp-idf-svc to ease handling of Nvs in case of C / ESP IDF tooling?

ivmarkov, 2023-01-16:
It does, but I would really appreciate if you add a few more as well. As in - for integers, bools etc. This way we can get feature parity with the native NVS API.

taunusflieger, 2023-01-22:
I'm fine with this - plan a PR for next week. I'm assuming we will have two fn for every type get & set explicit for each type and not a generics based solution with something like ```TypeId::of::<i8> == ```TypeId::of::<T>``` as this would create a dependency to ```std``` - correct?

ivmarkov, 2023-01-23:
> I'm fine with this - plan a PR for next week. I'm assuming we will have two fn for every type get & set explicit for each type and not a generics based solution with something like `TypeId::of::<i8> == TypeId::of::<T>` as this would create a dependency to `std` - correct?

`TypeId` is available in `core` so using it will not bring a dependency on `std`. It is more about what is more ergonomic:
* (1) A bunch of get_XXX / set_XXX methods, which cannot panic
* (2) A single get / set pair of methods, which inspects `TypeId` and unsafely transmutes `&T`. And panics when `T` is not one of the supported types

I think I'm in favor of (1).

skylve, 2023-01-23:
Hey folks, sorry I'm new to Rust so disregard if my idea doesn't make sense.

Could we have two traits (e.g `NvsGetter` and `NvsSetter`) that are only defined for the types that are natively supported by idf API (`u8`, `i16`, `u16`, `i32`, `u32`, `i64`, `u64`) and maybe a custom API for `String` (that takes a buffer)? We could use a macro to generate the getter / setter impl.

We could also redefine get_raw / set_raw directly in the `NvsPartitionId` impl for `EspNvs`, leveraging the getter/setters defined above, so that it doesn't mix Nvs and Storage abstraction.

ivmarkov, 2023-01-23:
> Hey folks, sorry I'm new to Rust so disregard if my idea doesn't make sense.
> 
> Could we have two traits (e.g `NvsGetter` and `NvsSetter`) that are only defined for the types that are natively supported by idf API (`u8`, `i16`, `u16`, `i32`, `u32`, `i64`, `u64`) and maybe a custom API for `String` (that takes a buffer)? We could use a macro to generate the getter / setter impl and we would have a single get / set pair methods.
> 
> Example [gist](https://gist.github.com/skylve/5a473009edf73813a2fab4769b54356d) (didn't try it out yet)

Also a possibility, but the custom API for blob / `&[u8]` (that you missed) as well as the one for `&str` is kind of ruining it, so not sure if worth it.

> We could also redefine `get_raw` /` set_raw` directly in the `NvsPartitionId` impl for `EspNvs`, leveraging the getter/setter defined above, so that it doesn't mix Nvs and Storage abstraction.

That would be difficult, but maybe possible. Let's start simple, with the get_XXX / set_XXX approach.

PR #214 (2023-01-27): added nvs get/set for primitive types
This PR resolves #206. It contains a set of set / get functions for Rust primitive types to access NVS. With this PR ```esp-idf-svc``` is closer to the native IDF NVS functionality.

```get_str()``` is special, as it does not return a string, but a slice of u8. The reason is that it should be usable for heap/heapless scenarios. The conversion into the destination string type is left for the user of this function.
closed 2023-01-27 by ivmarkov