esp-rs/esp-idf-svc #211: Panic on nvs_flash_deinit_partition when NvsCustom is dropped

NickUfer, 2023-01-06:
The following code crashes with error code 4353 (`ESP_ERR_NVS_NOT_INITIALIZED`)

```rust
fn main() {
    esp_idf_sys::link_patches();

    match EspCustomNvsPartition::take("config") {
        Ok(ok) => {
            println!("got partition")
        }
        Err(e) => {
            println!("esp error {}", e.code())
        }
    }

    loop {
        thread::sleep(Duration::from_secs(30));
    }
}
```

```
got partition
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: EspError(4353)', /home/nick/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.45.0/src/nvs.rs:119:74
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
abort() was called at PC 0x400ec3d6 on core 0
0x400ec3d6 - _ZN11panic_abort18__rust_start_panic5abort17hd168c25bed14c3b4E
    at ??:??

Backtrace: 0x40082bce:0x3ffb5b40 0x40085591:0x3ffb5b60 0x4008a30e:0x3ffb5b80 0x400ec3d6:0x3ffb5bf0 0x400ec3ca:0x3ffb5c10 0x400de515:0x3ffb5c30 0x400de4a4:0x3ffb5cb0 0x400e7707:0x3ffb5d20 0x400e75d9:0x3ffb5d50 0x400de237:0x3ffb5d80 0x400f4cf4:0x3ffb5db0 0x400f5467:0x3ffb5df0 0x400d3034:0x3ffb5e50 0x400d2a38:0x3ffb5e90 0x400d2b28:0x3ffb5eb0 0x4010ca43:0x3ffb5f00 0x400d2a14:0x3ffb5f20 0x400ebe31:0x3ffb5f40 0x400d2b57:0x3ffb5f60 0x400d40d3:0x3ffb5f90 0x4010dd68:0x3ffb5fb0
0x40082bce - panic_abort
    at /home/nick/IdeaProjects/esp32-tests/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/panic.c:408
0x40085591 - esp_system_abort
    at /home/nick/IdeaProjects/esp32-tests/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/esp_system.c:128
0x4008a30e - abort
    at /home/nick/IdeaProjects/esp32-tests/.embuild/espressif/esp-idf/release-v4.4/components/newlib/abort.c:46
0x400ec3d6 - _ZN11panic_abort18__rust_start_panic5abort17hd168c25bed14c3b4E
    at ??:??
0x400ec3ca - __rust_start_panic
    at ??:??
0x400de515 - rust_panic
    at ??:??
0x400de4a4 - _ZN3std9panicking20rust_panic_with_hook17he970e7307730fac7E
    at ??:??
0x400e7707 - _ZN3std9panicking19begin_panic_handler28_$u7b$$u7b$closure$u7d$$u7d$17h14ef4c42e31b3b43E
    at ??:??
0x400e75d9 - _ZN3std10sys_common9backtrace26__rust_end_short_backtrace17h06d82fdb5d0328ccE
    at ??:??
0x400de237 - rust_begin_unwind
    at ??:??
0x400f4cf4 - _ZN4core9panicking9panic_fmt17h3d71bf8d8318fafcE
    at ??:??
0x400f5467 - _ZN4core6result13unwrap_failed17hd42a26877a69ecb5E
    at ??:??
0x400d3034 - _ZN69_$LT$esp_idf_svc..nvs..NvsCustom$u20$as$u20$core..ops..drop..Drop$GT$4drop17h8f6a5910e1e0c86aE
    at ??:??
0x400d2a38 - _ZN5alloc4sync12Arc$LT$T$GT$9drop_slow17h593bebf4f09fa2edE
    at ??:??
0x400d2b28 - _ZN11esp32_tests4main17h15e60453b661ffe1E
    at ??:??
0x4010ca43 - _ZN3std10sys_common9backtrace28__rust_begin_short_backtrace17hc6be14b146155c8aE
    at ??:??
0x400d2a14 - _ZN3std2rt10lang_start28_$u7b$$u7b$closure$u7d$$u7d$17h7991eaaa6789e11bE.llvm.6876726762958448056
    at ??:??
0x400ebe31 - _ZN3std2rt19lang_start_internal17ha8c3fd326fb69b0fE
    at ??:??
0x400d2b57 - main
    at ??:??
0x400d40d3 - app_main
    at ??:??
0x4010dd68 - main_task
    at /home/nick/IdeaProjects/esp32-tests/.embuild/espressif/esp-idf/release-v4.4/components/freertos/port/port_common.c:141
```

I tried the same with `nvs_flash_init()` before the partition code, same problem

```rust
fn main() {
    esp_idf_sys::link_patches();

    let nvs = EspDefaultNvsPartition::take().unwrap();
    match EspCustomNvsPartition::take("config") {
        Ok(ok) => {
            println!("got partition")
        }
        Err(e) => {
            println!("esp error {}", e.code())
        }
    }

    let _ = nvs;

    loop {
        thread::sleep(Duration::from_secs(30));
    }
}
```

I saw the same problem seems to appear for `nvs_flash_deinit()`

https://github.com/esp-rs/esp-idf-svc/blob/ba1e19d59a0af0dd95ac6c67d0ed9d890059797d/src/nvs.rs#L63-L70

ivmarkov, 2023-01-07:
Have you created a custom partition labeled `config` before trying to use it?

taunusflieger, 2023-01-07:
I'm working successful with custom nvs partitions. You need to define the custom partition in the partitions.csv. In belows example it is the ```conf``` partition.
```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x4000,
otadata,  data, ota,     0xd000,  0x2000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, ,        0x14D000,
ota_0,    app,  ota_0,   ,        0x14D000,
ota_1,    app,  ota_1,   ,        0x14D000,
conf,     data, nvs,     ,        0x3000,
```
And as a second step you need to define the data storage layout and populate the partition with data. This can be done using the nvs_partition_gen.py tool.
```
~/esp/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py generate "../../weatherStationSecrets/anemometer/nvs.csv" conf.bin 0x3000 --outdir ./nvs
```
I would suggest to consult the ESP IDF C documentation for further details.

NickUfer, 2023-01-07:
Thanks for the info that I have to create these tables before usage. I just started programming an ESP32 with this rust library. I didn't program anything in C before, but I'm currently trying with the bindings.

@taunusflieger I tried to flash your partition table with `espflash` and the option `--partition-table`, sadly with no luck. 

```
Error: espflash::partition_table::invalid_partition_table

  × Failed to parse partition table
  ╰─▶ Invalid partition table
```

When I try to flash this simplified table, it works, but the SubType is not `nvs`.

```csv
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory,       ,  1M,
config,   data, fat,           ,  0x3000,
```

When I change the subType of the `config` partition to `nvs`, I get the same `Invalid partition table` error.

NickUfer, 2023-01-16:
I got it fixed. I updated espflash from 1.7.1 to rc version 2.0.0-rc.3 and now this partition table is accepted and there is no panic anymore:

```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory,       ,  1M,
config,   data, nvs,           ,  0x3000,
```

Thanks again for the help though 👍🏼

closed 2023-01-16 by NickUfer