esp-rs/esp-idf-svc #212: Garbage data in SPI when trying to use W5500 ETH chip

reinismu, 2023-01-16:
I used https://github.com/ivmarkov/rust-esp32-std-demo with W5500 chip. From what I have noticed it returns random data on SPI line (As far as I can see. Mostly zeros, but some numbers get in here and there).

These are ports I used

```rust
    #[cfg(feature = "w5500")]
    let eth = eth_configure(
        &sysloop,
        Box::new(esp_idf_svc::eth::EspEth::wrap(EthDriver::new_spi(
            peripherals.spi2,
            pins.gpio38, // int
            pins.gpio41, // sclk
            pins.gpio40, // sdo
            pins.gpio39, // sdi
            esp_idf_hal::spi::Dma::Disabled,
            Some(pins.gpio42), // cs
            Some(pins.gpio37), // rst Option::<gpio::Gpio28>::None, //
            SpiEthChipset::W5500,
            20.MHz().into(),
            Some(&[0x02, 0x00, 0x00, 0x12, 0x34, 0x56]),
            None,
            sysloop.clone(),
        )?)?),
    )?;
```

I tried using https://github.com/espressif/esp-idf/tree/master/examples/ethernet/basic and it worked perfectly fine

Used config
```
CONFIG_EXAMPLE_USE_W5500=y
CONFIG_EXAMPLE_ETH_SPI_HOST=1
CONFIG_EXAMPLE_ETH_SPI_SCLK_GPIO=41
CONFIG_EXAMPLE_ETH_SPI_MOSI_GPIO=39
CONFIG_EXAMPLE_ETH_SPI_MISO_GPIO=40
CONFIG_EXAMPLE_ETH_SPI_CLOCK_MHZ=36
CONFIG_EXAMPLE_ETH_SPI_CS0_GPIO=42
CONFIG_EXAMPLE_ETH_SPI_INT0_GPIO=38
CONFIG_EXAMPLE_ETH_SPI_PHY_RST0_GPIO=-1
CONFIG_EXAMPLE_ETH_SPI_PHY_ADDR0=1
```

I'm very confused now. What could have gone wrong here. 

Any tips on what else I could try? Currently waiting for "Saleae Logic 8" to investigate further on what happens there.

Thanks

p.s. I opened an PR to fail fast in this case https://github.com/espressif/esp-idf/pull/10554

ivmarkov, 2023-01-21:
It is not possible to use this driver without DMA being enabled.

closed 2023-01-21 by ivmarkov

reinismu, 2023-01-21:
> It is not possible to use this driver without DMA being enabled.

I have tried it in every possible way. It still is not working :/ Could you please reopen the ticket?

ivmarkov, 2023-01-21:
I'm unwilling to reopen it, because just recently other folks have used the driver successfully. The fact that it does not run for you does not mean the problem is in the driver.

For one, you seem to have the sdo and sdi li es reverted, at least compared to the basic C example you are using. And again - please enable DMA. If DMA is disabled in the demo, file a bug against the demo and I'll fix the code there.

ivmarkov, 2023-01-21:
BTW: Do you use it with ESP IDF 4.4 or 5.0? There is a big difference between the two w.r.t. ethernet, unfortunately.

reinismu, 2023-01-21:
> For one, you seem to have the sdo and sdi lines reverted, 

GPIO 39 - (MOSI) Master Out Slave In | | (SDI) Slave data In

Am I missing something?

> BTW: Do you use it with ESP IDF 4.4 or 5.0? There is a big difference between the two w.r.t. ethernet, unfortunately.

I use IDF 4.4

I hope that there is just some small stupid small mistake I am making, but ahh I have been at it on and off for more than a month. Getting desperate :/ 

Maybe there is a way I could get you as a contractor for a few hours to help me debug it?

ivmarkov, 2023-01-21:
OK so I just wasted some time reconnecting a w5500 module to a nodemcu lying around. As per the pin setup in the demo.

Then pulled, built (with ESP IDF 4.4) and flashed.
It works (after fixing a few import issues, and switching the DMA from Off to Auto.)

> > For one, you seem to have the sdo and sdi lines reverted,
> 
> GPIO 39 - (MOSI) Master Out Slave In | | (SDI) Slave data In

No. Google it. The whole point of retiring MOSI/MISO terminology was to get rid of the master/slave words. SDI means Serial Data In = MISO
Ditto for SDO - Serial Data Out = MOSI (as this stuff is from the point of view of the, ahem, master).

> 
> Am I missing something?
> I hope that there is just some small stupid small mistake I am making, but ahh I have been at it on and off for more than a month. Getting desperate :/

I strongly suggest you just try to flash the demo, unchanged. Using exactly the pins I've put in the demo (that is, assuming you are on a straight xtensa ESP32 chip). 

> Maybe there is a way I could get you as a contractor for a few hours to help me debug it?

Forget it. I'm too expensive :p

ivmarkov, 2023-01-21:
> No. Google it. The whole point of retiring MOSI/MISO terminology was to get rid of the master/slave words. SDI means Serial Data In = MISO Ditto for SDO - Serial Data Out = MOSI (as this stuff is [...]

Hmmmmmmm. The above is all exactly like this still, but I might still have confused it in the code, as - according to the new terminology, the "sdi" of the w5500 should go in the "sdo" of the esp32. And the other way around - the "sdo" of the w5500 should go in the "sdi" of the esp32 I think. And I'm pretty sure I connected sdi-to-sdi and sdo-to-sdo. But if that's the bug - just swap the wires.

But then again, first flash the demo. Exactly 1:1. To avoid the risk of introducing other variables to the equation. If it does not work, swap the sdi and sdo wires. I don't see what else could be wrong.

reinismu, 2023-01-21:
> No. Google it. The whole point of retiring MOSI/MISO terminology was to get rid of the master/slave words. SDI means  Serial Data In = MISO
> Ditto for SDO - Serial Data Out = MOSI (as this stuff is from the point of view of the, ahem, master).

Well, this seems like that stupid mistake I keep doing and it is not working...

Would like to slap those people who decided to do a terminology change. So many wasted hours. I bet 99.999% of people don't even care.

> I strongly suggest you just try to flash the demo, unchanged. Using exactly the pins I've put in the demo (that is, assuming you are on a straight xtensa ESP32 chip).

Will get in the office and will try now with my terminology fixed. Feels like that should do the trick.
I did use the code from the demo before as well. So wrong pins most likely

> Forget it. I'm too expensive :p

I would pay around a thousand for a few hours of your time. 

Anyways big thanks for pointing out the flaw! Will update if everything went fine on Monday.

ivmarkov, 2023-01-21:
Post here if it did work or if it still didn't.

ivmarkov, 2023-01-21:
> I would pay around a thousand for a few hours of your time.

FWIW this stuff is still just my hobby I'm doing over my spare time. JFYI. For now I want to keep it that way (unpaid). In the future, we'll see.

reinismu, 2023-01-23:
Changed to and it works
```rust
let _eth = w5500::init_w5500(
        &sysloop,
        Box::new(EspEth::wrap_all(EthDriver::new_spi(
            peripherals.spi2,
            pins.gpio10, // int
            pins.gpio13, // sclk
            pins.gpio12, // sdo
            pins.gpio4, // sdi
            esp_idf_hal::spi::Dma::Auto(4096),
            Some(pins.gpio11), // cs
            Option::<gpio::Gpio28>::None, //Some(pins.gpio37), // rst Option::<gpio::Gpio28>::None, //
            SpiEthChipset::W5500,
            36.MHz().into(),
            Some(&[0x02, 0x00, 0x00, 0x12, 0x34, 0x56]),
            None, //Some(1),
            sysloop.clone(),
        )?, netif)?),
    )?;
```

Wasted so much time on this
[image]

reinismu, 2023-01-23:
It interesting that it stops working when I set `mac_addr` to None or `rst` to GPIO :thinking:

ivmarkov, 2023-01-23:
> It interesting that it stops working when I set `mac_addr` to None or `rst` to GPIO 🤔

The SPI card does not have a burned-in MAC address, so you have to set  something. As per ESP-IDF documentation. RST should work actually, it is set and connected to the card in the demo.