esp-rs/esp-idf-svc #213: `EspWifi::start_scan` returns `ESP_FAIL` on `master` (succeeds on release)

hawkw, 2023-01-18:
On an ESP32-C3 (the Adafruit QT Py ESP32-C3 devboard), and on the `master` branch of this repo (commit a866ff3f7d68990f5f4b63d764b177a66b772740), calling `EspWifi::scan`  fails with the `ESP_FAIL` error code. On the current crates.io release, the WiFi scan succeeds.

The error appears to be returned by the `EspWifi::start_scan` function, rather than by `get_scan_results` --- I determined this by copying the implementation of `EspWifi::scan` into my project, and logging the results of the calls to both `start_scan` and `get_scan_results`.

hawkw, 2023-01-18:
It seems like several commits to `master` since v0.45.0 touched the WiFi scan code: 3d69a6522cb1bc71e899271ce7410754f4fa528d, 8aac6d0dd41953f16c58c85f14ebb7720c719309, 93591c1c57ba0e217ae33678e70212175648dde0, and b052a0d227da0e234c9d395f38aa9e4118aeaa5e in particular.

dragazo, 2023-03-07:
I just ran into this issue in my project as well with a different board: ESP32-S3 DevKitC-1. Unfortunately it seems like the only workaround has been to downgrade to latest release, which also forces me to use esp-idf v4.4 rather than v5.0.

ivmarkov, 2023-03-09:
@hawkw @dragazo we accept PRs!

hawkw, 2023-03-09:
> @hawkw @dragazo we accept PRs!

I'm happy to try and put a patch together, but I'm not at all sure what actually caused the regression --- if you have any suggestions for a starting point, I can try to get a fix together.

dragazo, 2023-03-10:
@hawkw From binary searching through the commits with a minimal test program, it looks like the culprit is 3d69a65 (later changes might also be problematic, but this was the first one to break it). I don't have time to look into it now, but I might get a chance next week.

teamplayer3, 2023-03-10:
Try to configure the wifi in station mode before scanning. The new scan interface can only be used after configuring the wifi in station mode manually.

```rust
wifi.set_configuration(...);
wifi.start();
let res = wifi.scan();
```

ivmarkov, 2023-03-10:
@teamplayer3 can it also be configured in ap or mixed mode during scanning?

teamplayer3, 2023-03-10:
> @teamplayer3 can it also be configured in ap or mixed mode during scanning?

@ivmarkov you mean the following scenario:

```rust
wifi.set_configuration(&Configuration::Client(ClientConfiguration::default()));
wifi.start();
wifi.start_scan();

wifi.set_configuration(&Configuration::Ap(AccessPointConfiguration::default()));

let res = wifi.get_scan_result();
```

ivmarkov, 2023-03-10:
No there is a thirdmode - mixed - where the radio is simultaneously in ap and sta mode. Is scanning possible in that mode? The other questions is whether scanning is possible in AP mode, but I guess the answer is no.

In any case neither of those are showstopper for anything.

ivmarkov, 2023-03-10:
@dragazo @hawkw please confirm that once you apply proper configuration and start the driver, scanning works. I want to close this once you confirm.

ivmarkov, 2023-03-10:
> Why would you need to start the wifi at all before scanning, let alone as a client? Isn't the point of scanning to know what's available before starting as a client? Is that just a change to esp-idf that you have to accommodate?
> 
> Also, I'll have to check later if this is still the case, but when I was trying to get it working in my project, I actually was starting it - but I might have only started it in AP mode since I didn't have the necessary information to connect as a (real) client yet. But apparently you can connect as a fake client using this default client config. Couldn't scan automatically detect this scenario and do it for you as backwards compatibility? Or at least this should be documented on `EspWifi::scan` and `EspWifi::start_scan`.

I do not remember how my code used to work before, but I won't be surprised if when you used to call scan, I started the client for you if not started already.

The point is that such automations get in the way of more advanced use cases, so we should better not do these.

As for documentation - sure - I'll put a one liner about the new behavior.

ivmarkov, 2023-03-10:
Confirmed: I used to stop the driver if started, reconfigure it in STA mode and then start it again.

dragazo, 2023-03-10:
@ivmarkov I just confirmed that starting as default client does indeed fix it for my test program. And now that you mention that it used to stop and restart in STA mode, I agree this is the better approach. As for the docs, I'd mention the STA mode thing, and also that `Default::default()` for client is sufficient - I would have probably tried this during debugging if I'd known that was valid.

closed 2023-03-11 by ivmarkov