esp-rs/esp-idf-svc #217: EspOta is not "async safe"

karlri, 2023-01-27:
It appears that if you await async tcp socket reads between `EspOta::new()` and `EspOta::complete()`, then you get a `stack overflow in task "pthread"` on call to `EspOta::complete()`. Note that I call all functions from task `Main`.

If no socket read future was awaited between `EspOta::new()` and `EspOta::complete()`, then it appears to work as intended.

I fixed my problem by turning async TcpStreams into sync blocking TcpStreams and doing sync reads. That makes it work as intended.

I don't have any idea what to do about this, I guess it's probably a bug in esp-idf? I can't seem to find anything in esp-idf that should indicate that certain functions must not be called during the ota process. I'm using esp-idf version 4.4.

ivmarkov, 2023-01-27:
English text will not help here. Show me concrete code and then we have something to discuss.

ivmarkov, 2023-02-02:
No code feedback provided, so I'll close for now. Feel free to re-open and provide a small code snippet, so we can start untangling what is going on.

closed 2023-02-02 by ivmarkov

karlri, 2023-02-02:
I managed to debug and pin down my mistake. I'm replying here in the hope that it may be useful to others. The std implementation for espidf does not support setting thread names, so all tasks spawned using std::thread API will be called "pthread".

The threads with too small stacks were "smol-1" and/or "async-io" that had been spawned by the respective crates. As those threads had names assigned, I simply assumed it was not those. How silly of me!

To increase the stack sizes of the specific treads only, it's possible to set a the thread size env var before smol::block_on:
```rust
    info!("WORKAROUND HACK: fixing smol async-io stack size...");
    env::set_var("RUST_MIN_STACK", "4096");
    let _ = smol::block_on(async {}); // starts the problematic threads with enough stacks
    env::remove_var("RUST_MIN_STACK");
    info!("smol async-io stack size fixed, blocking on main_task...");
    let _ = smol::block_on(main_task);` // threads are already running
```

If anyone is having issues debugging what "pthread" is overflowing its stack, here is a hacky patch for the Rust standard library that sets thread names for threads spawned through std::thread APIs.
```rust
    #[stable(feature = "rust1", since = "1.0.0")]
    #[track_caller]
    pub fn spawn<F, T>(self, f: F) -> io::Result<JoinHandle<T>>
    where
        F: FnOnce() -> T,
        F: Send + 'static,
        T: Send + 'static,
    {
        println!(
            "std::thread::Builder::spawn called from {}",
            Location::caller()
        );

        #[cfg(not(target_os = "espidf"))]
        unsafe {
            self.spawn_unchecked(f)
        }

        // If we're on espidf, this is our one chance to set the task name
        #[cfg(target_os = "espidf")]
        match &self.name {
            None => unsafe { self.spawn_unchecked(f) },
            Some(name) => {
                println!("setting espidf task name {name}");
                let name = CString::new(name.clone())
                    .expect("thread name may not contain interior null bytes");
                let name_ptr = name.as_ptr();
                use crate::ffi::{c_char, c_int};

                // Let the closure drop the name once thread exits.
                // Let's hope the lifetime is long enough :)
                let f = move || {
                    let _ = name;
                    f()
                };

                #[repr(C)]
                pub struct esp_pthread_cfg_t {
                    pub stack_size: usize,
                    pub prio: usize,
                    pub inherit_cfg: bool,
                    pub thread_name: *const c_char,
                    pub pin_to_core: c_int,
                }
                extern "C" {
                    pub fn esp_pthread_get_cfg(p: *mut esp_pthread_cfg_t) -> esp_err_t;
                    pub fn esp_pthread_set_cfg(cfg: *const esp_pthread_cfg_t) -> esp_err_t;
                    pub fn esp_pthread_get_default_config() -> esp_pthread_cfg_t;
                }

                pub type esp_err_t = c_int;
                const ESP_OK: c_int = 0;

                unsafe {
                    // get current
                    let mut esp_pth_cfg: esp_pthread_cfg_t = crate::mem::zeroed();
                    if esp_pthread_get_cfg(&mut esp_pth_cfg) != ESP_OK {
                        // if not set, this happens.
                        esp_pth_cfg = esp_pthread_get_default_config();
                    }
                    let orig_name_ptr = esp_pth_cfg.thread_name;

                    // set thread name for next spawn
                    esp_pth_cfg.thread_name = name_ptr;
                    assert_eq!(esp_pthread_set_cfg(&esp_pth_cfg), ESP_OK);

                    // spawn
                    let ret = self.spawn_unchecked(f);

                    // reset to old
                    esp_pth_cfg.thread_name = orig_name_ptr;
                    assert_eq!(esp_pthread_set_cfg(&esp_pth_cfg), ESP_OK);
                    ret
                }
            }
        }
    }

```

With the patch, you will get errors looking as follows:
```
std::thread::Builder::spawn called from /home/karlri/.cargo/registry/src/github.com-1ecc6299db9ec823/async-io-1.12.0/src/driver.rs:32:14
setting espidf task name async-io
***ERROR*** A stack overflow in task async-io has been detected.
```

karlri, 2023-02-02:
I assumed that smol block_on would run the main loop on the thread that I called block_on from, so I assumed that I would be using the large stack of the main task for async code. That's not the case, async code in block_on gets executed on a spawned thread with a small stack.

ivmarkov, 2023-02-02:
I should not be doing this on a weekday late evening, but since now we are talking in concrete, can't resist! :)

> I managed to debug and pin down my mistake. I'm replying here in the hope that it may be useful to others. The std implementation for espidf does not support setting thread names, so all tasks spawned using std::thread API will be called "pthread".

Indeed. Back in the time when I was upstreaming STD support for ESP-IDF, I passed on the `esp_pthread_set_cfg` approach, as that would've increased the diff size significantly. But now that folks are suffering, perhaps we should reconsider. BTW, the best way to do this is to perhaps actually upstream `esp_pthread_cfg_t` as well as the `esp_pthread_get/set_cfg` [here](https://github.com/rust-lang/libc/blob/master/src/unix/newlib/espidf/mod.rs#L1), and then only use these from the STD code.

(In the meantime, setting a `'static` null-terminated `&str` as a thread name also work with [this API](https://github.com/esp-rs/esp-idf-hal/blob/master/src/task.rs#L170). ... which does not support setting the stack size btw, due to a long standing bug in the ESP IDF STD support that I also need to finally address.)

> The threads with too small stacks were "smol-1" and/or "async-io" that had been spawned by the respective crates. As those threads had names assigned, I simply assumed it was not those. How silly of [...]

I know that the "demo" is still talking about using directly `smol` for doing async work, but in the meantime I think this is doing a dis-service to the advanced folks like you in the community, because `smol` (as in the top-level crate in the `smol-rs` org) comes pre-configured for "bigger" STD environments.

As you correctly found out, the `smol::block_on` code is delegating the blocking task to a thread pool of two threads (`smol-1` and `smol-2`), on which users have little control. Also - and if I'm not mistaken - the default operational mode of the smol executor is a [work stealing](https://smallcultfollowing.com/babysteps/blog/2023/02/01/async-trait-send-bounds-part-1-intro/) type, while for memory constrained MCUs - even if they do support STD - we want a sort of a hybrid between "single-threaded" and "work-dealing" mode, if you refer to the same post from Niko from above.

This does not mean abandoning the `smol-rs` ecosystem, quite the opposite! What it does mean however, is that we should be using *individual crates* from it - tailored/configured a bit to the MCU constraints - rather than the `smol` aggregator crate itself.

For example, I use:
* [This tailorisation](https://github.com/ivmarkov/edge-executor/blob/master/src/lib.rs#L85) of smol-rs's `async-task` - specifically, the `Executor<..., Local>` variation, that implements a thread-per-executor "work dealing" mode of sorts. You can still have multiple executors so as to model different task priorities, but each executor is confined to a single thread. So no "thread-pool" style surprises, and your futures do not have to be `Send` either!
* [This "block-on" variant](https://github.com/ivmarkov/edge-executor/blob/master/src/lib.rs#L349) that really executes on the current thread
*  The `async-io` (the "Reactor") crate as is, but it is really only polling (via the `polling` crate) the socket handles of your network streams, in a single, separate thread. I'm a bit surprised that you had thread-size issues with that, but well, could be. We can still tailor it so that it operates with a bigger stack size, regardless of the ESP IDF pthread stack size default, of course.

And it goes without saying that `embassy-futures` ([who needs `select!` anyway??](https://blog.yoshuawuyts.com/futures-concurrency-3/)), `embassy-sync` and `embassy-time` are super-useful in our context, and we do support them out of the box.

ivmarkov, 2023-02-02:
[This](https://github.com/ivmarkov/ruwm) might provide some extra context as to how I'm using async on top of ESP-IDF. Sadly, no "reactor" (a.k.a. `async-io`)-related stuff there yet, as I'm able to "get by" with the HTTP & MQTT services provided by ESP IDF itself, so no real need to async await on sockets from Rust code (yet). Once [this](https://github.com/ivmarkov/edge-net) takes off, then it would be a different story, but for now, only the [websocket-client-over-plain-http](https://github.com/ivmarkov/edge-net/blob/master/src/asynch/ws.rs) code-path in there is kinda working.