esp-rs/esp-idf-svc #220: Secure websocket server not working

reinismu, 2023-01-30:
Trying to use SSL with websockets errors. Just a simple `GET` request works

```
I (1241448) esp_https_server: performing session handshake
E (1244623) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x0050
E (1244623) esp_https_server: esp_tls_create_server_session failed
W (1244628) httpd: httpd_accept_conn: session creation failed
W (1244633) httpd: httpd_server: error accepting new connection
I (1247138) esp_https_server: performing session handshake
E (1250333) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (1250333) esp_https_server: esp_tls_create_server_session failed
W (1250338) httpd: httpd_accept_conn: session creation failed
W (1250343) httpd: httpd_server: error accepting new connection
```

```rust
        let cert_x509: X509 = X509::der(CERTIFICATE);
        let priv_x509: X509 = X509::der(PRIVATE_KEY);
        let conf = Configuration {
            server_certificate: Some(cert_x509),
            private_key: Some(priv_x509),
            ..Default::default()
        };
```

```rust
.ws_handler::<_, WsError>("/ws", move |con| {
```

Based on this example I assume that it should a supported feature.
https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_server/wss_server/main/wss_server_example.c

ivmarkov, 2023-02-02:
Based on the error messages, I find it very hard to believe that a GET request over TLS/HTTPS works, while the websockets don't. I mean, **both** go via the TLS tunnel yet it seems the TLS tunnel cannot be established in the first place?

Slo are you sure that "a simple (HTTPS) GET request works"?

reinismu, 2023-02-02:
> Based on the error messages, I find it very hard to believe that a GET request over TLS/HTTPS works, while the websockets don't. I mean, **both** go via the TLS tunnel yet it seems the TLS tunnel cannot be established in the first place?
> 
> So are you sure that "a simple (HTTPS) GET request works"?

Ye I'm positive. Those error messages also show up for GET requests, but it works out in the end.

reinismu, 2023-02-03:
Well I recompiled today and now it magically works.. Interesting.

Will close ticket for now. If something comes up will reopen

closed 2023-02-03 by reinismu